openscad/openscad #4922: OpenSCAD crashes with twisted linear_extrude

ErQrYfkrju, 2024-01-05:
**Describe the bug**
OpenSCAD crashes when using custom $fn value with code below:
```
module spiral(height=150, radius=25){
    linear_extrude(height=height, twist=-360*4){
        translate([radius-5,0,0])square(5,5);        
    }
    linear_extrude(height=height, twist=360*4){
        translate([radius-5,0,0])square(5,5);        
    }
}
spiral();
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create new project
2. Copy provided code
3. Render it
4. place $fn=60, on top of file or spiral($fn=60);
5. Render it again, my application crashes all the time.

**Expected behavior**
Expect smooth rendering

**Environment and Version info (please complete the following information):**
 - OS: Archlinux
 - System: Ryzen 1500x, 32Gb,
 - OpenSCAD Version OpenSCAD 2024.01.05 (pacaur -S openscad-git)

**Library & Graphics card information**
```
OpenSCAD Version: 2024.01.05
System information: Linux 6.1.58-1-lts #1 SMP PREEMPT_DYNAMIC Sun, 15 Oct 2023 18:38:43 +0000 x86_64 Arch Linux 8 CPUs 31.30 GB RAM
User Agent: OpenSCAD/2024.01.05 (Linux x86_64; Arch Linux)
Compiler: GCC "13.2.1 20230801" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Qt version: 5.15.11
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.78.0
lodepng version: 20210627
libzip version: 1.10.1
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 8.2.2
cairo version: 1.18.0
lib3mf version: 2.2.0
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /usr/bin
Documents Path: /home/alexey/.local/share
User Documents Path: /home/alexey
Resource Path: /usr/share/openscad
User Library Path: /home/alexey/.local/share/OpenSCAD/libraries
User Config Path: /home/alexey/.config/OpenSCAD
Backup Path: /home/alexey/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/alexey/.local/share/OpenSCAD/libraries
  /usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /home/alexey/.local/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /home/alexey/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/alexey/Library/Fonts
  /usr/share/fonts/100dpi
  /usr/share/fonts/75dpi
  /usr/share/fonts/OTF
  /usr/share/fonts/TTF
  /usr/share/fonts/Type1
  /usr/share/fonts/adobe-source-code-pro
  /usr/share/fonts/adobe-source-han-sans
  /usr/share/fonts/adobe-source-han-serif
  /usr/share/fonts/adobe-source-sans
  /usr/share/fonts/adobe-source-serif
  /usr/share/fonts/awesome-terminal-fonts
  /usr/share/fonts/caladea
  /usr/share/fonts/cantarell
  /usr/share/fonts/carlito
  /usr/share/fonts/croscore
  /usr/share/fonts/cyrillic
  /usr/share/fonts/droid
  /usr/share/fonts/encodings
  /usr/share/fonts/gentium
  /usr/share/fonts/gnu-free
  /usr/share/fonts/gsfonts
  /usr/share/fonts/inter
  /usr/share/fonts/joypixels
  /usr/share/fonts/junicode
  /usr/share/fonts/liberation
  /usr/share/fonts/libertinus
  /usr/share/fonts/misc
  /usr/share/fonts/noto
  /usr/share/fonts/noto-cjk
  /usr/share/fonts/sarasa-gothic
  /usr/share/fonts/tex-gyre
  /usr/share/fonts/ubuntu
  /usr/share/fonts/wenquanyi
  /usr/share/fonts/encodings/large
  /usr/share/fonts/wenquanyi/wqy-microhei
  /usr/share/fonts/wenquanyi/wqy-microhei-lite
  /usr/share/fonts/wenquanyi/wqy-zenhei

GL context creator: GLX (new)
GLX version: 1.4
PNG generator: lodepng

GLEW version: 2.2.0
OpenGL Version: 4.6.0 NVIDIA 535.113.01
GL Renderer: NVIDIA GeForce GT 1030/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

pca006132, 2024-01-06:
what experimental options did you enable?

t-paul, 2024-01-06:
> `Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output`

There are none enabled, they would have a `*` after the name when enabled.

pca006132, 2024-02-15:
can you reproduce with the latest development snapshot? seems fixed.

kintel, 2024-04-01:
Closing as fixed

closed 2024-04-01 by kintel