openscad/openscad #4925: Strange render lines with manyfold option on beta

fedekrum, 2024-01-07:
**Describe the bug**
I have used a library for bezier design ever since.
https://www.thingiverse.com/thing:2170645

Since nighty builds with  manifold option, things are SUPER FAST, but I get strange render lines.

[image]

with legacy rendering (slow) I get this:

[image]

with the newer manifold option (Super FAST) I get this:

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Get latest beta
2. download library
3. use my example file

```
use <bezier.scad>
$fn=180;

module easy_bezier(v,fuga,steps,h,dis){
    if (dis==1){  
        color([0.5,0.5,0.5,0.5]) translate([v[1][0],v[1][1],0.6]) circle(d=2);
        color([0.5,0.5,0.5,0.5]) translate([v[2][0],v[2][1],0.6]) circle(d=2);
    }
    if (dis==1) {
        color([1,0,0,0.5]) BezCubicFillet( v, fuga, steps, h); 
    }
    else  {
        color([0.5,0,0])  BezCubicFillet( v, fuga, steps, h);
    }

};

module aleta (h,steps=15){
    union(){
        easy_bezier( [[0.43,9.0],[1.9,7.0],[3.0,5.3],[4.1,2.0]], [6.5,5.5], steps, h,0);
        easy_bezier( [[4.1,2.0],[4.6,0.4],[5.0,0.3],[6.0,0.2]], [6.5,5.5], steps, h,0);
        easy_bezier( [[6.0,0.2],[7.5,0.22],[8.5,0.1],[9.5,0]], [6.5,5.5], steps, h,0);
        easy_bezier( [[9.5,0],[11.0,-0.1],[11.5,0.6],[11.5,1.5]], [6.5,5.5], steps, h,0);
        easy_bezier( [[11.5,1.5],[11.5,7.0],[6.8,10.0],[4.0,11.0]], [6.5,5.5], steps, h,0);
        easy_bezier( [[4.0,11.0],[1.7,12.0],[-1.2,11.0],[0.43,9.0]], [6.5,5.5], steps, h,0);
    }
};

difference() {
    cube(size = [20,20,2], center = false);
    x();        
}

    
module x (){
translate([5,5, 1])
aleta (h=2,steps=15);

}
```

**Environment and Version info (please complete the following information):**
 - OS: macOS 14 Sonoma
 - OpenSCAD-2024.01.05

**Library & Graphics card information**
```
OpenSCAD Version: 2024.01.05 (git b1160fa43)
System information: macOS 14.2.1 x86_64 iMac18,3 8 CPUs 24.00 GB RAM 
User Agent: OpenSCAD/2024.01.05 (git b1160fa43) (macOS 14.2.1 x86_64 iMac18,3)
Compiler: Clang "14.0.3 (clang-1403.0.22.14.1)" 64bit
MinGW build: No
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Qt version: 5.15.7
QScintilla version: 2.13.3
InputDrivers: 
GLib version: 2.76.3
lodepng version: 20210627
libzip version: 1.9.2
fontconfig version: 2.14.1
freetype version: 2.12.1
harfbuzz version: 6.0.0
cairo version: 1.18.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold*, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /Applications/-3D Print/OpenSCAD beta/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/fede/Documents
User Documents Path: /Users/fede/Documents
Resource Path: /Applications/-3D Print/OpenSCAD beta/OpenSCAD.app/Contents/Resources
User Library Path: /Users/fede/Documents/OpenSCAD/libraries
User Config Path: 
Backup Path: /Users/fede/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/Users/fede/Documents/OpenSCAD/libraries
/Applications/-3D Print/OpenSCAD beta/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/Users/fede/.local/share/fonts
/usr/local/share/fonts
/usr/share/fonts
/Users/fede/.fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/Users/fede/Library/Fonts
/System/Library/Fonts/Supplemental
GLAD version: 2.0.4
OpenGL Version: 2.1 ATI-5.2.4
GL Renderer: AMD Radeon RX 580 OpenGL Engine
GL Vendor: ATI Technologies Inc.
RGBA(8888), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(888-1), depth(0), stencil(0)

GL Extensions:
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix
```

kintel, 2024-01-07:
`BezCubicFillet()` doesn't seem to exist in the library you linked..

fedekrum, 2024-01-07:
You are right.
I am using an older version.
[Bezier_library_for_OpenScad.zip](https://github.com/openscad/openscad/files/13853969/Bezier_library_for_OpenScad.zip)
Try that one.

pca006132, 2024-01-12:
I think what you see is caused by floating point errors. In general, reserve slight overlap for boolean operands such that these surfaces can be removed cleanly. For your specific case, you probably want to do 2D offset before extrude. Removing these surfaces is a hard problem in general.