openscad/openscad #4926: Array indexing incorrectly

synco, 2024-01-07:
**Describe the bug**
Unusual behaviour due to not rounding the array index correctly?

**Code reproducing the issue**
```
puzzle = "1100000002200222100220000.112222100222200.1011121122122212001222";
dim = 4;
//!!! uncommenting following line gives different answer !!!
// dim = len(puzzle)^(1/3);          // Internally it's a float not quite 4. ?

echo(dim);      // Both cases the result says 4

module piece(s, c) {
    for (x = [0:1:(dim-1)]) {
        for (y = [0:1:(dim-1)]) {
            for (z = [0:1:(dim-1)]) {
                if ( s[dim*dim*x + dim*y + z] == c ) {
                    echo(dim*dim*x + dim*y + z);
                }
            }
        }
    }
}

piece(puzzle, ".");         // Correct answer is 25, 41.
                            // Using formula gives 26, 42.
```

Currently using OpenSCAD Version 2023.04.10. Is this still an issue in latest version?

UBaer21, 2024-01-07:
`echo()` will round the value so what is displayed is not always the value used for calculations. And using floats in ranges (for loop) is not ideal as the condition is never met it will nearly always stop 1 step before use `ceil()`.

thehans, 2024-01-07:
UBaer21 is correct.  `1/3` cannot be exactly represented as a floating point value, and so the result of that cube root exponent operation is not precise either.

To verify the source of the issue, for instances like this where the echo'd value is imprecisely rounded, you can use equality and comparison operators.
```
echo(64^(1/3) == 4); // false
echo(64^(1/3) < 4);   // true
```

closed 2024-01-07 by thehans

synco, 2024-01-07:
Since an array index always needs to be an integer, I'm wondering if it'd make more sense to internally convert the float using rounding instead of flooring. It'd sure make it a lot more intuitive for those who don't want to get into the nitty-gritty of floating point representation in binary. Like what would break if this was done?

jordanbrown0, 2024-01-07:
It would break anything that depends on the conversion being truncation rather than rounding. I don't have a use case handy, but it doesn't seem so obscure that one could assert that it would never happen.

I would also fear that it would just move the problem.  The same sort of floating point imprecision that can turn a result that is mathematically 1 into 0.9999999... could turn a result that is mathematically 0.5 into 0.499999... , and while 0.5 would round up to 1, 0.499999... would round down to zero.

synco, 2024-01-08:
@jordanbrown0  I understand your point, however it'd be rare if your floating maths error built up to ~0.5, and it's obviously a user-coding issue if they are trying to index around 0.5. (Actually it'd be another great opportunity to give a compiler warning if the index is rounded too much)
My point is that since types are hidden in OpenSCAD, (so much so that the echo() hides the issue too), I'd argue it's more important to make it work intuitively.
Love to see a good example for using truncation over rounding.

jordanbrown0, 2024-01-08:
Something like this seems plausible:
```
suitNames = [ "Hearts", "Diamonds", "Spades", "Clubs" ];
card = floor(rands(0,52,1)[0]);
suit = suitNames[card/13];
```

And with respect to 0.5, it wouldn't require that the error build up to 0.5.  Rather, it would require an intended result of 0.5, that calculates out to 0.49999999999999991 or less.  The problem with your example is the "cliff", that 0.9999999something is mathematically very close to one, but truncates to zero.  That same cliff exists for rounding at 0.5; a value that is infinitesimally smaller than 0.5 will round to zero rather than one.

synco, 2024-01-08:
@jordanbrown0, OK great example!!