openscad/openscad #4931: No two openscad runs export the same STL file (output not reproducable)

wocard2, 2024-01-09:
**Describe the bug**

If I run openscad from the command line multiple times with the same input file, exporting to STL, I get different file contents for each invocation. The input file does not use random numbers. Does openscad use random numbers for computing the output or is it worthwile to create a minimal input file that shows the problem?

**To Reproduce**

for i in $(seq -w 1 99); do openscad -o test-$i.stl --export-format asciistl test.scad; done
sha256sum test-*.stl | awk '{print $1}' | sort -u

**Expected behavior**

I expect all the files to be identical, ie, the second line should display just one line.

**Code reproducing the issue**

I'll provide code if openscad is supposed to create reproducable output.

**Screenshots**

The first two lines of two random specimen of the output:

solid OpenSCAD_Model
  facet normal 0.0559764 0.187705 -0.980629

solid OpenSCAD_Model
  facet normal -0.49160424 0.26276952 -0.8302274

**Environment and Version info (please complete the following information):**
 - OS: Gentoo Linux
 - System: x86_64
 - OpenSCAD Version: 2021.01 (Gentoo-provided)
 - OpenSCAD version: commit b1160fa4327f0a7914f6d48f3a5fed78fb4ab735 (different version, same problem)

**Library & Graphics card information**

Batch mode.

**Additional context**

t-paul, 2024-01-09:
Please check with a snapshot version and enable the `predictible-output` option via Preferences->Features or on command line with `--enable predictible-output`.

wocard2, 2024-01-09:
--enable=predictible-output does the trick indeed: So far, I got 4 identical STL files out of 4. Thanks!

wocard2, 2024-01-09:
That feature seems to be nowhere documented... And maybe it should be enabled by default.

wocard2, 2024-01-09:
...and the spelling should be changed to predictable-output.

pca006132, 2024-01-12:
With manifold enabled, we need to configure manifold to run deterministically when the option is enabled. There are some caveats though:

1. This will make things slightly slower because it disables some optimizations.
2. There is *no guarantee* that the output is the same across versions/compilers. It only ensures that the output is the same *for the same binary and input*.

nophead, 2024-01-12:
I run a Python script to make my STLs from GCAL deterministic. It just sorts the facets so the order doesn't matter. Will that work with Manifold or does it produce different facets on different compilers?

pca006132, 2024-01-12:
It is not enough for manifold, different compilers should produce the same output, but so far we have not tried that.

Reasons for non-deterministic behavior (across versions):
1. We assume mesh boolean operations are associative, but this is not really the case, which can cause subtle differences. So the result depends on the reordering heuristic, which may change across versions.
2. Our mesh simplification heuristic is not perfect, updating the version may change it.

pca006132, 2024-01-12:
The result from CGAL can also change if you use fast-csg due to inexact arithmetic, or any round-trip from PolySet due to it storing floating point numbers. Things like math function implementations can change with different versions of libc. I don't think it is reasonable to expect the exact same result across different versions, there is no software that can guarantee the exact same result across versions for all possible inputs. Floating point operation results can be different across different hardware as well due to different underlying arithmetic precision, e.g. x87 can use 80-bit precision for floating point computation and you need to explicitly turn this off if you want portable floating point results.

pca006132, 2024-01-12:
And according to https://stackoverflow.com/questions/27149894/does-any-floating-point-intensive-code-produce-bit-exact-results-in-any-x86-base, bitwise identical result is not guaranteed by the C/C++ standard. You need to do everything in exact arithmetic or write assembly and verify it on every single platform if you want bitwise identical results across versions even when the algorithm is the same.

wocard2, 2024-01-12:
My use case is this: I'm replacing a lot of magic constants in my code with formulas computing those values from a small set of constants describing my design. To verify those formulas, I compared the output before and after that substitition and was quite astonished that this didn't work.

pca006132, 2024-01-12:
For your specific use case, you should probably round your variables to specific precision, as the optimizations you do may cause the floating point number to change slightly. But other than that it should be fine.

nophead, 2024-01-12:
CGAL uses rationals, not floating point. I have had exactly the same STLs for years from OpenSCAD on different PCs and operating systems. Only the order of facets changed due to some container using machine addresses for ordering I think. 

My build systems needs to know if an STL has changed or not and Github needs to know if I publish a project.

pca006132, 2024-01-12:
fast-csg uses floating point, and the arithmetic in openscad uses floating point. I am not trying to force you into using manifold, I am just reminding you that the level of determinism you want never existed in the first place, maybe it was more deterministic, but it is not totally deterministic across versions and machines.

kintel, 2025-03-18:
This should be fixed by latest manifold (#5740)

closed 2025-03-18 by kintel