openscad/openscad #4934: Manifold doesn't support Minkowskis convexity

UBaer21, 2024-01-12:
Version 2024.01.03

code:
```
difference(){
  minkowski(convexity=5){
    difference(){
      cylinder(10,d=15);
      cylinder(30,d=14,center=true);
    }
    sphere(.5);
  }
  cube(100);
}
```
results in 
[image]

jordanbrown0, 2024-01-13:
It it's helpful, I think the convexity of a Minkowski sum is likely limited to the sum of the convexities of the individual operands, so it may be possible to do something reasonable by default.

pca006132, 2024-01-13:
I think the implementation forgot to propagate the convexity parameters. I wonder if we have the same issue with fast-csg as well.

PR #4935 (2024-01-14): Bugfix: Forward convexity from node to geometry for Manifold minkowski
Fixes #4934
closed 2024-01-14 by kintel