openscad/openscad #4938: Confused variable handling

smurfix, 2024-01-18:
**Describe the bug**
Overwriting a variable confuses OpenSCAD into assigning `undef`.

**To Reproduce**

```
k=3;
j=99;
k=j;
echo(j,k);
```

This results in:
```
[WARNING: k was assigned on line 1 but was overwritten in file linear_extrude.scad, line 3]
[WARNING: Ignoring unknown variable 'j' in file linear_extrude.scad, line 3]
ECHO: 99, undef
```
… which does not make much sense IMHO.

Deleting the first line causes the code to run without errors, as expected.

**Expected behavior**
This should only emit one warning and `k` should be `99`.

**Environment and Version info (please complete the following information):**
 - OpenSCAD Version 2021.01, Debian Stable

t-paul, 2024-01-18:
Reassignment is not allowed, it *should* be an error, but can't be (yet?) due to backward compatibility. There is no easy solution, maybe some sort of language version 2 may help getting rid of such cases.

What happens is that line 3 is effectively moved to replace line one, causing the unexpected warning.

smurfix, 2024-01-20:
OK, thanks.

closed 2024-01-20 by smurfix