openscad/openscad #4941: I have a bug in size handling and I'm not sure what to do.

Ajb2k3, 2024-01-20:
I have created a to generate panels based on 15mm extrusion.
Everything work fine if I use measurements of 50, 100, 200mm but as soon I try different sizes (75mm/74mm) one of the holes doesn't get made.

My script is as follows:
``
// -------------------------------------------
//
//   1515 Panel Maker by Adam Bryant
//
// -------------------------------------------

// Place holder for Size variables.
// M5Stack Extrusion is pre cut to 50, 100, 200, 500mm long.

Length=50; // X Dimension
Width=50; // Y Dimension
Height=15; //1515 Extrusion width
Screw_hole=3.2; // M3 Screw
Wall=4; // Wall thickness.

difference() {
cube([Length,Width,Height],center=true); //Initial Panel size.
translate([0,0,2]) // Moves Cavity Piece to correct height.

minkowski(){
cube([Length-Wall,Width-Wall,Height],center=true); // Cavity size.
sphere(1); // If this is increased, Wall thickness must be increased.
}

rotate([90,0,0])
cylinder(Length+2,Screw_hole/2,Screw_hole/2, center=true);

rotate([0,90,0])
cylinder(Width+2,Screw_hole/2,Screw_hole/2, center=true);
}
``

Can anyone see where I am going wrong?
I'm trying to learn from Justin Gohde and Marius Kintel's book

t-paul, 2024-01-20:
I think you need to swap the rotations of the 2 cylinders. Add a # in front of one to see which is which, like so:

```
#rotate([0,90,0])
cylinder(Width+2,Screw_hole/2,Screw_hole/2, center=true);
```

kintel, 2024-01-20:
t-paul is right, the two cylinders are just mixed up. Closing as this is not an OpenSCAD bug.

In the future, please use the mailing list or IRC channels to discuss modeling issues. GitHub tickets are for reporting defects in OpenSCAD itself.

closed 2024-01-20 by kintel