openscad/openscad #4942: Minkowski 2D and 3D

revk, 2024-01-20:
**Is your feature request related to a problem? Please describe.**
Just a common use case, e.g. Minkowski of text with a cone, etc. Have to linear_extrude the text first, and then allow for the extruded height in result.

**Describe the solution you'd like**
Allow a 2D and 3D minkowski, i.e. treat the 2D as a 3D infinitely thin.

**Describe alternatives you've considered**
As above, linear_extrude first and then allow for extruded height.

**Additional context**
Add any other context or screenshots about the feature request here.

[image]

UBaer21, 2024-01-20:
There is the experimental feature `roof()` that will chamfer a 2D into 3D.
Also there is WIP for
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/WIP/Offset_Extrude

revk, 2024-01-20:
Ooh, that covers my main use cases, though I suspect a general 2D+3D minkowski would be useful anyway. Thanks for pointing that out though.

GloinTheDark, 2024-02-24:
Please escalate this issue. I have use cases beyond simple chamfering. I am working on a router library. I am currently using minkowski() with "simulated" 2D objects (3D objects with very thin dimensions). This is very slow. Allowing 2D objects would reduce the complexity by at least a factor of 4. Another drawback of this approach is the visible artifacts incurred by using thin 3D shapes.
As I understand the Minkowski Sum algorithm it should be able to work with 2D or 3D input including the mixing of those.
As pointed out a 2D geometry is a 3D geometry with zero thickness. As such feeding it straight into the minkowski() operator might "just work" if the 2D checks are removed and the 2D geometry is converted to a "zero height extrusion".

jordanbrown0, 2024-02-24:
Erm... escalate it to whom?  Do remember that you're dealing with a small group of volunteers who do work based on their interests, knowledge, and available time.  (For instance, I can be helpful with language, UI, and general programming issues, but am useless when it comes to the geometry processing.)  This isn't a company, where management can just assign somebody to a task, or can spend the money to hire a new person.  The people who have the knowledge to work on geometry issues are already working on geometry issues - and, from what I've seen from the sidelines, they are universally important issues like overall performance and robustness.

GloinTheDark, 2024-02-24:
I understand. Escalate was a bad choice of words. If there is no one currently interested in solving this issue, I could take a crack at it.

jordanbrown0, 2024-02-24:
Thank you for understanding.  And yes, everybody would be delighted if you could take a look.  You don't mention what platform you're on, but if you want to work on Windows I can help get you started.  (And maybe on Linux, but there I'm not as familiar.)  But this is not the right forum for that discussion.  openscad (at) jordan.maileater.net if you have questions.  You might also want to join the mailing list.

GloinTheDark, 2024-02-24:
I decided to explore how much support OpenSCAD already had for degenerate (specifically zero volume) geometry.
I found that this works as expected...
`minkowski() {
    polyhedron(
        points = [
            [0,0],
            [10,0],
            [0,10],
        ], 
        faces = [
            [0, 2, 1],
            [0, 1, 2],
        ], 
        convexity = 1
    );
    sphere(r = 1, $fn = 36);
}
`
Which gives hope. This means that the algorithm works fine with degenerate 3D geometry.

GloinTheDark, 2024-02-24:
Um, this also works...
```
module line(start = [0,0,0], end = [10,0,0]){
    polyhedron(
        points = [
            start,
            (start + end) / 2,
            end
        ], 
        faces = [
            [0, 2, 1],
            [0, 1, 2],
        ]
    );
}

minkowski() {
    line([0,0,0], [10,10,10]);
    sphere(d = 1, $fn = 36);
}
```
`line()` has no surface area or volume.

revk, 2024-02-24:
I wonder if I can linear extrude to zero height validly?

UBaer21, 2024-02-24:
@GloinTheDark  be careful with these, e.g. a hull() of two lines to create a 2D will fail with minkowski while a hull() of two lines to create a 3D will work.

Harvie, 2025-01-26:
Yeah, i think it would be super useful to support 2D objects in minkowski as there's no real reason not to...

Instead of:

```c
minkowski() {
	cube(10);
	circle(d=4, $fn=100);
}
```

Now i have to do this:

```c
minkowski() {
	cube(10);
	cylinder(d=4, h=0.000001,  $fn=100);
}
```

[image]

kintel, 2025-01-26:
@Harvie There is some ongoing work to support this in #3505

revk, 2025-01-26:
Cool