openscad/openscad #4943: center()

yahbluez, 2024-01-20:
I miss the possibility to center or place an imported object.

It should be possibly to have a center() module that centers his childs.

center() import();
Now the imported object should be centered.

If center could take parameters to set one or more axes that would be nice to.

jordanbrown0, 2024-01-21:
Written the way you describe, it's not as easy as it seems, because in general during a preview you can't figure out anything at all about the child object.  You have to actually *do* the boolean operations, and preview doesn't actually do them.  (It does black magic instead.)  PR#4478 has mechanisms that can be used to implement such a module, at the expense of having to render the child.

There could also be centering (or, more generally, alignment) options on the import.  The import can see the data it's importing, so can know how to center it.

yahbluez, 2024-01-21:
Thank you

> There could also be centering (or, more generally, alignment) options on the import. The import can see the data it's importing, so can know how to center it.

How can i do that?

For example importing an STL and get the size xyz of a surrounding block and the position of one corner.

jordanbrown0, 2024-01-21:
*You* can't.  But OpenSCAD could more or less easily be modified to allow it.

jordanbrown0, 2024-01-21:
A hypothetical `center()` module could be like `resize()` and render its children - at a performance cost.  In addition to the performance cost, the obvious implementation would lose color, transparency, et cetera.

So it's not impossible to implement, but it wouldn't be pretty.

yahbluez, 2024-01-21:
I do not care about CPU time a function needs that implements a needed functionality.
The prerelease versions are so much faster than the old stable.
If i look how much cpu time is needed to do fancy stuff with imported STL files a center() will not add some real time to that.

I wonder if their is a way to motivate the openscad developers to add some useful things to openscad like the center() or an export() that could be called from inside an openscad script.

Using bash/python wrappers for that increases the workload a lot.

gsohler, 2024-01-21:
most primitive functions of openscad have somehow a center option,
but especially import() has not.
it would actually be very easy to have a center option in import(), and i
believe this would satisfy yahbluez needs ....

jordanbrown0, 2024-01-21:
Let's not discuss it here because it's off-topic for this issue, but `export()` per se is very unlikely to happen for security reasons.  If you'd like to discuss it, please take it to the mailing list or e-mail me privately at openscad(at)jordan(dot)maileater(dot)net.

wolfwood, 2025-03-11:
the center parameter to import is now working for all (except nef3) geometry file types.

I do agree that it is common to want to import 3D geometry that is sitting at 0 on the Z axis and wish it to stay that way. 

it is easier to "guess and check" the bottom of the object than the center, so I think the current all-axis centering is the right way if we can only have one thing.

However, I can see 3 ways to address this:
1. an additional `keep_z` parameter that modifies `center=true` for 3D objects and is ignored for 2D
2. a `center_xy` parameter that operates independently, and is identical to `center` for 2D objects
3. an `auto_z` parameter that will center Z unless the bounding box lower z coordinate is within epsilon of 0. this could potentially be on by default as I think the behavior is likely to match the user's desires. `auto_z` could be a boolean, or it could take the actual epsilon value with `undef` being used to disable the functionality.

I like 3, but I don't know if it is too magical. curious what @t-paul and @kintel would find acceptable.

t-paul, 2025-03-11:
Just `center()` seems much too specialized and badly defined for anything other than mostly simple and regular meshes.

As @wolfwood mentioned #5671 does already covers what is asked here by adding the `center` option to `import()`.

Auto-Z is already done in slicers, I'm not sure we need such a specialized module. OpenSCAD as tool for modifying random imported meshes is missing much more features than just that. I think the focus should be on making the existing workflow easier, getting some of the many long time pending ideas into the hands of users.

wolfwood, 2025-03-11:
I understand that the release is your top priority. 

I just had the idea that we could automatically detect when not to center Z, and realized that if centering releases as is, then changing the default centering behavior later would be a breaking change. 

I like the idea well enough to revert #5671 until after the next release if needed. But if you think that center should always center on all axes, and xy centering only should be an explicit request, then the point is moot.

yahbluez, 2025-03-11:
That are good news!

My request was based on the use of STL files.
Because i needed the center feature so much 
i already wrote a tool that reads the STL, 
centers it 
and saved the new file with a 
file name extension that 
includes the size of the bounding box.

If the module center() is already written, just add a function center() that returns the size of the bounding box.
Knowing the size of the bounding box of a centered objects  gives all control to the user script.
And the work for this function is already done because to center the import, the dimension of the bounding box are needed. 

I really like the concept of BOSL2 to have same named functions that give a lot of data about an object.

Thank you for implementing center()

jordanbrown0, 2025-03-12:
We have a number of operations that have `center` options, and there have been some proposals to make them able to center independently on each axis, or positive/negative justify on each axis.  If we do something like that, it should work the same for all operations.

One thing that STL/DXF/SVG/et cetera might want that I don't think has come up before is "leave the X axis the same as it was in the imported file" (for any or all of the three axes, of course).  We should include that concept.

yahbluez, 2025-03-14:
Having control how center works on each axis would be the best solution.
But again, without access to the bounding box vector or size of the bounding box and a way to center,  it still need external tools to get this necessary information from imported files. 

openscad internally knows every bounding box about every object. In my opinion it makes no sense at all to hide this information. The question is how to show this data? Like BOLS2 with same named functions or universal with a query function? The way languages like python handle attributes is very cool and stable against any future extension.

like cube().bounding_box()   or bounding_box(cube()).

jordanbrown0, 2025-03-14:
OpenSCAD internally knows the bounding box for every **primitive** object (including imported objects).  It does **not** know the bounding box for differences, intersections, rotations, and probably other operations I'm forgetting.  To get the bounding box for the results of those operations you have to actually calculate where all the surfaces are, and the previewer doesn't do that.  Full rendering does know, but takes longer.

In addition, the structure of the language makes retrieving even what data is available inconvenient; modules generate geometry, but modules cannot return values.

PR#4478 includes constructs that let you force rendering and extract all of this interesting data, but is waiting on review.

yahbluez, 2025-03-15:
Yah, things are sometimes not that easy. The difference between preview and rendering often fades out with actual versions. Rendering is always a huge CPU load but getting data for the cost of CPU time is maybe better than no data. possibly risk is the generation of circle references that end up in endless loops. Like we know from the solvers of other CAD programs.