openscad/openscad #4951: Timestamp on console messages

SteveLelievre, 2024-01-26:
**Is your feature request related to a problem? Please describe.**
Some of my models with minkowskis take a very long time to render (and sometimes even to preview), and I have a tendency to forget when I started it running. So when I come back to check my computer I have no way to estimate how long I still need to wait. If only I knew when I'd started the task running, i would be able to estimate how long it has to run - based on how long it took last time.

**Describe the solution you'd like**
I'd like every output to the console to start with a timestamp - but not a full timestamp, just hh:mm (24-hour format). Then I could see when I'd started tasks and that would help be guess how long is left (given that I know from previous runs how long it's going to take in total.)

Alternatively, but less desirable to me, a command that I choose to can add to the start of my more complex models, such as "timestamp", that does nothing but write a timestamp on the console. Or, possibly, some kind of a $now variable that is continuously updated so that I can add an echo($now).

**Describe alternatives you've considered**
None, except occasionally remembering to make a written note.

**Additional context**
Add any other context or screenshots about the feature request here.

jordanbrown0, 2024-02-05:
A few thoughts:
* Should timestamps appear (optionally) in CLI output?
* Should timestamps be full date-and-time, or should they be narrower hh:mm or hh:mm:ss, or selectable?
* Maybe instead of timestamping, show a timer somewhere saying how long it's been running.
* Could just emit the time when starting execution.  That would be trivial to implement and seems uncontroversial.
* Could always prefix every message (or every line) with a timestamp.
* Could track the time for each message and prefix it *when displaying*, so that you could turn timestamps on or off after the fact.
* Having date/time functions seems generally useful, e.g. for timestamping models.  Strings, arrays, or either?

SteveLelievre, 2025-02-05:
Now that I think about it a bit more, the jordanbrown suggestion of just emitting the time at start of a preview or a render would satisfy my needs.  Ideally as a 24-clock type value so something like "Exectution started at hh:mm:ss"

closed 2025-03-09 by jordanbrown0