openscad/openscad #4952: Port to iPadOS / iOS

dwt, 2024-01-28:
**Is your feature request related to a problem? Please describe.**
I would like to preview and work on OpenSCAD files on the go. Working with a keyboard can actually be quite comfortable on iPadOS and iOS, thus I would like to try it for an ultra light computing device.

**Describe the solution you'd like**
An OpenSCAD renderer, that either includes its own text editor, or uses directory or file sharing, to use the text editor of another application would be ideal. Especially if an external editor is used, I would hope that the port could be relatively simple.

**Describe alternatives you've considered**
I've tried installing it in ish, an x86 emulator that runs Linux, but was not able to get that environment stable. Using it in a-shell could be nice, but is probably too niche to bundle it there.

**Additional context**
None

t-paul, 2024-01-28:
This is impossible with current AppStore rules. It remains to be seen if the latest actions in EU have a chance of changing that.

Scopeuk, 2024-01-28:
an alternative may be one of the webasm ports, I believe safari should support running these implementations. one example is https://ochafik.com/openscad2

dwt, 2024-01-30:
Oh sad, I was under the impression that those problems have long been resolved. Ah damn. That leaves distribution through tools like the [alt store](https://altstore.io), which is way less popular, but free of restrictions.

t-paul, 2024-01-30:
I have not heard about any issues being resolved, but maybe I missed that. I'll have another look. It's probably still a big project limited by Apple restricting development to Apple hardware, so I'm out. So the web version seems to have a much higher chance of becoming a viable option.

kintel, 2024-02-05:
Closing as infeasible. A web version is likely the best alternative for mobile usage.

closed 2024-02-05 by kintel

jordanbrown0, 2024-02-05:
https://www.fsf.org/blogs/licensing/more-about-the-app-store-gpl-enforcement
seems like an authoritative statement of why GPL-licensed software can't be distributed through the Apple app store.

jesseditson, 2025-04-06:
Hey! I'm an iOS developer, and I'm curious if anyone has tried building this to a local device. I don't need an App Store version, and due to lack of Apple Silicon support on the current release I build OpenSCAD from source anyway. I'll give it a try, but curious if there's any prior art I should be aware of. I have not tried building a Qt project on iOS before.

t-paul, 2025-04-06:
The dev snapshot has full support for Apple Silicon. I can't comment on the Qt topic.

kintel, 2025-04-06:
Qt might be a challenge, but we also require desktop OpenGL which isn't available on iOS. The OpenGL issue will eventually be resolved, but not quite yet..

eduo, 2026-02-15:
> Helpful, thank you!

Hello! Did you ever get further on this?

I'm in the same boat. The App Store is a red herring since openscad for Mac isn't in the Mac App Store either. Many who would use openscad wouldn't have a problem compiling locally and/or sideloading.

I know I wouldn't. Qt would be less of a problem than OpenGL if porting, though.