openscad/openscad #4956: OpenSCAD RPC

tpimh, 2024-01-31:
**Is your feature request related to a problem? Please describe.**
Currently, the workflow with OpenSCAD works in a way that the model view, editor, console, error log, customizer, animate and viewport-control all fit in the same window, and are only meant to be interacted directly by the user.

**Describe the solution you'd like**
It would be nice to keep the model view, but use some other tools externally. This can be implemented by some sort of RPC. For example, an external process is sending a command "preview" with a parameter of the entire content of `.scad` file, and the main process would update the model view accordingly (or return an error in case of a malformed parameter). In this example the external process might be VSCode with an extension, and the main process is OpenSCAD with RPC enabled. This can also allow for 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

gsohler, 2024-01-31:
Hi Dima, this is yet already possible.
Just make the editor pane invisible and edit your SCAD file in another
editor.
Once you save your model in the other editor, OpenSCAD Preview will
automatically update.

tpimh, 2024-01-31:
Thanks for a quick response.

Well, yes this is one way to do it through a file. In OpenSCAD it's possible to trigger preview without saving the file by pressing F5. And other scenarios (console, customizer, viewport, etc) are not controllable from a file as far as I know. It can be implemented the same way, but for something that needs to be updated in real time like viewport-control using a file is not as efficient.

t-paul, 2024-01-31:
OpenSCAD has a DBus interface which allows for some additional interaction with a running instance. This is still flagged as experimental, so it's not enabled in release versions.
I don't think that will ever allow customizer UI interaction, that seems out of scope for an external interface. There are other means of asking OpenSCAD to report customizer values. An external interface that works on data level might be an option.

tpimh, 2024-01-31:
Thanks, I was not aware of DBus interface, that is a perfect backbone for what I had in mind.

Looking at the implementation now, I understand that it's Linux specific (will probably also work on *BSD if the dependencies are satisfied). That's fine with me as I am on Linux, and with certain hacks it should be possible to make it work on macOS and even Windows, but I am pretty sure these platforms have their own specifications for RPC.

gsohler, 2024-01-31:
> For a general interface, TCP seems easier.

Is there a list of "commands" which are desired ?
I have noticed this morning, that the D-BUS already implements quite some
view port control features.

tpimh, 2024-01-31:
No, I don't yet have a list of desired commands. I believe DBus interface covers my needs pretty well, but I haven't really delved into it yet.

Hentra, 2024-11-11:
I would love to have the features of the dbus interface available in a more generic approach for all platforms. This would allow a deep integration in existing editors using plugins. Are there currently any plans to build something of that sort?