openscad/openscad #4967: OpenSCAD renders too slowly in preview mode

coder2999, 2024-02-05:
**Environment**
System: Windows 10 LTSC 2021 (not VM)
CPU: Intel Core i9 13900K
GPU: Nvidia GeForce GTX 4060
RAM: 64GB

**Issue**
OpenSCAD renders too slowly in preview mode, when the model is complex, there will be obvious lag when dragging and scaling the model. By using the task manager, I found that OpenSCAD only uses two CPU cores and 50% of the GPU computing power.

pca006132, 2024-02-05:
Are you using stable or latest development snapshot? VBO should improve performance considerably. Also, how complex is your model? Preview performs CSG in the GPU, so it will be slow if the model is too complicated and requires doing an actual render.

coder2999, 2024-02-05:
> Are you using stable or latest development snapshot? VBO should improve performance considerably. Also, how complex is your model? Preview performs CSG in the GPU, so it will be slow if the model is too complicated and requires doing an actual render.

I tried both, I feel that the latest DEV version is a little faster than stable version.

I copied a code of "bolts and nuts" from Thingiverse and used it in my project, this piece of code makes the rendering speed so slow, maybe the code quality is too poor.

And what is VBO?

pca006132, 2024-02-05:
VBO is an experimental option that you have to enable explicitly, it improves FPS for preview. You can also try features such as manifold which improves render time.

coder2999, 2024-02-05:
> VBO 是一个实验性选项，您必须显式启用，它改进了预览的 FPS。您还可以尝试诸如流形之类的功能，以缩短渲染时间。

[image]

Could you tell me how to enable VBO?

pca006132, 2024-02-05:
Check the checkbox of `vertex-object-renderers`, and maybe you want to check `manifold` as well.

coder2999, 2024-02-06:
> Check the checkbox of `vertex-object-renderers`, and maybe you want to check `manifold` as well.

Danke, es klappt.

closed 2024-02-06 by coder2999

kintel, 2024-02-06:
Also note that this will become the default option very soon!