openscad/openscad #4975: A way to reset the current transform

lijon, 2024-02-06:
**Is your feature request related to a problem? Please describe.**
I'm working on a library where I'd like to run a list of children two times, and one of the times override the transform for some children. The use case is to have two modes, one assembled view and one for 3D printing where parts are layed out flat.

**Describe the solution you'd like**
I'd like to have a `reset_transform()` module that undoes any transforms.

```
translate([10,20,30]) rotate([0,180,90]) reset_transform() cube(10, centered=true);
```

The above example would draw the cube centered at the origin.

pca006132, 2024-02-08:
You can use `$variable` to do that, e.g.:

```scad
$NOTRANSFORM = 0;

module foo() {
  if ($NOTRANSFORM) {
    cube(10, center=true);
  } else {
    translate([10, 0, 0]) foo($NOTRANSFORM=1);
  }
}

foo();
```

jordanbrown0, 2024-02-08:
I'm having a hard time understanding a use case that wouldn't be better served by having two different transforms, one for the "assembled" view and one for the "print" view.  Could you give a larger example, like with two parts?

(I've done this a couple of ways in my projects.  My favorite, mostly because it's cool rather than because it's practical, is wrapper functions for translate and rotate that take two parameters and animate between the two, so that with animation running the model assembles and disassembles itself.  A more practical variation is to use `$preview`, and produce the assembled variant when `$preview` is true or the print version when it is false, so that I'm usually looking at the assembled view and it automatically switches to the print view when I render for export.)

jordanbrown0, 2024-02-08:
Understood, thanks.

lijon, 2024-02-08:
> The problem is that how would you define this reset transform thing? Should it reset the transform for children of union/intersection/subtraction/minkowski sum? What about implicit union? Does it have to depend on lazy union, which alters the behavior of existing code?

Hm, my thought was that `reset_transform()` would simply set the current node matrix to identity, and would thus affect all its children only. So, for example, this would translate cube D with A, and cube C with B:

```
translate(A) {
  reset_transform() translate(B) cube(C); // translated by B only, because A is "undone" by the reset
  cube(D); // translated by A as usual
}
```

But I haven't delved deep into the openSCAD source code so I might be misunderstanding something.

> I think maybe you should open an issue in BOSL2 issue tracker as well, this may be something that they are interested in implementing, and does not require modifying openscad to do that.

I'll consider that! It would be quite a huge change in BOSL2, modyfing all the various transform wrappers (both modules and functions) in there to conditionally apply a transform or not.

jordanbrown0, 2024-02-08:
That is, it would discard the transformation matrix applied by the node's ancestors.

jordanbrown0, 2024-02-08:
In a slightly deeper look, it walks the tree top-down, visiting each node before and after visiting the children.  For a transform node, the "after" visit unions the children and applies this node's transformation to them.  (Net, a lot like a bottom-up walk.)  Removing the transformations applied by parents would require somehow blocking them, not merely resetting a top-down transformation matrix.

https://github.com/openscad/openscad/blob/6d505e402691f349a195649cff28eb73f49e7eb0/src/geometry/GeometryEvaluator.cc#L719

lijon, 2024-02-08:
Oh, that's probably right. Transforms are applied bottom-up: `translate(A) rotate(B) shape()` first rotates, then translates, of course. So indeed it would not be as simple as resetting the current matrix, but as you say one would need to set a flag that would block any further transforms up the tree.

jordanbrown0, 2024-02-08:
You can do it in a library, and I have.  The library has to implement all of the transforms, and you have it pass down the accumulated transform matrix through a $ variable.  You invert the matrix when you want to reset back to the default coordinate system.  Or maintain an inverted matrix as you work your way down.

It's not a very viable solution to your problem, but it was an interesting experiment.  [Here's a message and an animation.](https://lists.openscad.org/empathy/thread/IUPFIPEISP2XYU3SRVFWG54TLGSXMWOT?hash=PNPDGDEH7NFGIMUBHUF47KMY74LSP3PQ#PNPDGDEH7NFGIMUBHUF47KMY74LSP3PQ)  Click on the picture at the end of the message to run the animation.  Unfortunately, the source formatting didn't survive the message archive.

You can implement transformations either way, top-down or bottom-up.  The order of the transformations is controlled by which way you do the matrix multiply.  OpenSCAD seems to do them bottom-up.  (Top-down seems like it might be more efficient.  In a top-down scheme, each transformation has to do a 4x4 matrix multiply, and then at the bottom the primitive objects have to multiply their coordinates by the accumulated matrix:  (number of transforms)*(4x4 multiply) + (number of points)*(4x1 multiply).  In a bottom-up scheme, each transform needs to multiply all of the points in its children by its matrix:  (number of transforms)*(number of points)*(4x1 multiply).)

jordanbrown0, 2024-02-08:
Markdown ate multiplications in those expressions.  Here's the fixed text, updated in place for the record.

You can implement transformations either way, top-down or bottom-up.  The order of the transformations is controlled by which way you do the matrix multiply.  OpenSCAD seems to do them bottom-up.  (Top-down seems like it might be more efficient.  In a top-down scheme, each transformation has to do a 4x4 matrix multiply, and then at the bottom the primitive objects have to multiply their coordinates by the accumulated matrix:  `(number of transforms)*(4x4 multiply) + (number of points)*(4x1 multiply)`.  In a bottom-up scheme, each transform needs to multiply all of the points in its children by its matrix:  `(number of transforms)*(number of points)*(4x1 multiply)`.)

pca006132, 2024-02-08:
I think both fast-csg and manifold performs the top-down scheme, or in other words *tree flattening* by collapsing transform nodes together.

jordanbrown0, 2024-02-08:
And that's one of the problems, that different implementations could do it differently.

pca006132, 2024-02-08:
I don't think it is much of a problem, this should not affect the final results? There will be small differences due to floating point issues but it should be fine. This is just a performance optimization IMO.

jordanbrown0, 2024-02-08:
It's a problem with trying to implement a node that resets the transformation for its children.  Such a node is reasonably straightforward in a top-down scheme, but difficult in a bottom-up scheme.

nophead, 2024-02-08:
All my modelling is done with objects in the correct place to make an assembly view, with fasteners added, etc. Since each STL is its own module without transforms and oriented for printing it is trivial to lay out several to form a build plate. I often use $preview to show the assembly view when for F5 and show the STLs laid out for printing with F6.  I don't see any need for reversing the transforms. I just call the STL module from two places, one with a transform and the other with a simple XY translation. E.g.

[image]

[image]

lijon, 2024-02-08:
> All my modelling is done with objects in the correct place to make an assembly view, with fasteners added, etc. Since each STL is its own module without transforms and oriented for printing it is [...]

Sure, that would work! But it would need the user (of my library) to define each aux part as a module, and call them at two places. What I'd like to do is to extract them from the assembled view. The extraction itself is simple, by using conditional generation depending on a $var, or using BOSL2 tags. But the problem (and reason for this feature request) is the undoing of the transforms applied to the part in the assembled view.

> You can do it in a library, and I have. The library has to implement all of the transforms, and you have it pass down the accumulated transform matrix through a $ variable. You invert the matrix [...]

Oh, yes that's indeed one solution! In my case it would need to be implemented in BOSL2 by rewriting all their transform modules.

BrownianHub, 2024-02-09:
Hi there, I recently created a little framework that might be of use here: 
[custom-transformations](https://github.com/BrownianHub/custom-transformations)

If you look at the examples.scad file in there, one possibility is to define a set of custom transformations in an array - this will be for assembling the full module. 
Then map the child index to the index of each of the transformations in a module, such that each child gets its own transformation.

Then for laying out all the parts, you could supply say either an empty transformation array or a variable (depends how you write the module) that will check to see if set, and instead display the child itself. OR you could also supply one specific different transformation (say displace on x-axis some distance) for all children inside the array and have all children do that instead.

lijon, 2024-02-10:
Actually this was quite easy to solve, no rewrite needed, I just include this file: https://github.com/lijon/jl_scad/blob/main/reset_transform.scad
(requires BOSL2).

I'll leave this issue open in case this is something to be considered for the built-ins in openscad?

BrownianHub, 2024-02-10:
> Actually this was quite easy to solve, no rewrite needed, I just include this file: https://github.com/lijon/jl_scad/blob/main/reset_transform.scad (requires BOSL2).
> 
> I'll leave this issue open in case this is something to be considered for the built-ins in openscad?

I should clarify, I am relatively new to OpenScad (been using it for about a few months total now). So, I only now just looked into BOSL2 (and some of the previous comments of this thread), and I just realized that basically what I came up with in my framework is basically what BOSL2 implemented under the hood, and they did it better ... and did it several years ago, haha.   
Welp, that's several weeks wasted. 😮‍💨 

But yeah, that's a good solution for your use case, and it should work.   

It makes me wonder though if it would make sense to have built-ins (as you may have alluded to) for retrieving the transformation matrix of already existing transforms?  So, instead of multiple frameworks potentially redefining the mult-matrices for `translate`, `rotate`,`scale`,etc, -- have them be built-in, and then allow users to reference them to do things like this as well as things like supplying custom-transformation functions as parameters to modules so that they can apply various transforms on their children().  But, that might be better served as a separate issue...