openscad/openscad #4976: Reboot attempt at working MSVC build

roel-v, 2024-02-07:
This builds on work earlier done in:

https://github.com/openscad/openscad/issues/2814 (2019)
https://github.com/openscad/openscad/issues/2857 (2019)
https://github.com/openscad/openscad/pull/4126 (2022)

I have a fork with a working msvc/vcpkg build here: https://github.com/roel-v/openscad/tree/vcpkg-build .

Instructions for trying this out:

- Install Visual Studio 2022. Check the 'Desktop development with C++' component in the VS installer.
- Install vcpkg in a convenient location with a short path (this is important), e.g. c:\vcpkg or d:\vcpkg. Instructions are at https://vcpkg.io/en/getting-started but an even shorter version is

cd d:\vcpkg
git clone https://github.com/Microsoft/vcpkg.git
.\vcpkg\bootstrap-vcpkg.bat

And then add d:\vckg to your %PATH% .

- If you didn't install vcpkg in d:\vcpkg, edit the build_vcpkg.bat batch file to update the path to vcpkg.
- Clone the repo above, run build_vcpkg.bat in it.

What that batch file does is first install all required packages through vcpkg, then it generates Visual Studio project files in the 'build' directory and finally builds Release and Debug versions.

For now, this works for a most basic definition of 'works'. Meaning, it generates an executable that starts up but that's about it. Most important issue is that it links against both Release and Debug executables in the Debug build.

TODO:

- Fix build configs (I think only in the local FindQt5Scintilla.cmake module) so that the Debug build correctly uses Debug versions of all transitive dependencies
- Check the changes to CMakeLists.txt still work OK on other platforms
- Fix build_vcpkg.bat so it doesn't use any absolute paths
- Set up CI for this build so it becomes clear if/when things stop working
- Check versions of all transitive dependencies. The commit history in the fork has details, but essentially all versions of dependencies are locked to one specific version. This means that either they are already out of date, or won't be updated when a dependency gets a new version in vcpkg. After moving to Qt6 this should become easier as it then will probably be possible to use a much simpler vcpkg setup, until that time a tool that checks versions and helps with manually upgrading version numbers would be useful (not hard to write either, just has to be done).
- Go through the earlier issues about this topic, see which ones of the todo items there are still relevant.

gsohler, 2024-02-07:
This sounds great!

Will this build allow for debug in a native debugger ? (step through the
code ,set breakpoints and watch crashes ?)

roel-v, 2024-02-07:
Yes - in fact this works in the commit I just did seconds ago.

kintel, 2024-02-07:
Thanks @roel-v - I added checkboxes to the TODO list. Let's check them off as items are resolved.

pca006132, 2024-02-18:
Curious how long does it take to build this? Also, from my experience, you probably don't want building dependencies as DLLs, but build them as .lib files that can be linked statically. I think it also allows you to step into the dependencies this way, and avoids the need to manage DLL locations.

pca006132, 2024-02-19:
Wow this is pretty fast, I was expecting something like half a day when you have that many dependencies. Sounds pretty good.

It is weird that static linking can cause those issues, I think in general they should not, but I am not experienced in Windows to say much about it. I think the current mxe build uses static linking, the linking time is reasonable (not fast but not too slow either), and there are other linkers (probably not for msvc though) that can link faster than GNU's LD.

For the issue with DLL location, I was just fixing issue with DLL location in manifold build, because we did not put everything in one directory :)
But yeah it is fine if you know the platform.

kintel, 2024-02-19:
fwiw, we use dynamic libs on macOS as well. It's all about knowing how to package it well to avoid surprises :)