openscad/openscad #4977: Ability to set solid name

DeepHorizons, 2024-02-07:
Hello,

I am using openscad to generate STLs for use in OpenFOAM. I am using the name of the solid to specify which part is what in the simulation. Unfortunately it seems I can only get openscad to generate `solid OpenSCAD_Model` in the STL. My workaround is to generate the STL as an ASCII file and modify the file afterwards. It would be nice if I could specify in the `.scad` file what the solid name should be so I can export it in binary format or not have to manually edit the file. It would also be extremely nice if I could specify multiple solids in the same STL. That way I can dynamically specify regions and bounding boxes for my simulation.

Example syntax would be something like:
```
// Specify a solid name if no region is specified, or is outside a region
solid_name "MyObject"

// Specify a new solid with the given name
solid_name "Region1" {
    my_module()
}
```

jordanbrown0, 2024-02-08:
Generating multiple "parts" from one file is covered under #1608.

Let's leave this one open for a request to have a way to control the "solid" value in STLs and perhaps more generally to supply metadata for exports.

pca006132, 2024-02-12:
Not sure if this is something you want, but you can use manifold c++/js/python binding to do this. There is no direct support for solid name, but you can manipulate the objects as values and export them one by one if needed.

DeepHorizons, 2024-02-14:
I took a stab at implementing this and made non-zero progress. You can see what I have here #4986
I like the idea of a special variable name `$solid_name` that can be set in the `.scad` file. However I am unsure where to begin looking to add such a variable and then extract that out in the export process. Could someone point me in the right direction?