openscad/openscad #4978: No holes in the importing SVG

Broshward, 2024-02-08:
**Describe the bug**
I using openscad at one of the stages of prototyping PCB. I'm importing an SVG image of PCB saved in Inkscape.
I see that there are no holes in the board. Is this a bug?
I see that the begin of Y-coordinate in other edge of the sheet. Is this a bug?

**To Reproduce**
Steps to reproduce the behavior:
1. Importing SVG
2. Preview (F5)
3. Ctrl+4
4. Ctrl+Shift+V 

**Expected behavior**
I expects the boards to be at the origin and with holes

**Screenshots**
[image]
[image]
[image]

**Environment and Version info (please complete the following information):**
 - OS: ArchLinuxARM
 - System: Khadas VIM3 Pro (aarch64)
 - OpenSCAD version 2021.01

**Library & Graphics card information**
alex@VIM3 ~ $ openscad --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2021.01
System information: Linux 5.19.4-ARCH #1 SMP Sat Aug 27 11:02:57 MSK 2022 aarch64 Arch Linux ARM 6 CPUs 3.70 GB RAM
User Agent: OpenSCAD/2021.01 (Linux aarch64; Arch Linux ARM)
Compiler: GCC "12.1.0"
MinGW build: No
Debug build: Yes
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Qt version: 5.15.12
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.76.5
lodepng version: 20180910
libzip version: 1.10.1
fontconfig version: 2.14.2 (runtime: 2.15.0)
freetype version: 2.13.2
harfbuzz version: 8.1.1 (runtime: 8.3.0)
cairo version: 1.17.8 (runtime: 1.18.0)
lib3mf version: 1.8.1
Application Path: /usr/bin
Documents Path: /home/alex/.local/share
User Documents Path: /home/alex
Resource Path: /usr/share/openscad
User Library Path: /home/alex/.local/share/OpenSCAD/libraries
User Config Path: /home/alex/.config/OpenSCAD
Backup Path: /home/alex/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/alex/.local/share/OpenSCAD/libraries
  /usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /home/alex/.local/share/fonts
  /home/alex/.local/share/flatpak/exports/share/fonts
  /var/lib/flatpak/exports/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /home/alex/.fonts
  /usr/share/fonts/100dpi
  /usr/share/fonts/75dpi
  /usr/share/fonts/OTF
  /usr/share/fonts/TTF
  /usr/share/fonts/Type1
  /usr/share/fonts/adobe-source-code-pro
  /usr/share/fonts/cantarell
  /usr/share/fonts/cyrillic
  /usr/share/fonts/encodings
  /usr/share/fonts/gnu-free
  /usr/share/fonts/gsfonts
  /usr/share/fonts/liberation
  /usr/share/fonts/misc
  /usr/share/fonts/noto
  /usr/share/fonts/noto-cjk
  /usr/share/fonts/ttf-linux-libertine
  /usr/share/fonts/ubuntu
  /usr/share/fonts/util
  /usr/share/fonts/encodings/large

MESA: error: ZINK: vkCreateInstance failed (VK_ERROR_INCOMPATIBLE_DRIVER)
glx: failed to create drisw screen
failed to load driver: zink
GLEW version: 2.2.0
OpenGL Version: 4.5 (Compatibility Profile) Mesa 23.3.5-arch1.1
GL Renderer: llvmpipe (LLVM 16.0.6, 128 bits)
GL Vendor: Mesa
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: GLX
PNG generator: lodepng
GLX version: 1.4
OS info: Linux 5.19.4-ARCH #1 SMP Sat Aug 27 11:02:57 MSK 2022
Machine: aarch64

**Additional context**
[image]

t-paul, 2024-02-08:
Please add the SVG file as attachment (e.g. adding ".txt"). When clicking on the inline SVG, I'm getting a totally different view, which probably already is the reason for the unexpected result.

Broshward, 2024-02-08:
[LED_flyback1.pcb_bottom.svg.txt](https://github.com/openscad/openscad/files/14203135/LED_flyback1.pcb_bottom.svg.txt)

kintel, 2024-02-08:
It's because the holes in your drawing aren't actually holes; they are filled white circles layered on top of the rest of the drawing.
To make OpenSCAD interpret them as holes, they have to be actual holes in SVG, typically done using https://www.w3.org/TR/SVG/painting.html#FillRuleProperty

We had a discussion a long time ago whether we should attempt guessing the intent of using white objects in SVG, but we didn't implement that, partially because it opens a whole can of worms..

closed 2024-02-08 by jordanbrown0