openscad/openscad #4981: crash when trying to export stl

DraconPern, 2024-02-11:
**Describe the bug**
Program exits and windows disappears.

**To Reproduce**
Steps to reproduce the behavior:
1. Install [OpenSCAD](https://www.openscad.org/) version 2024.02.04 (git f3de51827)
2. Open the files here https://www.printables.com/model/757297-gridfinity-rugged-storage-box-parametric-and-custo/files
3. Change Part to 'Bottom'
4. F6 to render
5. F7 to export stl
6. Program disappears. No warning, no error pop up. Main window just poofs.

**Expected behavior**
Not crash

**Code reproducing the issue**
cube(10); works.  
fast-csg, fast-csg-safer, manifold are turned on

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Windows 10
 - System: Intel PC 64-bit
 - OpenSCAD Version 2024.02.04 (git f3de51827)

**Library & Graphics card information**
OpenSCAD Version: 2024.02.04 (git f3de51827)
System information: Microsoft Windows 10 (10.0.19045) x86_64 20 CPUs 63.82 GB RAM
User Agent: OpenSCAD/2024.02.04 (git f3de51827) (Microsoft Windows 10 (10.0.19045) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20210627
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg*, fast-csg-safer*, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: C:/Program Files/Openscad
Documents Path: C:\Users\DraconPern\Documents
User Documents Path: C:\Users\DraconPern\Documents
Resource Path: C:/Program Files/Openscad
User Library Path: C:/Users/DraconPern/Documents/OpenSCAD/libraries
User Config Path: C:\Users\DraconPern\AppData\Local/OpenSCAD
Backup Path: C:/Users/DraconPern/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/DraconPern/Documents/OpenSCAD/libraries
C:/Program Files/Openscad\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/WINDOWS/fonts
C:/Users/DraconPern/AppData/Local/Microsoft/Windows/Fonts
C:/Users/DraconPern/.local/share/fonts
C:/usr/local/share/fonts
C:/usr/share/fonts
C:/usr/X11/lib/X11/fonts
[... 397 lines cut ...]
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

**Additional context**
Add any other context about the problem here.

DraconPern, 2024-02-11:
I did a clean install to make sure there wasn't any left over from previous install.  Couldn't find how to reset user setting so I don't know if that affects anything.

kintel, 2024-02-11:
Could try try disabling all the fast-csg options and enable just `manifold` instead?
fast-csg and manifold are mutually exclusive, but we don't currently have a good way of surfacing that in the GUI.

MWStonton, 2024-02-11:
Can't add to detail (other than also on macOS, 14.3, M2) having seen the same thing once in the last couple of days with recent dev. snapshot. Didn't record details as it was fine after restarting OpenSCAD. Was probably switching between rendering methods as I use manifold for speed while iterating design but often have to use older rendering methods before exporting as stl for many designs to prevent my slicer (for 3D print) finding errors.

pca006132, 2024-02-12:
what is the error in your slicer when the design is rendered with manifold?

MWStonton, 2024-02-12:
@pca006132 Not sure at the moment - just says there's an error with options to correct, close or ignore. Not obvious to me (haven't spotted visually if I ignore) what it is at the moment - when I get a spare moment I'll have a play with the design and try to find out which sections cause the issue and then see if it is obvious if ignored in the slicer. (Slicer is Flashforge FlashPrint). The design uses offset, lots of text, difference and intersections.

pca006132, 2024-02-12:
Sure, you can open an issue when you find a reproducible example. I think it is a manifoldness issue.

MWStonton, 2024-02-12:
I was thinking the same (had it in my mind (thought I remembered a message along the lines of  "design (rendered using manifold) is not manifold" which made me smile - maybe an older version of the slicer did say more. I assumed manifold was still a work in progress and so just worked around it for now. If I can spend a little time this evening and at least narrow down which part of the design is the issue and it may help with development, then I'll post it as a new issue

pca006132, 2024-02-14:
@MWStonton Do note that this is usually issue on openscad side, e.g. some primitives did not guarantee manifoldness. There were some changes recently regarding that, so it may fix the issue and you will not be able to reproduce it with recent versions :)

MWStonton, 2024-02-14:
@pca006132 I had reduced the code down to what created the issue for me and it was a difference where the original and part to be removed should be identical on one face (which was a cylinder and removing text which was shaped using intersection to the same thing on the outer face). I tried it this morning with the same version of OpenSCAD and same code and it doesn't fail (though the original "full" code it came from does).
This is an image of the text after render showing a sliver of material left over through the vertical of the "T". I assume the error might relate to that?
[image]
I have done similar things with added proud text where I overlap with the cylinder a little and I believe if I make the text to remove slightly bigger than the cylinder for the cut in, it works o.k. I did try a radius bigger by 0.001 and it seemed to fix things but 0.000001 didn't fix it.

pca006132, 2024-02-14:
It is for both input and output, STL does not preserve manifoldness. Yeah you should use 3mf when possible. For now, openscad does not preserve manifoldness when using 3mf as *input*, but it should not matter in your case.

gsohler, 2024-02-14:
pca006132,

can you quickly explain:

why is manifoldness lost when using stl ? (is it two cubes touching at one
corner only and its not clear which solid the corner belongs to ) ?

why is 3mf better than stl in this respect ?

Thank you for answering curious questions :)

pca006132, 2024-02-14:
@gsohler 

> is it two cubes touching at one corner only and its not clear which solid the corner belongs to

Yes, this is an example. We care about how the vertices are connected, and collapsing vertices when the position of them are the same will lose this topology information, thus losing manifoldness.

> why is 3mf better than stl in this respect ?

STL only store triangles as coordinate triples, so you cannot distinguish two vertices if they have the same coordinate. However, 3MF stores vertex position and connection information separately, so the topology information is not lost.

grufkork, 2024-02-14:
Can confirm having the same issue, using Manifold. OBJ export works as expected, STL crashes, producing malformed file. [2024.02.11]

MWStonton, 2024-02-14:
re: stl export error
I've tried a couple of files just exporting and re-exporting and seen no error (overwriting the same stl file, or creating a new one) BUT if I edit the design scad file  (I use an external editor), let OpenSCAD pick up the changes and re-preview and then manually render and re-export then the crash is quite common. I believe it happens whether the stl file is a new one or overwritten.

gsohler, 2024-02-14:
Yes , two cubes touching in one coordinate is really a very
interesting scenario. You can think of two separate objects.
But at the same time, you could also vision one object with a micro
connection formed from 6 faces which form a ring and are perpendicular to
each other.

If you read that in double-cube stl and look at the special coordinate, you
will find, that it's used by 6 faces.
Running the faces-around-corner algorithm, you will easily find out, that
there are 2 chains, thus two objects . Easy task to duplicate the vertex
and share evenly between the objects.
This could easily done  in the end of import_stl without big time expense.
After that I don't see this disadvantage over 3mf anymore.

pca006132, 2024-02-14:
Yes it can be done, but it is not as simple and you can write a paper about it (Cutting and stitching: converting sets of polygons to manifold surfaces), and the implementation is not trivial as well, we are currently using CGAL's implementation and porting it takes a lot of time. Another issue is that slicers don't implement this, so you will get warnings in slicers.

Regarding disadvantage over 3mf, the only task for stl is to represent a single 3D object, and it cannot do it without the help of some mesh fixing algorithm, so I cannot see the advantage of stl, apart from the weird popularity...

gsohler, 2024-02-14:
I understand your argument that it takes a lot of time to port
CGALS's implementation.  CGAL has the same score in Versatility and
Obfuscation.
But i am wondering why you need an any cutting and sticking of faces when
doing vertex separation of  a manifold stl input ?(this means that no faces
would intersect others nor there any points below surface)(apart from
inside negative volumes)
When you first create a edge mapping database(which face index is opposite
of an edge (specified by 2 vertices),
then it's just an O(N) task to find out if two or more vertices share the
same coordinate. I have done that before.

Even though nobody here appears to like STL I can think of quite some
reasons for its weird popularity
* its an very old format
* very established, everybody knows it
* its very simple, can easily be read and written

Not as easy to create 3mf. it is zipped and contains (few) different files
which need to be in sync to each other.
And many people put several objects into one STL when they share a design
with many parts readily placed for printing.
Even though the STL spec does not specify that STL can contain several
objects, its easily doable, practical, works well
and probably many people will cry if this got forbidden 🤣

pca006132, 2024-02-14:
It is one thing that you can find out the vertex that requires duplication, figuring out how to split the vertex into two (which face got which) is another issue. This is the reason why we need to find a cut. If you think it is a simple thing to do, you are welcome to submit a PR to manifold for that...

And I am not here to argue about why STL is a bad format nor I want to make it obsolete in openscad, I just want to point out that if users get manifoldness issue when importing stl into a slicer, they can just use 3mf. Whether or not writing stl/3mf by hand is easy is not of my concern.

I think we should not continue this thread here, free to open another thread to discuss about manifoldness issues. I think we are distracting the discussion here.

nophead, 2024-02-14:
STL works fine for my uses and the slicer I use doesn't accept any more recent formats. It can't represent anything with self intersections but I don't count those as real object that can be printed. The main problem is OpenSCAD snaps vertices to a grid and that can cause self intersection and break STL export even if the solid has no self intersections and could be represented in S TL. The only time STL format cause a problem is if the vertices are separated by less that the resolution of a float but that is many orders  magnitude less that the OpenSCAD grid.

gsohler, 2024-02-14:
I believe the reason why openscad snaps vertices to a grid is to find out
quickly, if two imported coordinates are meant to be the same.
But there are better approaches.
We could code a map function like:

space[3d-coordinate ] = information

when the underlying data is stored as a binary tree(actually octant ) its
pretty easy to quickly find out, how many/much information is stores at a
coordinate+search sphere-radius
I got code for that, when interested.

jordanbrown0, 2024-02-14:
> The only time STL format cause a problem is if the vertices are separated by less that the resolution of a float but that is many orders magnitude less that the OpenSCAD grid.

Not "many".  The OpenSCAD grid is around 10^-6 or 10^-7 (I have seen somebody say it's at 2^-20).  A single-precision float, as used in STL, is only good for about 7 digits.  The OpenSCAD grid appears to be absolute, while the ~7 digits of 32-bit floating point are relative to the size of the value.  If you're working with values around 100, the grid might well be smaller than the resolution of a float.

OpenSCAD arithmetic is 64-bit float, but STL is 32-bit float.

---

Grid snap is not just about imports; it's visible even with normal primitives.  I am told that without it vertices that are supposed to be identical drift apart and cause problems.  That shouldn't apply to "simple" cases where two vertices start the same and are transformed the same, but it might be an issue for cases where two vertices are derived through different paths and mathematically *should* be the same.

jordanbrown0, 2024-02-14:
> Yeah so I guess this is a known issue :)

Remember that the original complaint is that the application crashed, poof, gone, not merely artifacts or a CGAL error.  That's a different class of problem.

pca006132, 2024-02-14:
@jordanbrown0 yeah I am replying to the issue @MWStonton mentioned, not to the original post. And I suggest not to clutter this thread more...

closed 2024-02-16 by kintel

kintel, 2024-02-16:
Thanks @pca006132 for spotting a (in retrospect) painfully obvious bug!

PR #4990 (2024-02-16): fix append_stl infinite recursion
Closes #4981.

This is a simple infinite recursion.