openscad/openscad #4982: Double variable assignment warning not displayed in some cases it should be

MethylBromide, 2024-02-11:
**Describe the bug**
The documentation states that OpenSCAD will give a warning if the same variable is assigned twice, e.g. _WARNING: wheel_radius was assigned on line 3 but was overwritten on line 17_. However, this warning does not occur in all cases.

**To Reproduce**
Steps to reproduce the behavior:
Execute the script given below in the UI, and observe the console output.
This is what I get:

> Parsing design (AST generation)...
> Saved backup file: C:/Users/tyler/OneDrive/Documents/OpenSCAD/backups/tut1-backup-VcHabsyn.scad
> Compiling design (CSG Tree generation)...
> Compiling design (CSG Products generation)...
> Geometries in cache: 6
> Geometry cache size in bytes: 46840
> CGAL Polyhedrons in cache: 0
> CGAL cache size in bytes: 0
> Compiling design (CSG Products normalization)...
> Normalized tree has 2 elements!
> Compile and preview finished.
> Total rendering time: 0:00:00.165

**Expected behavior**
Expecting the console to display a warning message that wheel_rad was multiply assigned.

**Code reproducing the issue**
```
wheel_rad = 6;
scale([1.3,1,1]) {
translate([-20,15,0])
    rotate([90,0,15])
    cylinder(h=3,r=wheel_rad,center=true);
wheel_rad = 12;
translate([20,-15,0])
    rotate([90,0,0])
    cylinder(h=3,r=wheel_rad,center=true);
}
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: 64-bit Acer PC
 - OpenSCAD Version: 2024.02.10 (git 662e59003) from OpenSCAD website downloads page. Also observed issue with production release 2021.01

**Remarks**
I have seen this warning in other scripts. I don't know what the general rule is to reproduce the issue, but suspect it may have to do with nesting or indentation.

closed 2024-02-12 by pca006132

MethylBromide, 2024-02-13:
I have submitted a correction to the documentation to hopefully prevent such confusion in the future.

t-paul, 2024-02-13:
Thanks, where did you submit that? I would have assumed https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/General#Scope_of_variables but I don't see recent changes.

MethylBromide, 2024-02-15:
Here is the corrected document: https://en.wikibooks.org/wiki/OpenSCAD_Tutorial/Chapter_2#Parameterizing_parts_of_your_model
I think people start with the tutorial more than they do the manual. Having introduced them to variables and curly braces, it seems only fair to warn them why the variable they assigned at some point might not work anymore later in the script.