openscad/openscad #4988: search function returns incorrect value

rjcarlson49, 2024-02-15:
**Describe the bug**
When searching a vector of vectors for a vector, search returns [[]] instead of [] when no match is found.

**Code reproducing the issue**
FRONT = [0, 1, 0];
BACK = [0, -1, 0];
RIGHT = [1, 0, 0];
BOITTOM = [0, 0, -1];
b = [FRONT, BACK, RIGHT+BOTTOM];

echo(str([FRONT, BACK, RIGHT+BOTTOM]));
echo(str("FRONT ", FRONT, " ==> ", search([FRONT], [FRONT, BACK, RIGHT+BOTTOM])));
echo(str("RIGHT ", RIGHT, " ==> ", search([RIGHT], [FRONT, BACK, RIGHT+BOTTOM])));
echo(str("R+B ", RIGHT+BOTTOM, " ==> ", search([RIGHT + BOTTOM], [FRONT, BACK, RIGHT+BOTTOM])));

**Output**
Output of this code is 
ECHO: "[[0, -1, 0], [0, 1, 0], [1, 0, -1]]"
ECHO: "FRONT [0, -1, 0] ==> [0]"
ECHO: "RIGHT [1, 0, 0] ==> [[]]"     <==== Expected []
ECHO: "R+B [1, 0, -1] ==> [2]"

**Environment and Version info (please complete the following information):**
 - OS: macOS Sonoma 14.2.1
 - System: iMac with M3
 - OpenSCAD Version 24.01.15

rjcarlson49, 2024-02-15:
There's a type, BOITTOM ==>> BOTTOM

jordanbrown0, 2024-02-15:
First, I should note that the behavior of `search()` is quite unobvious and the the documentation does not fully explain it.

For your particular case, because the first argument is a vector, you are asking it to search for any of several values.  There's only one value (itself a vector) in your vector, but it's still a vector.  The function returns a vector of results, and if the particular value is not found then it returns [] for that value.

Let's move away from vectors-of-vectors because they confuse the issue; let's just search for numbers.

Search for one entry, that's there, get one result:
```
echo(search([2], [1,2,3]));
ECHO: [1]
```

Search for two entries, that are both there, get two results:
```
echo(search([2,1], [1,2,3]));
ECHO: [1, 0]
```

Three entries, one of them not found.  Get three results, missing entry reported as `[]`:
```
echo(search([2,4,1], [1,2,3]));
ECHO: [1, [], 0]
```

Two entries, one of them not found.  Two results, missing entry reported as `[]`":
```
echo(search([2,4], [1,2,3]));
ECHO: [1, []]
```

Searching for one entry, not found.  One result, missing entry reported as `[]`:
```
echo(search([4], [1,2,3]));
ECHO: [[]]
```

Moving over to a slightly simplified variant of your example:
```
FRONT = [0, 1, 0];
BACK = [0, -1, 0];
RIGHT = [1, 0, 0];
BOTTOM = [0, 0, -1];

echo(str("[BACK,RIGHT,FRONT] ==> ", search([BACK,RIGHT,FRONT], [FRONT, BACK, RIGHT+BOTTOM])));
```
yields
```
ECHO: "[BACK,RIGHT,FRONT] ==> [1, [], 0]"
```

So that's why it returns the result you see:  you asked it to search for a vector of values, and it returned a vector of results, and the one entry in that vector of results is the empty vector.

---

`search()` has a number of ... interesting ... behaviors but unfortunately at this point there's no telling how people might have come to depend on them, and so it can't really be fixed.  Maybe one day somebody will design a replacement.

closed 2024-02-15 by jordanbrown0

jordanbrown0, 2024-02-15:
#4989 for a proposal for a replacement.