openscad/openscad #4989: Simpler replacement for search()

jordanbrown0, 2024-02-15:
`search()` is very fast, but is very difficult to understand and very easy to use incorrectly.  It seems like it would be good to have a replacement that is fast but simple, perhaps sacrificing some of `search()`'s power.

Here's a straw-man proposal:
* `find(pattern, list, [limit=n])`
* `pattern` is any non-function[*] value for which equality is defined[**], or a function.
* `list` is, well, a list.
* `limit` sets the maximum number of entries to return.  Default is to return one, as a scalar.  Zero returns all matches.
* if `pattern` is a function, it's called as `f(list[i])` and returns `true` if matched and `false` if not.
* By default (`limit` undefined) returns the index of the first entry found, or `undef` if not found.
* If `limit` is specified, returns a list of indexes of entries found, which might be an empty list.

[*] if you really must search for a function, you can put it and the entries in the list into lists, so that you are searching for a one-entry list of functions.
[**] which currently excludes objects.

Limitations compared to `search()`:
* No facility for searching for "any of these", other than using a function.  (And no easy way to add it later, because pattern being a list means to search for that particular list.)  Calling it repeatedly is approximately the same performance.
* No facility for searching based on a column in a list-of-lists, other than using a function.  (Could be added later as a `column=n` argument, which could generalize to search lists-of-objects.)  Could also use list comprehension to generate a key-only list, and then `find()` on that.

Notes:
* OpenSCAD's treatment of strings as arrays of characters means that `pattern` could be a single character and `list` could be a single string, and you'd be looking for that one character in that string.  Perhaps this should be explicitly defeated, since it's not very useful.
* Does not attempt to do substring matching.
* Function execution is **very** slow compared to built-in comparison.  List comprehension solutions are probably similar in performance to a function-based match.
* It isn't clear that there are really use cases for values for `limit` other than undefined (first match, as a scalar) and zero (all matches, as a list).  That suggests that perhaps it should be a boolean `all` instead, or perhaps there should be `find()` and `findall()`.

Examples:
* `find(1, [3,1,4,1,5,9])` ==> `1`
* `find(1, [3,1,4,1,5,9], limit=0)` ==> `[1, 3]`
* `find([3,3], [ [1,1], [2,2], [3,3], [4,4] ])` ==> `2`
* `find("apple", [ "banana", "pear", orange", "apple", "plum" ])` ==> `3`
* `find("p", "apple")` ==> `1`
* `find(function (v) v > 5, [3,1,4,1,5,9])` ==> `5`
* `find(2, [3,1,4,1,5,9])` ==> `undef`
* `find(2, [3,1,4,1,5,9], limit=0)` ==> `[]`

jordanbrown0, 2024-02-15:
I should note that this proposal derives from a conversation that Adrian Mariano and I had in 2022.

rjcarlson49, 2024-02-16:
It would be much cleaner to always return a list. Then len(result) == 0 tests for not found, Len(result) > 0 tests for found. Len(result) == 1 tests for a unique match, with result[0] being the entry. Len(result) > 1 tests for multiple matches. Using undef as a legal return makes it necessary to use a compound predicate to test some of these conditions. If you always return a list, you can plug it straight into a for. An empty list will execute zero loops.

It would look even better if result.len could be used.

jordanbrown0, 2024-02-16:
You're not wrong, but "find the first entry" (which is often also the only matching entry) is a reasonably common case, and there the return-a-list scheme has problems:  on the average, it'll compare twice as many entries as it needs to, and the result is less obvious.

You only get undef if you asked for a single result, so it's never a three-way comparison.  There is no call that might return undef, a scalar, or a list. If the call asks for one entry, it'll either get undef or a scalar.  If it asks for more than one result, it'll get a list (which might be empty).

Remembering that, `is_undef(result)` seems comparable in complexity to `len(result)==0`.

- `rands()` has a somewhat similar behavior to what you propose:  it always returns a list of N random numbers.  A while back I was talking to somebody who was very confused by that.  They wanted one random number, and were getting back a list, and didn't know what to do.  They seemed to think that the `[0]` that they needed to stick on the end was a magic incantation.  In fact, they'd come up with a different magic incantation that worked:  `rands(...)*[1]`.  (It's left as an exercise for the reader why that works.)

Always returning a list means that a simple case - you only want a single result - becomes a bit more complicated:  you get a list.

Maybe this is an argument for having distinct `find()` and `findall()` functions, with `find()` returning a single index or `undef`, and `findall()` returning a possibly-empty list.

jordanbrown0, 2024-02-17:
Note that a userspace implementation of `findall()` is relatively simple:
```
function findall(v, list) = [ for (i=[0:1:len(list)-1]) if (list[i] == v) i ];
```
(Except that's limited to lists with fewer than a million entries.)
Based on experiments with `search()` and lists of numbers, the built-in form is about 20x faster than the list comprehension.

rjcarlson49, 2024-02-18:
I have a fairly strong view that undef should indicate a program error of some sort and not a "valid" return, even though you can test for it. If find1 returns a single scalar I would use -1 for a negative return, quite intuitive. For find_all, an empty list, [], seems perfect. The fact that the find1 and find_all use cases prefer different return types inclines me to have 2 different calls. But I would be OK with 1 call that always returned a list.

I stopped worrying about code execution time decades ago, at least for the code I write. I aim almost entirely for ease of understanding. If I come back to it in a month, I don't want to have spend time figuring it out. So I would probably go with find_one() and find_all(), so there's no ambiguity about find.

The person who was confused about rands must have been a beginning coder.

And thanks for addressing this.

jordanbrown0, 2024-02-21:
> I have a fairly strong view that undef should indicate a program error of some sort and not a "valid" return, even though you can test for it.

I think that `undef`, like `NULL` in C, should represent a "soft" error, a failure that's expected, like failing to find the thing you're looking for.  Program errors, bugs in the calling program, should not return `undef`; they should not return anything.  They should be **errors** and stop program execution.  (Yeah, if you don't have "stop on first warning" checked, it would have to return **something**, and that something would probably be `undef`.  But you're already in cascading-error territory, and so it doesn't bother me that `undef` gets overloaded for both purposes.)

nophead, 2024-02-21:
`undef` is fine for any function where the other returns count as true in an `if()` because `undef` counts as false without erroring.

jordanbrown0, 2024-02-22:
Unfortunately, that wouldn't work for `find()` since it returns indexes, and those start as zero, and zero is falsey.

rjcarlson49, 2024-02-22:
In what way doesn't it work? You can't say if (find(...)), but you can certainly say if (find(...)<0) or if (find(...)>=0). Or r = find(...); if (r>=0) ...

jordanbrown0, 2024-02-22:
I meant that, with `find()` returning `undef` for not-found, or for that matter if it returns -1 for not-found, `if(find())` will not tell you whether or not it found something.

With `find()` returning `undef` for not-found, finding the entry at index 0 is falsey and so equivalent to undef.
With `find()` returning -1 for not-found, a not-found is truthy, finding at index 0 is falsey, and finding later is truthy.

The fact that `undef` is falsey is useful in some cases, but not this one.