openscad/openscad #4997: 2024.02.04 (git cc7cbd6b5) Number School Works, but will only increment

ChickenPigAndEgg, 2024-02-19:
2024.02.04 (git cc7cbd6b5) is awesome in terms of increased rendering time.

However, the Number School will only increment, it will not decrement.
To be clear, mouse wheel forward => +1step ( have tried change the value of the step, no change)
To be clear, mouse wheel backward=> +1step ( have tried change the value of the step, no change).

Thank you for all the work you do!

ChickenPigAndEgg, 2024-02-19:
Oops, please edit the title and post to say Number Scroll, not School.

ChickenPigAndEgg, 2024-02-19:
This issue is repeatable an .exe installation of version 2024.02.17 (git f56889987).
It is repeatable whether OPENscad is run as admin or not.
When Modifier for Wheel Scroll is set to Left Mouse Button or Either, the wheel will increment and decrement with the Left Mouse Button.  However, with <ALT> either wheel direction increments.
I have tested this with 2 keyboards, (1 Logitech with support software, and 1 generic) and 2 mice (1 Logitech with support software, and 1 generic).
The issue is repeatable and consistent.

ChickenPigAndEgg, 2024-02-19:
Sorry, I am new here.  ALT enclosed in <> does not appear.  However, with PRESSING_the_ALT_key either wheel direction increments.

kintel, 2024-02-19:
Works as intended on macOS; this could be a Windows-specific issue.

ChickenPigAndEgg, 2024-02-19:
Issue also exists with the windows zip version 2024.02.17 (git f56889987).