openscad/openscad #5000: Crash in STL export after rotate_extrude

curtmcd, 2024-02-20:
I'm getting an assertion failure crash here:
```
src/io/export_stl.cc:131: // Tessellation already eliminated these cases.
src/io/export_stl.cc:132: uint64_t {anonymous}::append_stl(std::shared_ptr<const PolySet>, std::ostream&, bool): Assertion `p0 != p1 && p0 != p2 && p1 != p2' failed.
```
using the master branch checked out & built Feb 13 under Ubuntu 22.04.

It seems like the geometry created by rotate_extrude is in error, and/or the tessellation mentioned in the comment has not eliminated these cases.

To reproduce, I enter just the following, press F6, then press F7 to export STL. The crash happens during the STL export.
`rotate_extrude() square([10, 10]);`

It doesn't matter the value of angle or presence of the angle argument.
It doesn't matter CGAL vs Manifold.

It also crashes also on the code example from manual section 2D_to_3D_Extrusion:
`rotate_extrude($fn = 200) polygon( points=[[0,0],[2,1],[1,2],[1,3],[3,4],[0,5]] );`

These objects have an edge along the Y axis which is permitted.

UBaer21, 2024-02-20:
Did the crash closes scad without warning/error? And what happens with 3mf export?
I have seen some crashes on 2024.02.04 win when rendering or saving but couldn't reproduce.

pca006132, 2024-02-20:
Did you try the latest version? We fixed some cache related bugs recently.

kintel, 2024-02-20:
Try something after Feb 15 where this is fixed: https://github.com/openscad/openscad/pull/4990

curtmcd, 2024-02-21:
Just now I checked it out from scratch (Feb 20 dc2a0d1), re-built, and the bug is still there.
OpenSCAD prints the assertion failure to stderr then quits.
Export to 3mf does not crash, but reports an error in the console:
```
Rendering finished.
EXPORT-ERROR: Can't add triangle to 3MF model.
3MF export finished: /home/csm/3D_Printing.git/Math_Objects/Reuleaux/bug.3mf
```
It could be something in my build environment. However, a checkout from 2023 Dec 30 had the problem, while a checkout from 2022 Apr 27 does not. I may try to do a bisect, although it would be painful and manual because of git submodule hell.

pca006132, 2024-02-21:
1. Are your submodules up to date? Try `git submodule update --init --recursive`.
2. What is your OS version?

I think if you do a bisect, you probably don't have to update the submodules along with the bisect. The API of manifold has been pretty stable.

kintel, 2024-02-21:
Also, to eliminate a local issue, you could always try one of the recent dev binaries from https://openscad.org/downloads.html#snapshots

curtmcd, 2024-02-21:
The latest AppImage ran fine on Ubuntu 22.04/Jammy. As yet I have no idea why my build might be different, although I do build with EXPERIMENTAL=1.

@pca006132 Perhaps overkill, but to avoid confounding anything, for every test I started from scratch. Removed everything, cloned from github, checked out specific version, inited submodules, and built.

I ended up doing a full 9-stage manual bisect and found the exact commit where it stops working:

2023-12-03 16:41:36 fe2b635 = ok
2023-12-03 17:29:27 de1e7d8 = crash

Dependency check
```
$ ./scripts/check-dependencies.sh 
depname           minimum           found             OKness            
qt                5.4               5.15.3            OK                
qscintilla2       2.9               2.11.6            OK                
cgal              5.4               5.4               OK                
gmp               5.0               6.2.1             OK                
mpfr              3.0               4.1.0             OK                
boost             1.56              1.74              OK                
opencsg           1.4.2             1.4.2             OK                
glew              1.5.4             1.7.0             OK                
eigen             3.0               3.4.0             OK                
glib2             2.0               2.72.4            OK                
fontconfig        2.10              2.13.             OK                
freetype2         2.4               2.11.1            OK                
harfbuzz          0.9.19            2.7.4             OK                
libzip            0.10.1            1.7.3             OK                
bison             2.4               3.8.2             OK                
flex              2.5.35            2.6.4             OK                
make              3                 4.3               OK                
double-conversion 2.0.1             2.0.1             OK       
```

pca006132, 2024-02-21:
If we have an extremely sliver/small triangle that differs by less than the resolution of float (23 bits) but different when represented as doubles (52 bits), it will become the same when we export to STL (the assertion is checking double, but we eventually outputs float). I think we removed quantization when exporting STL (at least it is not obvious to me where we do it), not sure if we want to add it back.

Also, @curtmcd maybe you can try this:
```diff
--- a/src/geometry/PolySetBuilder.cc
+++ b/src/geometry/PolySetBuilder.cc
@@ -148,7 +148,7 @@ std::unique_ptr<PolySet> PolySetBuilder::build()
   polyset->setConvexity(convexity_);
   polyset->isTriangular = true;
   for (const auto& face : polyset->indices) {
-    if (face.size() > 3) {
+    if (face.size() != 3) {
       polyset->isTriangular = false;
       break;
     }
```

I think we forgot the case where there the face can be degenerate...

kintel, 2024-02-21:
What about faces with >=3 vertices, but some indices are duplicated, causing degenerate triangles or self-intersections. Perhaps the builder should remove those (at least the degenerate ones) while building?

kintel, 2024-02-21:
Perhaps we need to start unit-testing some of the core classes, to establish a more formal contract, as it's easy to introduce bugs which gets magically fixed by some downstream feature, perhaps only in some cases.

pca006132, 2024-02-21:
True, the builder may get a degenerate triangle, merge vertices together and end up with a triangle with duplicated indices. Manifold can output degenerate triangles, i.e. vertices can have different indices but refers to the same position.

Yeah I think we should write out the assumption of those core classes. Their invariants are pretty unclear now.

pca006132, 2024-02-21:
Ah! Did you build with `-DCMAKE_BUILD_TYPE=Release` or `RelWithDebInfo`? I think for non-debug builds assertions are disabled. Yeah this is indeed a bug. I see degenerate triangles in my stl output as well (although it seems that other programs handle them just fine). Btw it doesn't matter which backend you are using: `rotate_extrude` directly gives you a polyset :).

curtmcd, 2024-02-21:
I only built like this: `mkdir build && cd build && cmake .. -DEXPERIMENTAL=1 && make`
Let me know if I can help further!

kintel, 2024-02-21:
I wonder if we should, generally, write some output sanity checkers for output file formats, and run such checks as part of tests. e.g. like `stlsanitytest` (https://github.com/openscad/openscad/blob/dc2a0d1c28e41016abf4f91ffbc4c6c707c4f64a/tests/CMakeLists.txt#L991C18-L991C35) but better.

kintel, 2024-02-22:
I opted for adding a run-time sanity-check assert of PolySet when in debug mode instead. This is pretty aggressive, but will only crash in debug mode, but will catch such issues much earlier in the pipeline.

kintel, 2024-03-17:
@curtmcd It took some time to finalize this, but try with latest master.

closed 2024-03-17 by kintel

curtmcd, 2024-03-20:
> @curtmcd It took some time to finalize this, but try with latest master.

I find that the issue is gone in latest master 23d5bb96ee8f84bb82dbccec8b840f0956e0b21c. Thank you!

PR #5007 (2024-03-17): Remove degenerate polygons in PolySetBuilder
Also includes some PolySetBuilder refactoring.

Fixes #5000

Note: rotate_extrude() could be improved to better handle manifold corner cases. Deferring to #5046