openscad/openscad #5003: Color and/or Material Export

hansonry, 2024-02-20:
**Is your feature request related to a problem? Please describe.**
As a game developer I would like to export UV maps and textures from openscad as well as geometry. 

**Describe the solution you'd like**
For starters, just getting UVs for faces all tied to a single pixel in a generated pallet texture would be amazing. I would be fine with using the color system already in place for the previews.

Assuming I had a magic wand; I can imagine having "levels" of geometry.  You pick a level and everything above that would be geometry (mesh) and everything below that would be a ray traced textures (color, normal, etc...) applied to geometry. This way you could dynamically create LOD models. Everything could be self contained inside a single scad script, no imported textures or UVs.

**Describe alternatives you've considered**
Writing my own tool, giving up and working in blender with a mouse of all things.... *gasp*

An attempt I made at solving my own problem from scratch: https://gitlab.com/hansonry/sdf-modeler

**Additional context**

I am willing to put some coding effort towards getting this feature up, so I am mostly looking for thoughts and ideas and places where I could start "hacking" to get a prototype working. From there I imagine we could have some more concrete design discussions.

I have been talking to some people in IRC and it seems like there is some interest. More specifically interest in exporting materials in general, mostly for 3D printing. I suspect I am one of the few (maybe only me) interested in gamedev.

Ideally, we can find a way to satisfiy both the printers and the gamedevs (me). But face materials and volumetric materials are two different things (as I have been learning in the IRC). So I see this as a challenge, we don't want to block one by designing ourselves into a corner with the other.

Someone in the IRC posted this manifold Issue that did some texture mapping. So this seems feasible at least.

* https://github.com/elalish/manifold/discussions/683

I searched the issues for related topics:

* #1608
* #4671

Links about UV textures because I am not sure how familiar they are in the openscad community:

* https://en.wikipedia.org/wiki/UV_mapping
* https://www.pluralsight.com/blog/film-games/understanding-uvs-love-them-or-hate-them-theyre-essential-to-know

gsohler, 2024-02-20:
>    - elalish/manifold#683

In my branch I can at least display textures on the solid's faces. you
might get an impression  on pythonscad.org
with further input and more directions, maybe we can work towards
exporting uv maps ?

hansonry, 2024-02-20:
@gsohler Do you plan on merging your branch back into master or is it only for pythonscad?

gsohler, 2024-02-20:
yes of course,

still trying to get 1st part of my work merged here

https://github.com/openscad/openscad/pull/4841

However I am not the owner of the openscad repo and it's not in my hands.
For the time being my work is available in another place.
I hope that my homepage becomes obsolete some time in the future.

hansonry, 2024-02-20:
@gsohler That sounds interesting to me. Seems like that might be an easier path to get to what I am looking for.

gsohler, 2024-02-20:
Sure!

What I have done is just the ability to display textures instead of colors
on faces in openscad preview.(F5)
you can load several textures from jpeg files and activate them with
color(texture=xxx)
UV mapping is rather primitive but viable. I am happy to receive
improvement suggestions.

BTW: pythonscad can also do SDF like your previous project using libfive,
and you can literally code the SDF equations alongside your python
code(with libfive).
Checkout the accompanying library which does some primitives but also some
nice facets. happy to receive feedback here, too

collinalexbell, 2024-07-16:
I want this for gamedev stuff

kintel, 2024-07-16:
We have basic color support since #5185.
Export is currently limited to OFF.

collinalexbell, 2024-07-17:
Oh. Nice. The assimp importer library supports OFF.

jordanbrown0, 2025-04-13:
I suspect that color/material per se is covered by #1608.  What seems unique here is the desire for UV maps, and perhaps that should be reflected in the title.