openscad/openscad #5004: Doesn't stop execution on errors

jordanbrown0, 2024-02-21:
**Describe the bug**
With "stop on the first warning" set, execution stops on the first warning.
However, it does *not* stop on errors(!).

**To Reproduce**
```
polyhedron() cube();
echo("hello");
```
correctly does not print "hello".

```
polyhedron(points=5);
echo("hello");
```
reports an error (ERROR: Unable to convert points = 5 to a vector of coordinates in file , line 1), without a stack trace, but then continues and prints "hello".

**Expected behavior**
Terminate with a stack trace.

**Environment and Version info (please complete the following information):**
Windows 11, OpenSCAD 2023.12.31.

**Additional context**
The problem appears to be at
https://github.com/openscad/openscad/blob/dc2a0d1c28e41016abf4f91ffbc4c6c707c4f64a/src/utils/printutils.cc#L119
where it handles two cases:
* Throws on warning if that's enabled.
* Records a deferred exception on error if suppressing throws.
but an ordinary `message_group::Error` without `no_throw` does nothing.

It seems like the second half of the `if` should just check for `Error`, and not check `no_throw`.

Parkinbot, 2024-03-10:
the problem goes far worse. 
The compiler doesn't stop on polyhedron errors. In a complex sweep operation you can easily create millions of ERROR: lines in the console without any mean to stop the run. 
 
```
obj=[[0,0], [0,1], [1,1], [1,0]];  
polyhedron(obj[0], [0, 1, 2, 3]); 

```
[ERROR: Unable to convert points[0] = 0 to a vec3 of numbers in file , line 2](https://github.com/openscad/openscad/issues/2,C:/Users/rhutt/Documents/OpenSCAD/libraries)
[ERROR: Unable to convert points[1] = 0 to a vec3 of numbers in file , line 2](https://github.com/openscad/openscad/issues/2,C:/Users/rhutt/Documents/OpenSCAD/libraries)
[ERROR: Unable to convert faces[0] = 0 to a vector of numbers in file , line 2](https://github.com/openscad/openscad/issues/2,C:/Users/rhutt/Documents/OpenSCAD/libraries)
[ERROR: Unable to convert faces[1] = 1 to a vector of numbers in file , line 2](https://github.com/openscad/openscad/issues/2,C:/Users/rhutt/Documents/OpenSCAD/libraries)
[ERROR: Unable to convert faces[2] = 2 to a vector of numbers in file , line 2](https://github.com/openscad/openscad/issues/2,C:/Users/rhutt/Documents/OpenSCAD/libraries)
[ERROR: Unable to convert faces[3] = 3 to a vector of numbers in file , line 2](https://github.com/openscad/openscad/issues/2,C:/Users/rhutt/Documents/OpenSCAD/libraries)
Compiling design (CSG Products generation)...

kintel, 2024-03-10:
I agree that we need a way of stopping on errors. Currently, both the test frameworks and some people may rely on the current behavior, so I'd suggest that, as a start, we silently introduce a UI + cmd-line setting for this, potentially merged with the existing `hardwarning` setting.

The question is how to (short-term) structure it for backwards compatibility, until we clean up tests etc., and (longer-term) set up good defaults.

Long-term, I'm not sure if this is appropriate for OpenSCAD, but from the compiler world (e.g. clang), we have the concept of first deciding to upgrade warnings to errors (`--Werror`) , then how to deal with how many errors to accept before aborting (`-ferror-limit=`), with a bunch of ways of tuning this behavior with additional flags.

The current behavior is indeed a bit weird; default is like clang's `-ferror-limit=0` (unlimited), but with `--hardwarning`, all warnings are upgraded to fatal errors, while the error limit is still unlimited.