openscad/openscad #5011: Add a "height" parameter for rotate_extrude to help with threading

aprotim, 2024-02-24:
If trying to build threaded parts of any kind in OpenSCAD, there's a persistent frustration that the only in-built method to extrude a shape both vertically and rotationally is to use `linear_extrude`, which keeps the extruded polygon parallel to the XY plane. Ideally, when creating threads (or similar helical structures, like corkscrews, corkscrew marble channels, etc), it would be possible to use 2D geometry to describe the vertical cross-section of the helix and extrude that along a helical path. 

**Describe the solution you'd like**
In a similar way that the `angle` parameter of `linear_extrude` extends linear (vertical) extrusion to do rotation as well, an optional `height` parameter for `rotate_extrude` could extend rotate extrusion to include vertical translation as well.

**Describe alternatives you've considered**
Common solutions involve implementing the rotation in OpenSCAD using `hull()` (which only works with convex cross-sections, or using `linear_extrude()` as in [this user-created library](https://www.thingiverse.com/thing:1958354), but these methods are painfully slow, and ad hoc for a function that is useful in many modeling contexts.

**Additional context**
I have implemented the `height` parameter in my copy of openscad (PR incoming), and the speedup for spiralizing a semi-complex (non-convex) thread cross-section is 100x in preview mode compared to the linked `spiral_extrude` library and 200x compared to my home-brewed `hull()`-based implementation (done by decomposing the cross-section into convex positive shapes and convex negative shapes, and differenceing the hulls). For a full render, the difference is even starker. The C++ implementation finishes in < 0.15s. Both other methods take 10s of minutes. (I'm still waiting for the faster one to finish, it's been at least 20 minutes. I can give actual benchmark numbers later, if required.)

aprotim, 2024-02-24:
Ah, ok. The spiral extrude method ended up with

Total rendering time: 0:33:55.396

pca006132, 2024-02-25:
Did you try snapshot with manifold?

pca006132, 2024-02-25:
The `spiral_extrude` library renders in 0.5s when manifold is enabled... don't think it is so slow.

aprotim, 2024-02-26:
Even with manifold, for a small example (attached as a .txt), spiral_extrude takes about 5x longer to render and has significant noticeable artifacts, whereas rotate_extrude renders precisely the intended geometry (for a given segmentation resolution) at a fraction of the time.

You can use the customizer on the attached script to toggle between spiral_extrude and rotate_extrude with a height parameter, if you apply my patch in your local tree.  
[screw top w spiral extrude.txt](https://github.com/openscad/openscad/files/14409270/screw.top.w.spiral.extrude.txt)

jimb15122, 2025-05-13:
Has this been put in a nightly build yet?  It would make generating screw theads so much easier.

jordanbrown0, 2025-05-13:
No.  https://github.com/openscad/openscad/blob/72c9919d63116f8e711f3566ae34e9eb63a2d6e6/src/core/RotateExtrudeNode.cc#L50
(You don't have to be able to read C++ to see that there's no "h" or "height" parameter specified.)

There are any number of thread libraries that will build you standard (or non-standard) threads.  Not quite as satisfying as DIY, but maybe easier.  For instance: https://github.com/BelfrySCAD/BOSL2/wiki/threading.scad

gsohler, 2025-05-13:
Look into here:

https://imgur.com/a/AxW90oR

or

at the end of

https://pythonscad.org/tutorial/site/python_new/

coryrc, 2025-12-15:
So there should be two parts to make this easy:

1. The plane shouldn't only be YZ, it should optionally be tilted to be perpendicular to the end of the helix. Thus polygon defines the cross-section of the thread (e.g. buttress is right triangle).
2. There should also be a rotate_extrusion *function* which reports where the start and end faces are. Technically that's available if you examine the geometry, but it's not trivial to get, yet the C++ code will already exist to calculate, so provide it to the users.