openscad/openscad #5014: Icons and frame information is hard to read on high DPI/retina displays

amatulic, 2024-02-26:
Closed issue #3786 apparently changed the toolbar icons to SVG, but I find all the UI elements hard to make out with my eyes on a retina display. Everything is too small. I prefer to work with OpenSCAD on my lower-resolution computers because I can not only see, but aim my mouse better.

**Solution I'd like**
Two possible implementations I'd like to see:
- A preferences choice for magnification of UI elements.
- Internal code change to define UI element sizes in points or web pixels, rather than pixels on the local display hardware.

thehans, 2024-03-05:
The UI scaling is set by the window manager.  Here is what mine looks like on Linux Mint with Cinnamon desktop.
[image]

The OpenSCAD UI changes in accordance to the system Display options.

I don't believe application specific scaling overrides should exist, as this is best handled globally, by the system.

jordanbrown0, 2024-03-05:
I agree that we should work with global scaling preferences.  On Windows 11, we only sort of do.

[ Note:  the screen shots below are from a 100dpi screen.  They might look reasonable at a smaller dot density, but the small text has upper-case letters about 2mm tall. ]

Tweaking the scale setting at (right-click on desktop) > Display Settings > Scale & Layout > Scale, OpenSCAD immediately changes some things, but not all.  Here's what I see when I move from 100% to 200%...

[image]
[image]
[image]
[image]

It doesn't seem to have re-laid-out some of the stuff, and the tool-bar icons didn't change size.

When you restart OpenSCAD, there are some things that are not scaled:

[image]
[image]
[image]

Whether we should have application-specific scaling (or, in general, whether applications should have application-specific scaling) is less clear to me.  For any particular user, different applications have different levels of importance and different habits associated with them.  If I'm using OpenSCAD exclusively as a viewer, using an external editor, maybe I want to have everything other than the main view pane be as small as possible.  For some windows I value readability more than I value cramming more information into the window, and for others it's the other way around.  At a finer-grained level maybe I can read text more readily than I can recognize icons, or the other way 'round.  Maybe I want my title bar and menu bar as small as possible (because I know what's on them and so don't really need to *read* them) and want my editor and console relatively larger.  I'm not suggesting that we have per-UI-element control, but it's not silly.

amatulic, 2024-03-05:
> It doesn't seem to have re-laid-out some of the stuff, and the tool-bar icons didn't change size.

That's exactly the problem I wrote about when I created this issue. The toolbar icons don't change size.

jordanbrown0, 2024-03-06:
The toolbar icons (at least for the two primary toolbars - editor pane and view pane) do change size for me, when I restart OpenSCAD.  But it would be better to not require the restart, and even with a restart there's a bunch of stuff that stays small.

thehans, 2024-03-06:
@amatulic Please include information about OS you are using

amatulic, 2024-03-07:
@thehans - done.

thehans, 2024-03-07:
Here is Qt's documentation on High DPI  https://doc.qt.io/qt-5/highdpi.html

For mac it looks like we may need to add a setting into our `info.plist`
```
<key>NSHighResolutionCapable</key>
<string>True</string>
```

There are some notes about Windows also there, but it is still not entirely clear to me what configuration makes the most sense.
(If you set the application to DPI **unaware**, then that means the system will compensate for that and scale things accordingly?  Conversely if it is DPI aware, does that mean we need a bunch of extra code to adjust our UI sizes dependent upon DPI? )  
Also just the idea of multiple screen with differing DPI is already giving me a headache.

@amatulic 
Were you adjusting scaling while the app was open, or have you tried restarting the app after making system changes?
As Jordan mentioned, I have also noticed that restarting was required to see the full effects of UI scaling adjustment, even on Linux.
But the toolbar icons definitely change size for me, on Linux anyways.