openscad/openscad #5021: glm package missing in scripts/mingw-x-build-dependencies.sh

Chron-O-John, 2024-02-27:
**Describe the bug**
glm is missing in the packages, altough it is necesarry to cross-compile

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the tutorial for cross-compilation in readme.md
2. See Error

**Expected behavior**
glm is found and used

**Code reproducing the issue**
after adding glm to scripts/mingw-x-build-dependencies.sh compiling goes a few steps further (until it fails again, complaining about missing functions, but I have to further investigate for a detailed follow-up)

**Environment and Version info (please complete the following information):**
WSL2 - ubuntu on a Windows10-Machine

kintel, 2024-02-28:
We don't use glm, but Manifold does. Perhaps this is a recent Manifold build system regression. @pca006132 does this ring a bell?

pca006132, 2024-02-28:
weird, can you post the error messages? I thought we should automatically pull it.

imperfectinventions, 2024-03-16:
Hi, I also ran into this issue when trying to cross build for the first time. 

This is the error I got:

`In file included from /etc/opt/openscad/src/geometry/cgal/CGALCache.cc:5:
/etc/opt/openscad/src/geometry/manifold/ManifoldGeometry.h:5:10: fatal error: glm/glm.hpp: No such file or directory
    5 | #include <glm/glm.hpp>
      |          ^~~~~~~~~~~~~`

Adding "glm" to the packages list in the mingw build dependencies file did get me past this error, but then I ran into other errors about conversion from ManifoldGeometry into CGALHybridPolyhedrons and many other data types.

pca006132, 2024-03-16:
Can you post the complete build log? Manifold should normally search for glm and download it automatically.

imperfectinventions, 2024-03-16:
Sure, is this what you're looking for? Or is there a file that the build process created?

[build_stdout.txt](https://github.com/openscad/openscad/files/14623791/build_stdout.txt)

pca006132, 2024-03-16:
No, but the cmake log, e.g. similar to the "Build and Test" part in https://github.com/openscad/openscad/actions/runs/8227281463/job/22494941114

btw, are you building 2023-11-22? there are *a lot* of changes recently, and I recommend trying the latest one (you are already using snapshot anyway).

imperfectinventions, 2024-03-16:
Let me see if I can get the cmake log for you. Trying to build again after switching to the master branch for MXE in the mingw-x-build-dependencies.sh file. I'm running the "mingw-x-build-dependencies.sh 4" command it says in the README for cross building. Is there another script I should run instead? 

This is my first time building OpenSCAD, last time I installed the release version (great program by the way, the work you all have done is incredible).

pca006132, 2024-03-16:
and just to let you know there is a [development snapshot download](https://openscad.org/downloads.html#snapshots) section in the website so you can try the latest stuff without waiting for it to compile

imperfectinventions, 2024-03-24:
Thanks for telling me about the development snapshot download. I was unable to successfully build again, but it wasn't the glm issue this issue is about. I assume my environment isn't set up right on my end.

imperfectinventions, 2024-03-24:
I was able to successfully download the snapshot download, so from my end, the problem is solved.

kintel, 2024-03-24:
@imperfectinventions Are you just building the default binaries using `./scripts/release-common.sh mingw64` (i.e. without any experimental features like Manifold) ?

imperfectinventions, 2024-03-24:
That was what I was trying to do, yes.

kintel, 2024-03-24:
Thx, I'll take a look. We don't currently run non-experimental builds on any CI, so minor build issues may easily sneak in there..

imperfectinventions, 2024-03-24:
Makes sense! Thank you all for your help!

kintel, 2024-09-19:
This works again now. I assume there was a bug which we fixed as part of manifold integration cleanup.

closed 2024-09-19 by kintel