openscad/openscad #5027: Incorrect display at difference

kberkut, 2024-03-03:
**Describe the bug**
Surfaces subject to subtraction are not displayed.

**To Reproduce**
Repeat the simple code
`$fn = $preview ? 100 : 100;
Dh = 50;
Hh = 50;
Dotv = 8;
TH = 5;
module work()
{
	difference()
	{
		cylinder(d = Dh, h = Hh);
    cylinder(d = Dotv, h = Hh + TH);
    translate([0,0,-1]) cylinder(d = Dh - 2 * TH, h = Hh - TH);
	}
}
work();`

**Expected behavior**
Clear visibility of the surfaces of the shape that has undergone the difference operation

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
[image]
[image]

**Environment and Version info (please complete the following information):**
 - OS: Arch Linux                                                                                                                                                                               
 Kernel: x86_64 Linux 6.7.6-arch1-2
 DE: GNOME 45.3
 WM: Mutter

 - System: 
 CPU: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx @ 8x 2.1GHz
 GPU: AMD Radeon Vega 8 Graphics (radeonsi, raven, LLVM 16.0.6, DRM 3.57, 6.7.6-arch1-2)
 RAM: 5194MiB / 9884MiB
 - OpenSCAD Version   2024.03.02.ai18721 (git e3e5f8e0c), appimage

**Additional context**
It is displayed normally if applied: render()
But then the coloring of the colors of the cut surfaces is not displayed.
Function "manifold" may be enable or disable - does not affect. Other all function is off.
[image]

kintel, 2024-03-03:
This seems like a duplicate of the issue discussed at the bottom of #4861. Could you verify whether downgrading from Mesa 24 to Mesa 23 solves the issue?

kberkut, 2024-03-04:
> This seems like a duplicate of the issue discussed at the bottom of #4861. Could you verify whether downgrading from Mesa 24 to Mesa 23 solves the issue?

Downgraded the version. Yes, I confirm. Something has changed in the new versions of openscad, apparently there is a bug somewhere.

[image]

kintel, 2024-03-04:
Thanks, closing as duplicate of #4861

closed 2024-03-04 by kintel

kintel, 2024-04-01:
This was a bug in Mesa. Fixed in Mesa-24.0.4: https://docs.mesa3d.org/relnotes/24.0.4.html