openscad/openscad #5030: 3mf Export Error with lazy union

UBaer21, 2024-03-05:
Version 2024.02.22 
features manifold + lazy union

certain values and geometry positions cause "EXPORT-ERROR: Can't add triangle to 3MF model."
And exporting an empty 3mf file. But renders without error or warning.

Adding a cube outside the geometry (of certain size and distance) the export works fine!

without lazy union i get 
ERROR: [manifold] Input mesh is not closed!
ERROR: [manifold] Surface_mesh -> Manifold conversion failed: NotManifold

example code:
```
size=20; // smaller 18 works

matrixString=[for(i=[0:size^2]) round(rands(0,1,1,i)[0]) ];

module qrCodePattern(matrixString=matrixString, dimension=size) {
    translate([1,1,-1])linear_extrude(2,convexity=5)union(){
        for (y = [0 : dimension-1]) {
            for (x = [0 : dimension-1]) {
                if (str(matrixString[y * dimension + x]) == "1") {
                    translate([x , (dimension - 1 - y) , 0]) square(1, center = true);
                }
            }
        }
    }
}

union(){
  difference(){
    cube(size+1);
    color("navy")qrCodePattern();
  }
  *translate([-5,-5])cube(1); // without this 3mf-export fails
  translate([-100,-2])cube(.001); // this doesn't help
}
```

Adding an offset to overlap the cubes helps but this code is only to demonstrate the problem as i am getting it also in far more complex designs also with proper overlap.

salamanders, 2024-05-11:
Getting this as well.  Complex model, lazy union, attempting to export to 3mf, `"EXPORT-ERROR: Can't add triangle to 3MF model.`

kintel, 2024-05-11:
This should work fine now just with manifold enabled (no lazy union required).

kintel, 2024-05-11:
@salamanders I think you're experiencing a different issue. Could you open a new ticket with an example?

closed 2024-05-11 by kintel