openscad/openscad #5032: Manifold surface cover when using small offsets

UBaer21, 2024-03-05:
Please first consider that general questions about using OpenSCAD or how to write code for specific models are better asked on the mailing list. See https://openscad.org/community.html for more details.

**Environment**
v2024.02.22 on  Windows

**Issue**
offset(-.0001) produces a face covering the surface of the difference when rendering

[image]

code:
```
size=60; // only on larger sizes
matrixString=[for(i=[0:size^2]) round(rands(0,1,1,i)[0]) ];
module qrCodePattern(matrixString=matrixString, dimension=size) {
    translate([1,1,-1])linear_extrude(2,convexity=5)offset(-.0001)union(){
        for (y = [0 : dimension-1]) {
            for (x = [0 : dimension-1]) {
                if (str(matrixString[y * dimension + x]) == "1") {
                    translate([x , (dimension - 1 - y) , 0]) square(1, center = true);
                }
            }
        }
    }
}

  difference(){
    cube(size+1);
    color("navy")qrCodePattern();
  }
```
preview looks fine
[image]
render
[image]

kintel, 2024-03-06:
Reproducible on latest master on macOS as well.

@pca006132 Any ideas? I've seen similar things when playing around with Manifold's 2D triangulator, but I haven't tried to diagnose this yet.

pca006132, 2024-03-06:
sounds like a bug in the triangulator. is it possible to get a minimal example?

kintel, 2024-03-06:
It seems tricky to trigger this on a minimal example, but I'll dig around a bit.

kintel, 2024-03-06:
Some observations:
If you replace the last construct with any of these, it works:
```
difference() {
  translate([0,0,-size-1]) cube(size+1);
  color("navy")qrCodePattern();
}

difference() {
  cube(size+1);
  translate([0,0,size+1]) color("navy")qrCodePattern();
}
```

Also, both `union` and `intersection` work where `difference` fails.

closed 2024-09-20 by kintel