openscad/openscad #5035: Grammar railroad diagram

mingodad, 2024-03-08:
I've just added this project grammar to https://mingodad.github.io/parsertl-playground/playground/ an `Yacc/Lex` compatible online editor/tester (select `Openscad parser (partially working)` from the `Examples` then click `Parse` to see a parse tree for the content in `Input source`) and it also produces an `EBNF` understood by (IPV6) https://www.bottlecaps.de/rr/ui or (IPV4) https://rr.red-dove.com/ui to generate a nice navigable railroad diagram to help document/debug/develop the grammar.
```
//
// EBNF to be viewd at  (IPV6) https://www.bottlecaps.de/rr/ui or (IPV4) https://rr.red-dove.com/ui
//
// Copy and paste this at https://www.bottlecaps.de/rr/ui in the 'Edit Grammar' tab
// then click the 'View Diagram' tab.
//

input::=
	  /*%empty*/
	| input TOK_USE
	| input statement

statement::=
	  ';'
	| '{' inner_input '}'
	| module_instantiation
	| assignment
	| TOK_MODULE TOK_ID '(' parameters ')' statement
	| TOK_FUNCTION TOK_ID '(' parameters ')' '=' expr ';'
	| TOK_EOT

inner_input::=
	  /*%empty*/
	| inner_input statement

assignment::=
	  TOK_ID '=' expr ';'

module_instantiation::=
	  '!' module_instantiation
	| '#' module_instantiation
	| '%' module_instantiation
	| '*' module_instantiation
	| single_module_instantiation child_statement
	| ifelse_statement

ifelse_statement::=
	  if_statement
	| if_statement TOK_ELSE child_statement

if_statement::=
	  TOK_IF '(' expr ')' child_statement

child_statements::=
	  /*%empty*/
	| child_statements child_statement
	| child_statements assignment

child_statement::=
	  ';'
	| '{' child_statements '}'
	| module_instantiation

module_id::=
	  TOK_ID
	| TOK_FOR
	| TOK_LET
	| TOK_ASSERT
	| TOK_ECHO
	| TOK_EACH

single_module_instantiation::=
	  module_id '(' arguments ')'

expr::=
	  logic_or
	| TOK_FUNCTION '(' parameters ')' expr
	| logic_or '?' expr ':' expr
	| TOK_LET '(' arguments ')' expr
	| TOK_ASSERT '(' arguments ')' expr_or_empty
	| TOK_ECHO '(' arguments ')' expr_or_empty

logic_or::=
	  logic_and
	| logic_or OR logic_and

logic_and::=
	  equality
	| logic_and AND equality

equality::=
	  comparison
	| equality EQ comparison
	| equality NEQ comparison

comparison::=
	  addition
	| comparison '>' addition
	| comparison GE addition
	| comparison '<' addition
	| comparison LE addition

addition::=
	  multiplication
	| addition '+' multiplication
	| addition '-' multiplication

multiplication::=
	  unary
	| multiplication '*' unary
	| multiplication '/' unary
	| multiplication '%' unary

unary::=
	  exponent
	| '+' unary
	| '-' unary
	| '!' unary

exponent::=
	  call
	| call '^' unary

call::=
	  primary
	| call '(' arguments ')'
	| call '[' expr ']'
	| call '.' TOK_ID

primary::=
	  TOK_TRUE
	| TOK_FALSE
	| TOK_UNDEF
	| TOK_NUMBER
	| TOK_STRING
	| TOK_ID
	| '(' expr ')'
	| '[' expr ':' expr ']'
	| '[' expr ':' expr ':' expr ']'
	| '[' ']'
	| '[' vector_elements optional_trailing_comma ']'

expr_or_empty::=
	  /*%empty*/
	| expr

list_comprehension_elements::=
	  TOK_LET '(' arguments ')' list_comprehension_elements_p
	| TOK_EACH vector_element
	| TOK_FOR '(' arguments ')' vector_element
	| TOK_FOR '(' arguments ';' expr ';' arguments ')' vector_element
	| TOK_IF '(' expr ')' vector_element
	| TOK_IF '(' expr ')' vector_element TOK_ELSE vector_element

list_comprehension_elements_p::=
	  list_comprehension_elements
	| '(' list_comprehension_elements ')'

optional_trailing_comma::=
	  /*%empty*/
	| ','

vector_elements::=
	  vector_element
	| vector_elements ',' vector_element

vector_element::=
	  list_comprehension_elements_p
	| expr

parameters::=
	  /*%empty*/
	| parameter_list optional_trailing_comma

parameter_list::=
	  parameter
	| parameter_list ',' parameter

parameter::=
	  TOK_ID
	| TOK_ID '=' expr

arguments::=
	  /*%empty*/
	| argument_list optional_trailing_comma

argument_list::=
	  argument
	| argument_list ',' argument

argument::=
	  expr
	| TOK_ID '=' expr

//Tokens

TOK_MODULE ::= "module"
TOK_FUNCTION ::= "function"
TOK_IF ::= "if"
TOK_ELSE ::= "else"
TOK_LET ::= "let"
TOK_ASSERT ::= "assert"
TOK_ECHO ::= "echo"
TOK_FOR ::= "for"
TOK_EACH ::= "each"

TOK_TRUE ::= "true"
TOK_FALSE ::= "false"
TOK_UNDEF ::= "undef"

TOK_USE ::= "use"[ ]+"<"[^>]+">"

LE ::= "<="
GE ::= ">="
EQ ::= "=="
NEQ ::= "!="
AND ::= "&&"
OR ::= "||"
```

kintel, 2024-03-08:
This reads like an announcement, not an issue. Please use the mailing list for announcements & discussion.

If you were intending to report a bug in OpenSCAD’s grammar or parser, we need more information.

closed 2024-03-08 by kintel

t-paul, 2024-03-08:
That still does not explain what actions on OpenSCAD side are asked for. It's nice to see the grammar added to this web application, but what would you expect to be changed in the OpenSCAD application?

mingodad, 2024-03-08:
Add the railroad diagram to the documentation ?

kintel, 2024-03-09:
The bottlecaps diagrams are generally very pretty, and providing those as part of OpenSCAD docs would be interesting. I think we'd want to automatically generate them from our source code though, to avoid having to manually maintain a separate grammar (which will fall out of maintenance pretty quickly). If you have the capability for such automation, I'd be interested!

BenHanson, 2024-03-25:
I've updated read_bison.hpp so that it successfully reads your parser.y, so that is a big part of the auto generation of the format listed above concluded.

kintel, 2024-03-29:
Nice! I guess the next missing piece is to put together a script that can build and run these tools and generate the wanted diagrams, then carve out some space in our docs or website for this. Not sure anyone reading this ticket has time to look into this, but it could be a good discussion for the mailing list or IRC to look for volunteers.