openscad/openscad #5044: Générer du code avec ChatGPT

vincent-olmos, 2024-03-15:
Il semble que ChatGPT soit peu fiable pour la génération de code OpenSCAD !
Copilot ne semble pas faire mieux.

kintel, 2024-03-15:
Please use English for OpenSCAD discussions. This sounds like you're trying to report a weakness in ChatGPT, perhaps open an issue with OpenAI?

closed 2024-03-15 by kintel

vincent-olmos, 2024-03-15:
I'm French and it seems that ChatGPT is unreliable for OpenSCAD code generation!
Copilot doesn't seem to do any better.

jordanbrown0, 2024-03-15:
Yes, it is basically unusable.  It tends to put together mostly-legal programs that don't really have much to do with the shape requested.  But the OpenSCAD development team doesn't have any way to affect that picture.  I don't think even OpenAI can directly affect it.  My understanding of the technology is that nobody specifically taught it to write OpenSCAD programs; it learned on its own by reading web pages.

vincent-olmos, 2024-03-15:
OpenAI est sensé apprendre tout seul !

jordanbrown0, 2024-03-15:
> OpenAI is supposed to learn on its own!

Yep.  Not much we can do about it, other than write OpenSCAD programs and post them on the web for it to see.