openscad/openscad #5045: On Ubuntu 23.10 QScintilla2 Installed but Not Found

cversek, 2024-03-16:
**Describe the bug**
I want to try building the latest OpenSCAD from source on Ubuntu.
So I followed the instructions.  When I ran 
```
sudo ./scripts/uni-get-dependencies.sh
```
In the output was the line:
```
Note, selecting 'libqscintilla2-qt5-dev' instead of 'libqt5scintilla2-dev'
```
Then I ran
```
./scripts/check-dependencies.sh
```
And got the output:
```
depname           minimum           found             OKness            
qt                5.4               5.15.10           OK                
qscintilla2       2.9               unknown           NotOK             
cgal              5.4               5.6               OK                
gmp               5.0               6.3.0             OK                
mpfr              3.0               4.2.1             OK                
boost             1.56              1.74              OK                
opencsg           1.4.2             1.5.              OK                
glew              1.5.4             1.7.0             OK                
eigen             3.0               3.4.0             OK                
glib2             2.0               2.78.0            OK                
fontconfig        2.10              2.14.             OK                
freetype2         2.4               2.13.1            OK                
harfbuzz          0.9.19            8.0.1             OK                
libzip            0.10.1            1.7.3             OK                
bison             2.4               3.8.2             OK                
flex              2.5.35            2.6.4             OK                
make              3                 4.3               OK                
double-conversion 2.0.1             2.0.1             OK                
```
I stopped here.  Thanks.

**Environment and Version info:**
 - OS: Ubuntu 23.10
 - System: Intel PC 64-bit
 - OpenSCAD Version - latest source

cversek, 2024-03-16:
If it helps at all, here is the output of the command `dpkg -L libqscintilla2-qt5-dev | grep /usr/include`:
```
/usr/include
/usr/include/x86_64-linux-gnu
/usr/include/x86_64-linux-gnu/qt5
/usr/include/x86_64-linux-gnu/qt5/Qsci
/usr/include/x86_64-linux-gnu/qt5/Qsci/qsciabstractapis.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qsciapis.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscicommand.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscicommandset.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscidocument.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qsciglobal.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexer.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerasm.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexeravs.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerbash.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerbatch.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexercmake.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexercoffeescript.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexercpp.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexercsharp.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexercss.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexercustom.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerd.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerdiff.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexeredifact.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerfortran.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerfortran77.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerhex.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerhtml.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexeridl.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerintelhex.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerjava.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerjavascript.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerjson.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerlua.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexermakefile.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexermarkdown.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexermasm.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexermatlab.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexernasm.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexeroctave.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerpascal.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerperl.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerpo.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerpostscript.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerpov.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerproperties.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerpython.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerruby.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerspice.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexersql.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexersrec.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexertcl.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexertekhex.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexertex.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerverilog.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexervhdl.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexerxml.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscilexeryaml.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscimacro.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qsciprinter.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qsciscintilla.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qsciscintillabase.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscistyle.h
/usr/include/x86_64-linux-gnu/qt5/Qsci/qscistyledtext.h
```

cversek, 2024-03-16:
@kintel 
Ah, yes I did try to build - without success.  I should have posted this sooner!
```
$ cd build && cmake .. -DEXPERIMENTAL=1
CMake Deprecation Warning at submodules/mimalloc/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- 
-- No build type selected, default to: Release
-- Override standard malloc (MI_OVERRIDE=ON)
-- Use the C++ compiler to compile (MI_USE_CXX=ON)
-- 
-- Library base name: mimalloc
-- Version          : 1.7
-- Build type       : release
-- C++ Compiler     : /usr/bin/c++
-- Compiler flags   : -Wall;-Wextra;-Wno-unknown-pragmas;-fvisibility=hidden;-ftls-model=initial-exec;-fno-builtin-malloc
-- Compiler defines : 
-- Link libraries   : /usr/lib/x86_64-linux-gnu/libpthread.a;/usr/lib/x86_64-linux-gnu/librt.a
-- Build targets    : static
-- 
-- Eigen: 3.4.0
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.56") found components: filesystem system regex program_options 
-- Boost: 1.74.0
-- Harfbuzz: 8.0.1
-- Fontconfig: 
-- Glib: 2.78.0
-- OpenCSG: 1.5.0
-- USE_GLAD/USE_GLEW not specified: Defaulting to glew
-- GLEW: 2.2.0
-- OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so;/usr/lib/x86_64-linux-gnu/libGLX.so;/usr/lib/x86_64-linux-gnu/libGLU.so
-- Using header-only CGAL
-- Targeting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.66")  
-- Boost include dirs: /usr/include
-- Boost libraries:    
-- Using gcc version 4 or later. Adding -frounding-math
-- CGAL: 5.6.0
-- CGAL: Using target CGAL::CGAL CGAL::CGAL_Core
-- libzip: 1.7.3
-- Freetype: 2.13.1
-- LibXml2: 2.9.14
-- HIDAPI: disabled
-- SpNav: disabled
-- CAIRO_INCLUDEDIR: /usr/include
-- CAIRO_INCLUDE_DIRS: /usr/include/cairo;/usr/include/libpng16;/usr/include/freetype2;/usr/include/pixman-1
-- CAIRO_LIBRARIES: cairo
-- Cairo: 1.18.0
-- Flex: 2.6.4
-- Bison: 3.8.2
-- lib3mf: disabled
-- Qt5: 5.15.8
CMake Error at /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Qt5QScintilla (missing: QT5QSCINTILLA_INCLUDE_DIR) (Required
  is at least version "2.8.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindQt5QScintilla.cmake:112 (find_package_handle_standard_args)
  CMakeLists.txt:543 (find_package)

-- Configuring incomplete, errors occurred!
```

I even tried, unsuccessfully, setting and environment variable like so:
```
export QT5QSCINTILLA_INCLUDE_DIR=/usr/include/x86_64-linux-gnu/qt5/Qsci
```

Thanks for the help, I am sure you have many Ubuntu 23.10 users, just not ones trying to build from source ;)  What I'm excited to try is the new CSG backend(s?).

t-paul, 2024-03-16:
Please run:
```
cmake .. -DEXPERIMENTAL=1 -Wno-dev --trace-expand --trace-source=FindQt5QScintilla.cmake
```
and post the associated logs after the line `-- Qt5: 5.15.8`

cversek, 2024-03-17:
@t-paul 
Here is the relevant output from that command:
```
-- Qt5: 5.15.8
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(46):  find_path(QT5QSCINTILLA_INCLUDE_DIR NAMES qsciscintilla.h HINTS /home/cversek/mambaforge/include/qt/;/home/cversek/mambaforge/include/qt/QtWidgets;/home/cversek/mambaforge/include/qt/QtGui;/home/cversek/mambaforge/include/qt/QtCore;/home/cversek/mambaforge/.//mkspecs/linux-g++;/usr/include PATH_SUFFIXES qt5 Qsci )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(52):  set(QT5QSCINTILLA_INCLUDE_DIRS QT5QSCINTILLA_INCLUDE_DIR-NOTFOUND )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(55):  set(_VERSION_FILE QT5QSCINTILLA_INCLUDE_DIR-NOTFOUND/qsciglobal.h )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(56):  if(EXISTS QT5QSCINTILLA_INCLUDE_DIR-NOTFOUND/qsciglobal.h )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(68):  set(_QT5QSCINTILLA_VERSION_MATCH TRUE )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(69):  if(QScintilla_FIND_VERSION AND QT5QSCINTILLA_VERSION_STRING )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(82):  find_library(QT5QSCINTILLA_LIBRARY NAMES qt5scintilla2 qscintilla2-qt5 qscintilla2 qscintilla2_qt5 HINTS Qt5::Widgets )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(87):  set(QT5QSCINTILLA_LIBRARIES /usr/lib/x86_64-linux-gnu/libqscintilla2_qt5.so )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(89):  IF(QT5QSCINTILLA_LIBRARY AND QT5QSCINTILLA_INCLUDE_DIR )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(99):  if(QT5QSCINTILLA_INCLUDE_DIR )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(102):  elseif(QT5QSCINTILLA_LIBRARY )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(103):  string(REGEX REPLACE (.*)/lib[/|32|64].* \1 QT5QSCINTILLA_ROOT_DIR /usr/lib/x86_64-linux-gnu/libqscintilla2_qt5.so )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(108):  include(FindPackageHandleStandardArgs )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(109):  if(CMAKE_VERSION VERSION_LESS 2.8.3 )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(111):  else()
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(112):  find_package_handle_standard_args(Qt5QScintilla REQUIRED_VARS QT5QSCINTILLA_LIBRARY QT5QSCINTILLA_INCLUDE_DIR _QT5QSCINTILLA_VERSION_MATCH VERSION_VAR QT5QSCINTILLA_VERSION_STRING )
CMake Error at /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Qt5QScintilla (missing: QT5QSCINTILLA_INCLUDE_DIR) (Required
  is at least version "2.8.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindQt5QScintilla.cmake:112 (find_package_handle_standard_args)
  CMakeLists.txt:543 (find_package)

-- Configuring incomplete, errors occurred!
```
Disclosure, I am running this from within a base conda (mambaforge) environment if that might have anything to do with it.

t-paul, 2024-03-17:
As the first line shows, that environment breaks `cmake`. Normally it looks like:
```
/home/tp/projects/o/openscad-branch/cmake/Modules/FindQt5QScintilla.cmake(46):  find_path(QT5QSCINTILLA_INCLUDE_DIR NAMES qsciscintilla.h HINTS /usr/include/x86_64-linux-gnu/qt5/;/usr/include/x86_64-linux-gnu/qt5/QtWidgets;/usr/include/x86_64-linux-gnu/qt5/QtGui;/usr/include/x86_64-linux-gnu/qt5/QtCore;/usr/lib/x86_64-linux-gnu/qt5//mkspecs/linux-g++ PATH_SUFFIXES qt5 Qsci )
```

It may help adding `-DQT5QSCINTILLA_INCLUDE_DIR=/usr/include/x86_64-linux-gnu` to `cmake`, it may also need the library path specified. In any case it does not seems to be an OpenSCAD issue at this point.

cversek, 2024-03-17:
@t-paul 
Thanks for the tip.  I ran:
```
cmake .. -DEXPERIMENTAL=1 -Wno-dev --trace-expand --trace-source=FindQt5QScintilla.cmake -DQT5QSCINTILLA_INCLUDE_DIR=/usr/include/x86_64-linux-gnu
```
With better looking results, albeit some warnings related to mambaforge (conda env):
```
Put cmake in trace mode, but with variables expanded.
Put cmake in trace mode, but output only lines of a specified file. Multiple options are allowed.
-- 
-- No build type selected, default to: Release
-- Override standard malloc (MI_OVERRIDE=ON)
-- Use the C++ compiler to compile (MI_USE_CXX=ON)
-- 
-- Library base name: mimalloc
-- Version          : 1.7
-- Build type       : release
-- C++ Compiler     : /usr/bin/c++
-- Compiler flags   : -Wall;-Wextra;-Wno-unknown-pragmas;-fvisibility=hidden;-ftls-model=initial-exec;-fno-builtin-malloc
-- Compiler defines : 
-- Link libraries   : /usr/lib/x86_64-linux-gnu/libpthread.a;/usr/lib/x86_64-linux-gnu/librt.a
-- Build targets    : static
-- 
-- Eigen: 3.4.0
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.56") found components: filesystem system regex program_options 
-- Boost: 1.74.0
-- Harfbuzz: 8.0.1
-- Fontconfig: 
-- Glib: 2.78.0
-- OpenCSG: 1.5.0
-- USE_GLAD/USE_GLEW not specified: Defaulting to glew
-- GLEW: 2.2.0
-- OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so;/usr/lib/x86_64-linux-gnu/libGLX.so;/usr/lib/x86_64-linux-gnu/libGLU.so
-- Using header-only CGAL
-- Targeting Unix Makefiles
-- Using /usr/bin/c++ compiler.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.66")  
-- Boost include dirs: /usr/include
-- Boost libraries:    
-- Using gcc version 4 or later. Adding -frounding-math
-- CGAL: 5.6.0
-- CGAL: Using target CGAL::CGAL CGAL::CGAL_Core
-- libzip: 1.7.3
-- Freetype: 2.13.1
-- LibXml2: 2.9.14
-- HIDAPI: disabled
-- SpNav: disabled
-- CAIRO_INCLUDEDIR: /usr/include
-- CAIRO_INCLUDE_DIRS: /usr/include/cairo;/usr/include/libpng16;/usr/include/freetype2;/usr/include/pixman-1
-- CAIRO_LIBRARIES: cairo
-- Cairo: 1.18.0
-- Flex: 2.6.4
-- Bison: 3.8.2
-- lib3mf: disabled
-- Qt5: 5.15.8
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(46):  find_path(QT5QSCINTILLA_INCLUDE_DIR NAMES qsciscintilla.h HINTS /home/cversek/mambaforge/include/qt/;/home/cversek/mambaforge/include/qt/QtWidgets;/home/cversek/mambaforge/include/qt/QtGui;/home/cversek/mambaforge/include/qt/QtCore;/home/cversek/mambaforge/.//mkspecs/linux-g++;/usr/include PATH_SUFFIXES qt5 Qsci )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(52):  set(QT5QSCINTILLA_INCLUDE_DIRS /usr/include/x86_64-linux-gnu )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(55):  set(_VERSION_FILE /usr/include/x86_64-linux-gnu/qsciglobal.h )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(56):  if(EXISTS /usr/include/x86_64-linux-gnu/qsciglobal.h )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(68):  set(_QT5QSCINTILLA_VERSION_MATCH TRUE )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(69):  if(QScintilla_FIND_VERSION AND QT5QSCINTILLA_VERSION_STRING )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(82):  find_library(QT5QSCINTILLA_LIBRARY NAMES qt5scintilla2 qscintilla2-qt5 qscintilla2 qscintilla2_qt5 HINTS Qt5::Widgets )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(87):  set(QT5QSCINTILLA_LIBRARIES /usr/lib/x86_64-linux-gnu/libqscintilla2_qt5.so )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(89):  IF(QT5QSCINTILLA_LIBRARY AND QT5QSCINTILLA_INCLUDE_DIR )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(90):  SET(QT5QSCINTILLA_FOUND TRUE )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(99):  if(QT5QSCINTILLA_INCLUDE_DIR )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(100):  string(REGEX REPLACE (.*)/include.* \1 QT5QSCINTILLA_ROOT_DIR /usr/include/x86_64-linux-gnu )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(108):  include(FindPackageHandleStandardArgs )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(109):  if(CMAKE_VERSION VERSION_LESS 2.8.3 )
/home/cversek/gitwork/openscad/cmake/Modules/FindQt5QScintilla.cmake(111):  else()
[... 72 lines cut ...]
    runtime library [libharfbuzz.so.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/cversek/mambaforge/lib
    runtime library [libfontconfig.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/cversek/mambaforge/lib
    runtime library [libglib-2.0.so.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/cversek/mambaforge/lib
    runtime library [libfreetype.so.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/cversek/mambaforge/lib
    runtime library [libxml2.so.2] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/cversek/mambaforge/lib
    runtime library [libX11.so.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/cversek/mambaforge/lib
    runtime library [libQt5DBus.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/cversek/mambaforge/lib

  Some of these libraries may not be found correctly.

-- Generating done (0.0s)
-- Build files have been written to: /home/cversek/gitwork/openscad/build
```
I also tried first running `conda deactivate` first but that didn't seem to fix the warnings.  I guess I might have to run a shell without the "conda initialize" junk that its installer puts in `~/.bashrc`.

I'm currently running make and will report back on whether the build completes and the executable check outs. 

Anyway I will close this issue but maybe comment back here on any additional relevant findings.  Thanks for the rapid resolution!

closed 2024-03-17 by cversek

cversek, 2024-03-17:
As the warnings might suggest the code compiles but fails to find its libs:
```
~/gitwork/openscad/build$ ./openscad 
./openscad: /home/cversek/mambaforge/lib/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by ./openscad)
./openscad: /home/cversek/mambaforge/lib/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by /lib/x86_64-linux-gnu/libboost_program_options.so.1.74.0)
```

Next I'm going to try purging conda from my shell and rebuilding.

cversek, 2024-03-17:
@kintel  @t-paul 
This is what a  `conda` installation puts into `.bashrc`:
```
# >>> conda initialize >>>
# !! Contents within this block are managed by 'conda init' !!
__conda_setup="$('/home/cversek/mambaforge/bin/conda' 'shell.bash' 'hook' 2> /dev/null)"
if [ $? -eq 0 ]; then
    eval "$__conda_setup"
else
    if [ -f "/home/cversek/mambaforge/etc/profile.d/conda.sh" ]; then
        . "/home/cversek/mambaforge/etc/profile.d/conda.sh"
    else
        export PATH="/home/cversek/mambaforge/bin:$PATH"
    fi
fi
unset __conda_setup

if [ -f "/home/cversek/mambaforge/etc/profile.d/mamba.sh" ]; then
    . "/home/cversek/mambaforge/etc/profile.d/mamba.sh"
fi
# <<< conda initialize <<<
```
I guess its the aggressive PATH manipulation that causes most of the problems.  

With that removed, the basic build instructions work like a charm.  I'm looking forward to trying out all the new features since the `2021.01` build!

cversek, 2024-03-17:
@t-paul 
Yeah thanks for showing me where the new binaries are.  It does feel good to build it right from the github repo.  I don't know if I'm qualified to dig into the internals and change things, but it's nice to have the option.

I just assumed that the Tags were the latest builds and you weren't doing them any more ;) It seems like at least the Ubuntu people made the same assumption and have not updated their package in a few years.   I also have an M3 Mac and the `brew` package is also stuck at `2021.1`.  I guess this must be a decision about what is considered stable and where the development priorities are presently?  I love OpenSCAD and have been using it for over 10 years, though I'm hardly an expert.  The CGAL render step has often been a barrier to doing more complex generative design work for me, so was hoping for new backends that work better on modern multicore systems.   I read a little in the Issues about Manifold and it got me excited to try compilation, though I actually have no clue what it is all about.  I haven't benchmarked anything yet, but will do so in the coming weeks!

cversek, 2024-03-20:
Wow, I just wanted to comment on a little benchmark I ran using the "Manifold" backend option, my reason for wanting the latest code in the first place.  It took a render for this ASCII Table model from 13.5 min (CGAL) down to 5 seconds (Manifold)!

[image]

This is awesome, much love to all the devs that made it happen.