openscad/openscad #5053: Union of closed shapes fails

w1ebr, 2024-03-18:
**Describe the bug**
When the program tries to merge two closed shapes it says "ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron."

**To Reproduce**
Run the program (F6)

**Environment and Version info (please complete the following information):**
 - OS: macOS Sonoma
 - System: M3 Pro
 - OpenSCAD Version 2023.3.17 from OpenSCAD website
[Crocus.zip](https://github.com/openscad/openscad/files/14640288/Crocus.zip)

thehans, 2024-03-18:
SVG line drawings like this often contain self intersections, which result in non-manifold geometry after extrusion.  
You can correct for this by wrapping the import with a miniscule offset.
```
... offset(delta=0.01) import("drawing2.svg");
```

kintel, 2024-03-18:
Also, try to enable `Preferences->Features->manifold` - this new geometry backend is quite a bit more robust.

w1ebr, 2024-03-19:
Manifold said the results were not manifold :-)
Offsetting it by a small amount works.
The issue (I think) is that the SVG is the result of a trace operation in Inkscape and there are self-intersecting Bézier curves in the vector image.

kintel, 2024-03-19:
With manifold enabled, your design renders perfectly on OpenSCAD-2024.03.17 on an M1 Mac, without any offset needed.

w1ebr, 2024-03-19:
Were you able to get the Union() to work also? I am running the same version and it will create the pattern (stained glass frame) fine, but doesn't want to perform a union with the filled version of th same shape.

w1ebr, 2024-03-19:
Thanks for all of the help

kintel, 2024-03-22:
The union worked fine after enabling manifold. Are you still having a problem with that?

w1ebr, 2024-03-22:
No more problem, thank you!

reedy, 2024-04-08:
Thanks!

I was having a similar problem with OpenSCAD-2024.04.05 using https://github.com/scottbez1/splitflap/blob/master/3d/tools/flap_container.scad (it's got another issue which I've fixed in https://github.com/scottbez1/splitflap/pull/235)

> Also, try to enable `Preferences->Features->manifold` - this new geometry backend is quite a bit more robust.

For anyone else falling into the same issue and stumbled upon this task, this indeed fixed the problem. But I did have to restart OpenSCAD for this to take effect.

I don't know if this is intended, but wasn't completely obvious after changing the preference (no prompt etc), but it didn't make any apparent changes. Quick app restart, and it works!

kintel, 2024-04-08:
You have to restart, or do `Design->Flush caches`. We don't spend too much effort on UI or docs for experimental features, until they make it out of the experimental stage..

reedy, 2024-04-08:
> You have to restart, or do `Design->Flush caches`. We don't spend too much effort on UI or docs for experimental features, until they make it out of the experimental stage..

That's fair!

I do note that one explicitly says it needs a restart:

[image]

Is it worth adding a line of text before the list to say that a restart/cache flush may be required when enabling/disabling some of these options?

t-paul, 2024-04-08:
I agree for "normal" settings. We are talking here about special **development** settings that are **not** available in the release version. Polishing those does not make too much sense.

closed 2026-07-02 by github-actions[bot]