openscad/openscad #5058: Building on Mac OS X Using Homebrew Fails due to Qscintilla2 Versioning

cversek, 2024-03-21:
**Describe the bug**
When following the build step:
```
./scripts/macosx-build-homebrew.sh
```
Followed by:
```
 mkdir build && cd build && cmake .. -DEXPERIMENTAL=1    
```
The `cmake` script gives the following error:
```
CMake Error at /opt/homebrew/Cellar/cmake/3.28.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Qt5QScintilla (missing: QT5QSCINTILLA_LIBRARY) (found
  suitable version "2.14.1", minimum required is "2.8.0")
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.28.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindQt5QScintilla.cmake:112 (find_package_handle_standard_args)
  CMakeLists.txt:543 (find_package)

-- Configuring incomplete, errors occurred!
```

**Environment and Version info (please complete the following information):**
 - OS: macOS 14.4 Sonoma
 - System: Apple Silicon M3 Max
 - OpenSCAD Version - Latest Source from Github
 
**Additional context**
According to  `brew info qscintilla2`:
```
==> qscintilla2: stable 2.14.1 (bottled)
Port to Qt of the Scintilla editing component
https://www.riverbankcomputing.com/software/qscintilla/intro
/opt/homebrew/Cellar/qscintilla2/2.14.1_3 (155 files, 8.3MB) *
  Poured from bottle using the formulae.brew.sh API on 2024-03-18 at 11:39:41
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/q/qscintilla2.rb
License: GPL-3.0-only
==> Dependencies
Build: pyqt-builder ✘, sip ✘
Required: pyqt ✔, python@3.12 ✔, qt ✔
```
The package `qt` is a dependency, but that package version via `brew info qt` is:
```
==> qt: stable 6.6.2 (bottled), HEAD
```
thus libraries for the `qt5` version of `qscintilla2` are missing.
The breaking change appears to have been introduced in this commit to `homebrew`: https://github.com/Homebrew/homebrew-core/commit/ce9bebfee9fa353602dc637bdc70a2d804aa3e77

I don't have a workaround yet but I expect that it involves either installing `qscintilla2@2.13.1` and somehow getting the build to use that instead of the head's version 2.14.1 OR maybe trying to build everything against qt6?  Either way I am very new to `homebrew` and compiling on OSX in general.

kintel, 2024-03-21:
See also https://github.com/Homebrew/homebrew-core/pull/166414.
The ideal short-term workaround would be to get a qt5 version of QScintilla added to Homebrew.

The workaround for building OpenSCAD from source is to build the dependencies using https://github.com/openscad/openscad/blob/master/scripts/macosx-build-dependencies.sh

Out of curiosity, are you building manually vs. downloading the nightly builds for a reason (other than wanting to) ?

cversek, 2024-03-21:
@kintel 
Yeah I know about the builds now, and it does work for me.  I guess I just wanted to see if it would easy to compile locally, and then I did some extra digging into the failure so that I could be at least a little helpful :)

kintel, 2024-03-21:
Sounds good, let’s leave this ticket open to track this.

Btw., building does from source using the above script should be fully automatic as well, and any problems with it are reportable as well.

kintel, 2024-03-22:
Btw., this also breaks our CI, so it's pretty high priority to fix. I just keep hoping someone will pick up the Homebrew issue so I don't have to do it :)

kintel, 2024-03-22:
Workaround:
```
curl -o qscintilla2.rb https://raw.githubusercontent.com/Homebrew/homebrew-core/da59bcdf7f1dadf70e30240394ddc0bd6014affe/Formula/q/qscintilla2.rb
brew unlink qscintilla2
brew install qscintilla2.rb
```

cversek, 2024-03-22:
@kintel 
So I pulled the latest changes to try out your workaround.  The build environment seems ok, but there is some noisy warnings and errors on my end, which may just be artifacts of previous attempts:
```
% ./scripts/macosx-build-homebrew.sh 
Fri Mar 22 02:43:06 EDT 2024: Updating homebrew
==> Updating Homebrew...
Warning: No remote 'origin' in /opt/homebrew/Library/Taps/cversek/homebrew-local-tap, skipping update!
==> Updated Homebrew from 7b2bfee363 to d704e007c7.
Updated 1 tap (homebrew/core).
==> New Formulae
ingress2gateway                                                   rtabmap
==> Outdated Formulae
cmake                                                             libomp

You have 2 outdated formulae installed.
You can upgrade them with brew upgrade
or list them with brew outdated.

real	0m4.666s
user	0m1.139s
sys	0m1.465s
Fri Mar 22 02:43:11 EDT 2024: Listing homebrew configuration
HOMEBREW_VERSION: 4.2.13-96-gd704e00
ORIGIN: https://github.com/Homebrew/brew
HEAD: d704e007c7a40f97d1ed3005be4539cc2cd52ee4
Last commit: 5 hours ago
Core tap HEAD: 6b9f22abafb887533d9dc5dd06f604b52ba7c24e
Core tap last commit: 22 minutes ago
Core tap JSON: 22 Mar 06:43 UTC
Core cask tap JSON: 22 Mar 06:43 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_MAKE_JOBS: 16
HOMEBREW_SORBET_RUNTIME: set
Homebrew Ruby: 3.1.4 => /opt/homebrew/Library/Homebrew/vendor/portable-ruby/3.1.4/bin/ruby
CPU: 16-core 64-bit arm_palma
Clang: 15.0.0 build 1500
Git: 2.39.3 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Curl: 8.4.0 => /usr/bin/curl
macOS: 14.4-arm64
CLT: 15.3.0.0.1.1708646388
Xcode: 15.1
Rosetta 2: false

real	0m0.611s
user	0m0.223s
sys	0m0.120s
Error: No such keg: /opt/homebrew/Cellar/python@2
Fri Mar 22 02:43:13 EDT 2024: Installing or updating formula boost
boost 1.84.0_1
Warning: boost 1.84.0_1 already installed
==> `brew cleanup` has not been run in the last 30 days, running now...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
Removing: /Users/cversek/Library/Caches/Homebrew/cmake--3.28.3... (16.4MB)
Removing: /Users/cversek/Library/Caches/Homebrew/libomp--18.1.1... (567.3KB)
Removing: /opt/homebrew/Cellar/librsvg/2.57.2... (22 files, 37MB)
Removing: /Users/cversek/Library/Caches/Homebrew/librsvg--2.57.2... (11.7MB)
Removing: /opt/homebrew/Cellar/pango/1.52.0... (68 files, 3.4MB)
Removing: /Users/cversek/Library/Caches/Homebrew/pango--1.52.0... (808.6KB)
Removing: /opt/homebrew/Cellar/sqlite/3.45.1... (11 files, 4.8MB)
Removing: /Users/cversek/Library/Caches/Homebrew/sqlite--3.45.1... (2.2MB)
Removing: /opt/homebrew/Cellar/xz/5.6.0... (166 files, 2.7MB)
Removing: /Users/cversek/Library/Caches/Homebrew/xz--5.6.0... (684.4KB)
Pruned 0 symbolic links and 4 directories from /opt/homebrew

real	0m3.234s
user	0m1.339s
sys	0m1.071s
Fri Mar 22 02:43:16 EDT 2024: Installing formula pkg-config
[... 451 lines cut ...]
/opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:561:10: error: member reference base type 'Data' (aka 'int') is not a structure or union
        d->data()[d->size] = '\0';
        ~^ ~~~~
/opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:561:20: error: member reference base type 'Data' (aka 'int') is not a structure or union
        d->data()[d->size] = '\0';
                  ~^ ~~~~
/opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:698:9: error: unknown type name 'QStringDataPtr'
        QStringDataPtr dataPtr = { fromLatin1_helper(str, (str && size == -1) ? int(strlen(str)) : size) };
        ^
/opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:952:37: error: 'QString::Data' (aka 'int') is not a class, namespace, or enumeration
    inline QString(const Null &): d(Data::sharedNull()) {}
                                    ^
/opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:955:46: error: 'QString::Data' (aka 'int') is not a class, namespace, or enumeration
    inline bool isNull() const { return d == Data::sharedNull(); }
                                             ^
/opt/homebrew/opt/qt@5/lib/QtCore.framework/Headers/qstring.h:1067:11: error: member reference base type 'Data' (aka 'int') is not a structure or union
{ return d->size; }
         ~^ ~~~~
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/OpenSCAD.dir/OpenSCAD_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [CMakeFiles/OpenSCAD.dir/all] Error 2
make: *** [all] Error 2
```
Not quite sure what to make of these errors in the `qt@5` package core library.

cversek, 2024-03-22:
@kintel 

I had a feeling that the wrong version of `qt` was being pulled into the build, so I purged `qt6` from my `brew` environment and the compile seems to work now.

Complete commands with extra qt6 removal steps:
```
./scripts/macosx-build-homebrew.sh 
brew uninstall pyqt
brew uninstall qt
mkdir build && cd build && cmake .. -DEXPERIMENTAL=1
make
ctest
```

I'm currently running `ctest` I'll post back with any further errors.
I'm not sure what the right steps should be for someone who uses the same `brew` environment for compiling with qt6 and qt5 (not a concern for me).  I'm just posting these problems for anyone else who runs into the same issue.

cversek, 2024-03-22:
Not bad, only 2 tests failed:
```
99% tests passed, 2 tests failed out of 1624

Total Test time (real) = 1096.26 sec

The following tests did not run:
	313 - cgalpngtest_nothing-decimal-comma-separated (Disabled)
	336 - cgalpngtest_background-modifier2 (Disabled)
	337 - cgalpngtest_child-background (Disabled)
	350 - cgalpngtest_highlight-and-background-modifier (Disabled)
	352 - cgalpngtest_highlight-modifier2 (Disabled)
	355 - cgalpngtest_import-empty-tests (Disabled)
	361 - cgalpngtest_linear_extrude-scale-zero-tests (Disabled)
	363 - cgalpngtest_minkowski3-difference-test (Disabled)
	397 - cgalpngtest_testcolornames (Disabled)
	465 - cgalpngtest_nef3_broken (Disabled)
	474 - cgalpngtest_empty-shape-tests (Disabled)
	560 - opencsgtest_child-background (Disabled)
	648 - opencsgtest_issue1165 (Disabled)
	651 - opencsgtest_issue1215 (Disabled)
	671 - opencsgtest_issue3158 (Disabled)
	689 - opencsgtest_nef3_broken (Disabled)
	806 - throwntogethertest_minkowski3-erosion (Disabled)
	811 - throwntogethertest_nullspace-difference (Disabled)
	839 - throwntogethertest_testcolornames (Disabled)
	859 - throwntogethertest_issue1089 (Disabled)
	870 - throwntogethertest_issue1215 (Disabled)
	885 - throwntogethertest_issue1803 (Disabled)
	908 - throwntogethertest_nef3_broken (Disabled)
	919 - throwntogethertest_internal-cavity (Disabled)
	920 - throwntogethertest_internal-cavity-polyhedron (Disabled)
	997 - csgpngtest_background-modifier2 (Disabled)
	998 - csgpngtest_child-background (Disabled)
	1011 - csgpngtest_highlight-and-background-modifier (Disabled)
	1013 - csgpngtest_highlight-modifier2 (Disabled)
	1022 - csgpngtest_linear_extrude-scale-zero-tests (Disabled)
	1024 - csgpngtest_minkowski3-difference-test (Disabled)
	1038 - csgpngtest_primitive-inf-tests (Disabled)
	1058 - csgpngtest_testcolornames (Disabled)
	1097 - csgpngtest_issue1258 (Disabled)
	1125 - csgpngtest_transform-nan-inf-tests (Disabled)
	1126 - csgpngtest_nef3_broken (Disabled)
	1373 - dxfpngtest_nullspace-2d (Disabled)
	1383 - dxfpngtest_text-empty-tests (Disabled)
	1414 - dxfpngtest_nothing-decimal-comma-separated (Disabled)
	1440 - svgpngtest_nullspace-2d (Disabled)
	1450 - svgpngtest_text-empty-tests (Disabled)
	1616 - openscad-viewoptions-edges_view-options-tests (Disabled)
	1617 - openscad-viewoptions-axes-scales-edges_view-options-tests (Disabled)
	1618 - openscad-viewoptions-wireframe_view-options-tests (Disabled)
	1653 - relative-output_csg_run (Disabled)
	1654 - relative-output_csg_check (Disabled)
	1665 - relative-output_ast_run (Disabled)
	1666 - relative-output_ast_check (Disabled)

The following tests FAILED:
	1270 - pdfexporttest_centered (Failed)
	1271 - pdfexporttest_simple-pdf (Failed)
```

kintel, 2024-03-22:
I think the PDF checks are false positives - it's probably due to the conversion from PDF to PNG for comparison (ghostscript updated in Homebrew; I get the same error locally).

haikusw, 2024-04-28:
I'm running into a build failure on macOS X (v13.6.6) on Apple Silicon due to qscintilla2:

Building using:
> % mkdir build && cd build && cmake .. -DEXPERIMENTAL=1
>...
>-- Qt5: 5.15.13
> CMake Error at /opt/homebrew/Cellar/cmake/3.29.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
>   Could NOT find Qt5QScintilla (missing: QT5QSCINTILLA_LIBRARY
>   QT5QSCINTILLA_INCLUDE_DIR) (Required is at least version "2.8.0")
> Call Stack (most recent call first):
>   /opt/homebrew/Cellar/cmake/3.29.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
>   cmake/Modules/FindQt5QScintilla.cmake:112 (find_package_handle_standard_args)
>   CMakeLists.txt:543 (find_package)

As per the ReadMe.md I first ran but got an Error related to qscintilla2.rb
> % ./scripts/macosx-build-homebrew.sh
> ...
> real	0m0.889s
> user	0m0.584s
> sys	0m0.122s
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100  4414  100  4414    0     0  24646      0 --:--:-- --:--:-- --:--:-- 24659
> Unlinking /opt/homebrew/Cellar/qscintilla2/2.14.1_2... 0 symlinks removed.
> Error: Failed to load cask: qscintilla2.rb
> Cask 'qscintilla2' is unreadable: wrong constant name #<Class:0x0000000108f7e618>
> Warning: Treating qscintilla2.rb as a formula.
> Warning: qscintilla2 2.14.1_2 is already installed, it's just not linked.
> To link this version, run:
>   brew link qscintilla2

which seems likely to be the root issue.  I don't know how to fix this; the workaround in the ./scripts/macosx-build-homebrew.sh doesn't seem to be working for me.

----
I tried building dependencies from source as per option 1 in the ReadMe.md instructions via:

```
source scripts/setenv-macos.sh
./scripts/macosx-build-dependencies.sh
```
but that got an error relating to `aclocal` during `hdiapi-hidapi-0.12.0` build:

> Building hidapi 0.12.0 ...
> Archive:  hidapi-0.12.0.zip
> 76108294092c023a4ece99eb3219559cea0d5066
>    creating: hidapi-hidapi-0.12.0/
> …
>   inflating: hidapi-hidapi-0.12.0/windows/hidtest.vcproj  
>   inflating: hidapi-hidapi-0.12.0/windows/hidtest.vcxproj  
>   inflating: hidapi-hidapi-0.12.0/windows/hidtest.vcxproj.filters  
> + autoreconf --install --verbose --force
> autoreconf: export WARNINGS=
> autoreconf: Entering directory '.'
> autoreconf: configure.ac: not using Gettext
> autoreconf: running: aclocal --force -I m4
> Can't exec "aclocal": No such file or directory at /opt/homebrew/Cellar/autoconf/2.72/share/autoconf/Autom4te/FileUtils.pm line 299.
> autoreconf: error: aclocal failed with exit status: 2
> 

So I'm unable to build OpenScad on macOS (v13.6.6) using either option for getting the dependencies installed.

Was going to try looking into fixing a bug but sort of need to get it building first :)

kintel, 2024-04-28:
Option 1 (dependencies via homebrew) is temporarily a bit annoying since Homebrew upgraded qscintilla to Qt6 without making a Qt5 version available. You could try this:
```
brew uninstall --force qscintilla2
./scripts/macosx-build-homebrew.sh
```

Option 2 (locally built dependencies): You probably didn't install automake (brew install automake), as mentioned in build prerequisites at https://github.com/openscad/openscad?tab=readme-ov-file#building-for-macos

haikusw, 2024-04-29:
Thank you.  I installed the build prerequisites but it seems I missed this one (sigh).  Sorry to trouble you.    Still won't build.  Too many hours just to get it to build and too much pain; moving on to something else.

PR #5468 (2024-12-03): Make Qt6 default for macOS builds
Also fixes #5058
closed 2024-12-03 by kintel