openscad/openscad #5060: Poor transparency calculation for previewing extruded hollow shapes with holes cut into them

MethylBromide, 2024-03-22:
**Describe the bug**
When an extruded 2D shape with internal holes is perforated via the "difference" function, large portions of the remaining surface vanish from preview.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new SCAD file.
2. Paste in the below code and execute. The script creates two objects -- one is an extruded flat shape with an interior hole, and the other is the same shape with slots cut into it via the difference function.
3. Move the perspective around to see the issue. Depending on angle of view, large portions of the surface become transparent.

**Expected behavior**
This is a simple enough shape I would expect the preview and the rendered version to look nearly identical.

**Code reproducing the issue**
`/* [Hidden] */
$fa = 1;
$fs = 0.4;
for (slots=[false, true])
    translate([slots?40:0,0,0])
    difference() {
        linear_extrude(height=18) {
            difference() {
                offset(r=4) polygon([[15,0], [0,15], [-15,0], [0,-15]]);
                offset(r=3) polygon([[15,0], [0,15], [-15,0], [0,-15]]);
        }   }
        if (slots) {
            translate([0,0,-1]) cylinder(17, r=9);
            for(z=[1:3:13])
                translate([12,0,z]) cube([12,12,1], center=true);
        }
    }
`

**Screenshots**
[image]
[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11.
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2024.03.22 (git 968835698) from OpenSCAD website.

**Additional context**
This is severely impacting my ability to create complex models to post on Thingiverse for use with the Customizer, because huge swaths of my model turn into gaping holes in the preview. People can't see what they're getting.

UBaer21, 2024-03-22:
`linear_extrude(height=18, convexity=5) `
Add the convexity parameter for correct preview display.

t-paul, 2024-03-22:
See also https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#preview-convexity

closed 2024-03-22 by t-paul