openscad/openscad #5061: Is this the right place to ask question about OpenSCAD

dbyy, 2024-03-22:
Hi,
I just started using OpenSCAD and wondering whether is this right place to ask questions about OpenSCAD?
Can this be used like a 'discussions forum' or is there somewhere else a better place for that?
e.g. 
I'm using OpenSCAD v 2101.01 as flatpak and the scripts I'm saving having a weird name like '.xdp_mymodel.MJVNK2' ???
What do I do wrong?
Thanks for answering my question.

kintel, 2024-03-22:
This is for reporting bugs. For discussions and questions, please use the mailing list or IRC: https://openscad.org/community.html

Leaving this open for a bit in case it's a symptom of a real problem, so the people familiar with flatpak has a chance to comment.

closed 2024-03-22 by dbyy

t-paul, 2024-03-22:
Please include more information when posting there. I just tried with the Flatpak release version (2021.01) and it behaves just normal saving a file in a subfolder of the home directory.

Are you using the Release or Beta version? What's the exact steps and in which folder is the file saved?

dbyy, 2024-03-23:
Hello,
It seems to me it's the 'stable' flatpak version 2101.01 ("flatpak list").
Help > About only says it's version 201.01.
I saved the script on a mapped NAS drive (/mnt/nas_documents/).
On the other side, when I created a (dummy) script on the NAS location and then opened it in OpenSCAD, edited and saved it again, it's working properly.
Later today:
I went back and tried the same procedure again (opened OpenSCAD, created a 'dummy' script, saved it to a NAS location) and this time it was working just fine.
I'm not sure what happened earlier today what I have described in my initial posting. But I would say it is working now.
I apologize for the confusion I created :-( 
Thank you.