openscad/openscad #5065: Display colors when rendering using F6

kintel, 2024-03-24:
Currently, `color()` can be used to set the color of (both negative and positive) objects, but only works in Preview Mode (F5).

Example:
```
difference() {
    union() {
      color("darkred") cube(20);
      color("darkgreen" ) sphere(10);
    }
    color("cyan") cylinder(r=8, h=30);
}
```
[image]

It would be great if this was extended to also support Render Mode (F6).

Using Manifold, this should be possible, as demoed by https://manifoldcad.org/#Tetrahedron%20Puzzle.
This feature is likely a generalization of #5047

gsohler, 2024-03-24:
My work on that is stuck currently, as
the order of the input sets described in runIndex is not predictable for me
in my place even documentation states, that
they are sorted by originalID.
sometimes their order is A, B, sometimes th eorder is  B,A and I did not
follow the approach by matching the sets by triangle count ;)

hope to get the right inspiration soon.

ps the example given matching by triangle count would work well as its
unlikely that green, blue and red have same counts.

kintel, 2024-03-24:
Any dependency on implicit ordering is likely to fail, as it any dependency on likely triangle count of child nodes. Why not just map from originalID, as these IDs are already available in MeshGL?

yahbluez, 2024-04-09:
Would like to see that feature too.

ochafik, 2024-06-14:
> Using Manifold, this should be possible, as demoed by https://manifoldcad.org/#Tetrahedron%20Puzzle. This feature is likely a generalization of #5047

Manifold supports per-vertex colors, not per-face, which leads to color blends on faces that get split during csg ops (cc/ @elalish)

Besides, there are bits of the rendering (e.g. minkowski) where we convert manifolds to PolySet and back, so we'd need to preserve colors in those instances.

A costly but maybe simpler hacky alternative would be to do per-color rendering of the model:
- Collect all colors contained in each subtree
- Do top-level rendering for each global color
    - Skip nodes that don't contain said color
    - Node transforms to ensure proper color handling, e.g. `union() { color(ca) a(); color(cb) b()};` -> `color(ca) { intersection() { a(); b(); } difference() { a(); b() }} color(cb) { difference() { b(); a(); }}` (first color in union wins the common parts)
- Assemble top-level nodes into an [assimp](https://github.com/assimp/assimp) scene node each w/ their color -> gives glb, 3mf color support for free

elalish, 2024-06-14:
>Manifold supports per-vertex colors, not per-face, which leads to color blends on faces that get split during csg ops (cc/ @elalish)

Not true! In fact, Manifold just has generic property support - nothing specific to color, so it's up to you how to use them. But it does have a very flexible and powerful API that you can use for applying colors in many ways.

1. Manifold can have multiple property sets per vertex, which is what allows us to do Boolean operations on meshes with properties *without* creating incorrect blends. 
2. Face colors can be created using this same method, but more likely you actually want swaths of faces with the same color, often due to an entire input mesh having a constant color. Manifold can handle this without any properties by using our `OriginalID` - we keep track of the runs of triangles that are from each separate input mesh, so those can easily be put into separate draw calls with different rendering materials / colors if you keep track of the mapping of `OriginalID` to material. We have a simple [example](https://manifoldcad.org/three) of this and you can look at the source in Chrome dev tools.

kintel, 2024-06-15:
Thanks for the demo! I read through the important part of the code when I first opened this ticket and I believe it will be doable to track OriginalID from our PolySet class. I ended up down-prioritizing this though, as getting Manifold ready for release is better use of time right now than adding more features.

gsohler, 2024-06-15:
I was also exploring the OriginalID property during Unioning 3 shifted
cubes in 3 different colors.
But i had severe issues to correctly map back the originalID property of
the result set  back to a color.

The Ratio in between the set of different originalID's in the result  was
simply not matching to the expected correct result.

ochafik, 2024-06-15:
I took a stab at using Manifold's OriginalID mechanism: https://github.com/openscad/openscad/pull/5180 (thanks @elalish for the explanation! also... could this be supported in 2D? would be a good incentive to switch OpenSCAD's 2D rendering to manifold too)

I threw in a dependency to assimp to support exporting to color-aware file formats. Technically 3MF / updating `export_3mf.cc` should have been enough, but PrusaSlicer doesn't do anything with the colors.

While that PR will probably be very useful for previews in the [Web Playground](https://ochafik.com/openscad2/), I think for multimaterial 3D printing we'll need distinct manifold solids of each color (hope I'm wrong haha). I've started playing with the idea I listed [above](https://github.com/openscad/openscad/issues/5065#issuecomment-2167812697).

elalish, 2024-06-15:
Great! I've considered making a more advanced `Polygon` akin to `MeshGL` vs `Mesh` that could contain ID/property info, but it's kind of low on my priority list. If anyone else is feeling inspired, PRs are welcome. 

Indeed, for multi-material printing you'll definitely want multiple manifolds that fit like jigsaw pieces - our `Split` method is ideal for this. 

@gsohler I've tested the IDs stuff extensively, so I'm pretty confident it works, but it's not always intuitive at first. If you can file an issue on Manifold with a simple example, I'll be happy to help out.

ochafik, 2024-06-16:
So, I've pushed a color-enabled WASM build of https://github.com/openscad/openscad/pull/5180 to this [web playground demo](https://ochafik.com/openscad3/#%7B%22params%22%3A%7B%22sourcePath%22%3A%22%2Fplayground.scad%22%2C%22source%22%3A%22%2F*%5Cn%20%20Hello%20there!%5Cn%5Cn%20%20If%20you're%20new%20to%20OpenSCAD%2C%20please%20learn%20the%20basics%20here%3A%5Cn%20%20https%3A%2F%2Fopenscad.org%2Fdocumentation.html%5Cn%5Cn%20%20There%20are%20lots%20of%20amazing%20libraries%20in%20the%20OpenSCAD%20ecosystem%5Cn%20%20(see%20this%20list%3A%20https%3A%2F%2Fopenscad.org%2Flibraries.html).%5Cn%5Cn%20%20Some%20of%20these%20libraries%20are%20bundled%20with%20this%20playground%5Cn%20%20(search%20for%20%5C%22demo%5C%22%20or%20%5C%22example%5C%22%20in%20the%20file%20explorer%20above)%5Cn%20%20and%20can%20be%20included%20directly%20from%20your%20models.%5Cn%5Cn%20%20Any%20bugs%20(this%20is%20an%20Alpha!)%20or%20ideas%20of%20features%3F%5Cn%20%20https%3A%2F%2Fgithub.com%2Fopenscad%2Fopenscad-playground%2Fissues%2Fnew%5Cn*%2F%5Cn%5Cn%2F%2F%20Click%20on%20Render%20or%20hit%20F6%20to%20do%20a%20fine-grained%20rendering.%5Cn%24fn%3D%24preview%20%3F%2020%20%3A%20100%3B%5Cn%5Cncolor(%5C%22red%5C%22)%5Cntranslate(%5B-24%2C0%2C0%5D)%20%7B%5Cn%20%20union()%20%7B%5Cn%20%20%20%20cube(15%2C%20center%3Dtrue)%3B%5Cn%20%20%20%20sphere(10)%3B%5Cn%20%20%7D%5Cn%7D%5Cn%5Cncolor(%5C%22green%5C%22)%5Cnintersection()%20%7B%5Cn%20%20cube(15%2C%20center%3Dtrue)%3B%5Cn%20%20sphere(10)%3B%5Cn%7D%5Cn%5Cncolor(%5C%22blue%5C%22)%5Cntranslate(%5B24%2C0%2C0%5D)%20%7B%5Cn%20%20difference()%20%7B%5Cn%20%20%20%20cube(15%2C%20center%3Dtrue)%3B%5Cn%20%20%20%20sphere(10)%3B%5Cn%20%20%7D%5Cn%7D%5Cn%5Cncolor(%5C%22black%5C%22)%5Cntranslate(%5B0%2C%20-30%2C%20-12%5D)%5Cn%20%20linear_extrude(1)%5Cn%20%20%20%20text(%5C%22OpenSCAD%20Playground%5C%22%2C%20halign%3D%5C%22center%5C%22%2C%20valign%3D%5C%22center%5C%22)%3B%5Cn%5Cn%22%2C%22features%22%3A%5B%22manifold%22%2C%22fast-csg%22%2C%22lazy-union%22%2C%22assimp%22%5D%2C%22renderFormat%22%3A%22glb%22%7D%2C%22view%22%3A%7B%22layout%22%3A%7B%22mode%22%3A%22multi%22%2C%22editor%22%3Atrue%2C%22viewer%22%3Atrue%2C%22customizer%22%3Afalse%7D%2C%22color%22%3A%22%23f9d72c%22%2C%22showAxes%22%3Atrue%2C%22showShadows%22%3Atrue%7D%7D).

I uses the fast face coloring / Manifold OriginalID mechanism for previews, and my experimental hacky CSG tree rewrites for solids (only glTF (.glb) downloads on the demo so far, sorry, I think I stripped it out of the assimp wasm build)

It also uses https://modelviewer.dev/ for glTF rendering, which mean we get automatic AR rendering of rendered models (at least on my iPhone). cc/ @elalish 🎉👌

On a local build of the branch, w/ `--enable=manifold --enable=assimp --enable=color-solids`, it outputs 3MF files that BambuSlicer recognizes as multiple objects (one per color), each of which can be assigned a material.

ochafik, 2024-06-19:
Did a tiny bit of digging re/ outputting surface colors and letting slicers turn them into multimaterial prints. PrusaSlicer (& BambuStudio) store the [multimaterial face painting](https://help.prusa3d.com/article/multi-material-painting_262620) data in each triangle of their model. Each `.3mf` is a ZIP file containing an XML `3D/3dmodel.model` like this:

[image]

They [store](https://github.com/prusa3d/PrusaSlicer/blob/8601d829d799084511e9daa1056b2051ff955a45/src/libslic3r/Format/3mf.cpp#L1978) a serialized attribute `slic3rpe:mmu_segmentation` (or `paint_color` [for compat.](https://github.com/prusa3d/PrusaSlicer/issues/12502) w/ BambuStudio) which is deserialized in [FacetsAnnotation::set_triangle_from_string](https://github.com/prusa3d/PrusaSlicer/blob/8601d829d799084511e9daa1056b2051ff955a45/src/libslic3r/Model.cpp#L2112) and [TriangleSelector::TriangleSplittingData::update_used_states](https://github.com/prusa3d/PrusaSlicer/blob/8601d829d799084511e9daa1056b2051ff955a45/src/libslic3r/TriangleSelector.cpp#L1700). It all seems extremely complicated but the resulting encoding is very simple when the face is fully colored w/ a single material, so should be straightforward to generate multimaterial 3mf files w/ just a tiny bit of post-processing of the 3mf (I'll probably file a feature request to assimp).

ochafik, 2024-07-02:
FYI the upcoming version of the web demo now has full color support (and [full Customizer UI](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Customizer) as icing on the cake):

https://ochafik.com/openscad3

I've implemented a crude color -> extruder color mapping for the export to 3MF. It results in multimaterial files in BambuStudio / PrusaSlicer!

[image]

[image]

It will take a while to upstream the changes (see https://github.com/openscad/openscad/pull/5185) but would be great to get initial feedback!

kintel, 2024-07-20:
Closing as this is technically implemented now.
We can open new tickets for further evolutions like additional file format support, 2D colors, volumetric colors etc.

closed 2024-07-20 by kintel