openscad/openscad #5067: Customizer Textbox support

ThePilcrow, 2024-03-25:
Stated is in the Customizer manual section "The text box example only works in release version 2021.01, it may not work in future versions!"

Indeed in the OpenSCAD-2024.03.23 Development Snapshots version this option is no longer working.
The result is that certain options, that is a vector with more than 4 elements, can no longer be configured.

The problem, or rather the bug, is that there is no alternative way to configure a vector with more than 4 elements.
Hence, at least my design( https://www.printables.com/model/294305-storage-box-generator-v100 ) can no longer be used!

Thus, provide a way to configure/use vectors with more than 4 elements in the Customizer.

jordanbrown0, 2024-03-25:
This is a subset of #3817, but since it might be simpler than the full nested-vector support requested there I'll leave it open.

gchwelos, 2024-12-02:
+1 for similar purpose.