openscad/openscad #5070: Does not see user color schemes (Mac OS 13.6.4)

ron4mac, 2024-03-27:
**Describe the bug**
Preferences does not offer color scheme .json that is placed in Application Support/OpenSCAD/color-schemes folders.

**To Reproduce**
Steps to reproduce the behavior:
1. Place a color scheme .json file in what should be the appropriate place in ~Library/Application Support/OpenSCAD/color-schemes/render
2. and/or place in Library/Application Support/OpenSCAD/color-schemes/render
3. start the OpenSCAD application
4. See that no additional color schemes are offered in the preferences

**Expected behavior**
Additional color schemes should be available for selection and use.

**Environment and Version info (please complete the following information):**
 - OS: macOS 13.6.4 Ventura
 - System: Apple M1 silicon
 - OpenSCAD Version 2024.03.04 (git 7dda63a4d) installed from OpenSCAD.org

kintel, 2024-03-27:
I can reproduce - will look into it

closed 2024-03-28 by kintel

ron4mac, 2024-03-28:
👍🏻

PR #5071 (2024-03-28): Bugfix: Color schemes no loading due to uninitialized pointer
Fixes #5070