openscad/openscad #5072: Intersection not working, creating union instead

xypron, 2024-04-01:
**Describe the bug**
The OpenSCAD manual has this example:
`intersection(){cube(12, center=true); sphere(8);}`
The rendered image is the union not the intersection

**To Reproduce**
[OpenSCAD](https://www.openscad.org/) version 2021.01 as provided by Ubuntu 24.04.

Steps to reproduce the behavior:
1. Add $fn = 100;intersection(){cube(12, center=true); sphere(8);}
2. Click on 'F5'
3. Observe that the union of cube and spehere are displayed.

**Expected behavior**
An intersection should be displayed.

**Code reproducing the issue**
$fn = 100;intersection(){cube(12, center=true); sphere(8);}

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 22.04
 - System: amd64
 - OpenSCAD Version 2021.01 as provided by Ubuntu 24.04.

**Library & Graphics card information**
OpenSCAD Version: 2021.01
System information: Linux 6.8.0-20-generic #20-Ubuntu SMP PREEMPT_DYNAMIC Mon Mar 18 11:23:20 UTC 2024 x86_64 Ubuntu Noble Numbat (development branch) 8 CPUs 13.52 GB RAM
User Agent: OpenSCAD/2021.01 (Linux x86_64; Ubuntu Noble Numbat (development branch))
Compiler: GCC "13.2.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.12
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.79.3
lodepng version: 20180910
libzip version: 1.7.3
fontconfig version: 2.15.0
freetype version: 2.13.2
harfbuzz version: 8.3.0
cairo version: 1.18.0
lib3mf version: 1.8.1
Application Path: /usr/bin
Documents Path: /home/zfsdt/.local/share
User Documents Path: /home/zfsdt/Documents
Resource Path: /usr/share/openscad
User Library Path: /home/zfsdt/.local/share/OpenSCAD/libraries
User Config Path: /home/zfsdt/.config/OpenSCAD
Backup Path: /home/zfsdt/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/home/zfsdt/.local/share/OpenSCAD/libraries
/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/home/zfsdt/.local/share/fonts
/usr/share/plasma/fonts
/usr/local/share/fonts
/usr/share/fonts
/var/lib/snapd/desktop/fonts
/home/zfsdt/.fonts
/usr/share/fonts/croscore
[... 417 lines cut ...]
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_NV_vdpau_interop
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

**Additional context**
n/a

xypron, 2024-04-01:
- When pressing 'F5' the union is displayed.
- When pressing 'F6' the intersection is displayed.

t-paul, 2024-04-01:
Please check you don' t have "Thrown-Together" (F12) display mode active. Try F10 (Display surface) + F5. This is also visible in the "View" menu.

xypron, 2024-04-01:
"Thrown-Together" is not active.

There seems to be a real issue in the display chain. See this example with difference:

[image]

kintel, 2024-04-01:
Duplicate of #4861: This was a bug in Mesa. Fixed in Mesa-24.0.4: https://docs.mesa3d.org/relnotes/24.0.4.html

closed 2024-04-01 by kintel