openscad/openscad #5079: Linear_extrude with 3d vector as height yields inapropriate result

gsohler, 2024-04-08:
**Describe the bug**
height parameter of linear_extrude incorrecly used

**To Reproduce**
Steps to reproduce the behavior:

Use this code:

linear_extrude(height=[0,0,10]) square(1);

It always extrudes 100 for all vector values

**Expected behavior**
Either output an error or create oblique prism 

**Code reproducing the issue**

linear_extrude(height=[0,0,10]) square(1);

**Environment and Version info (please complete the following information):**
Windows version 2023.12.28 (git bdb7b78ac) and probably many others

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

kintel, 2024-04-08:
This is because there is no type checking. If the argument isn't a number, it will fall back to the default: https://github.com/openscad/openscad/blob/dfa7855cfa83958008c50ebf5a3a701e64ff22b9/src/core/LinearExtrudeNode.cc#L92

I expect this to be the case for a lot of built-in modules.

gsohler, 2024-04-08:
interpreting the vector correctly could  open many possibilities ....

https://imgur.com/a/bJKjPAd

jordanbrown0, 2024-04-12:
Looks to me like an error that isn't being detected and reported.

commit cc546b31 (2024-04-13): backport linear_extrude with vector to upstream, fixes #5079 (#5080)
closed 2024-04-13 by kintel