openscad/openscad #5083: Splat 2D/3D functions

khionu, 2024-04-09:
**Is your feature request related to a problem? Please describe.**
Sometimes you want to use the same value for all dimensions of a transform or other operation.

**Describe the solution you'd like**
Providing a `splat` function that takes a single number and applying it to all axis. Very simply:
```scad
function splat2d(val) { [val, val] }
function splat3d(val) { [val, val, val] }
// alternatively
function splat(val, 2d=false) {
  2d ? [val, val] : [val, val, val]
}
```

**Prior art**
- Bevy Game Engine's [`Vec3`](https://docs.rs/bevy/latest/bevy/prelude/struct.Vec3.html#method.splat) and [`Vec2`](https://docs.rs/bevy/latest/bevy/prelude/struct.Vec2.html#method.splat).

t-paul, 2024-04-09:
Use swizzling:
```
v = [1, 2, 3];
echo(v.xx); // ECHO: [1, 1]
echo(v.yyy); // ECHO: [2, 2, 2]
```

closed 2024-04-09 by t-paul

khionu, 2024-04-09:
I'm struggling to find documentation on this feature (and in general, actually). It's been hard to search for things because WikiBook's search doesn't limit the scope to the user manual.

t-paul, 2024-04-09:
Looks like it's also missing in the CheatSheet. It's not in the release version, only dev snapshots have it.

Created #5084 as reminder for the documentation.

khionu, 2024-04-09:
Reflecting, I don't think swizzling helps here. I'm not starting with an existing vector, I'm trying to create a new vector from a number literal. I could see this _maybe_ being helpful if I had a set of numbers I was going to splat from, but I'd be limited to only vectors with a length of three, and so I'd have to create `ceil(N / 3)` vectors.

nophead, 2024-04-09:
Perhaps [val].xx and [val].xxx

t-paul, 2024-04-09:
`echo([3].xxxx);`
`echo([for (n = [0,2,5,7,9]) [n].xxxx]);`

The output limit for swizzling is 4 (inputs are `xyzw` and `rgba`).

khionu, 2024-04-09:
Ahh, ok. I misread the [docs on dot-access notation](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/General#Dot_notation_indexing), and inferred that it wouldn't work on other-sized vectors.

This looks like it's sufficient to not need splat functions, thank you!