openscad/openscad #5085: Setting to do full Renders in place of Preview

khionu, 2024-04-10:
**Is your feature request related to a problem? Please describe.**
I'm currently experiencing the issue in #4861, and am waiting on Nixpkgs to update mesa on unstable. I use an external editor, and having to keep going back to OpenSCAD to rerender is a bit annoying. Additionally, with the new renderer being much more performant, the full render is fast enough for the scale of projects I'm working with.

**Describe the solution you'd like**
An opt-in to change the behaviour on Preview to instead do a full Render.

nophead, 2024-04-10:
Wrap your problematic code with render() { }.

khionu, 2024-04-11:
That's decent for me right now, but if you're collaborating on something, it's less so.

pca006132, 2024-04-11:
Curious if you can just use a flake and pull the correct mesa version? That way you don't have to wait for the channel update.

khionu, 2024-04-12:
I could probably do it myself, but they're already running into things to sort out on the staging branch, so I'm not sure if I really want to try. Also, upgrading mesa has a large cascade of rebuilding, so it's not a casual swap.