openscad/openscad #5086: path_extrudes() not loved by Manifold (or even CGAL, maybe...)

butcherg, 2024-04-11:
**Describe the bug**
terminate called after throwing an instance of 'std::out_of_range'
  what():  Vec out of range
Thread 1 "openscad" received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737251305408) at .      

**To Reproduce**
$ openscad --enable=manifold --enable=lazy-union --export-format binstl -o cistern.stl  cistern.scad

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code reproducing the issue**
I've narrowed it down to path extrude operations, both BOSL2 path_sweep() and a path_extrude() I downloaded from Thingiverse

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 11, Ubuntu 22.04]
 - System: PC 64-bit, 
 - OpenSCAD Version: Windows - 2024.04.08 nightly build, Ubuntu -  master branch as of #dfa785

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.
Ubuntu, from the command line:
 ./openscad --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.04.11
System information: Linux 6.5.0-27-generic #28~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 15 10:51:06 UTC 2 x86_64 Ubuntu 22.04.4 LTS 24 CPUs 15.53 GB RAM
User Agent: OpenSCAD/2024.04.11 (Linux x86_64; Ubuntu 22.04.4 LTS)
Compiler: GCC "11.4.0" 64bit
MinGW build: No
Debug build: Yes
Boost version: 1_74
Eigen version: 3.4.0
CGAL version, kernels: 5.4, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.15.3
QScintilla version: 2.11.6
InputDrivers:
GLib version: 2.72.4
lodepng version: 20210627
libzip version: 1.7.3
fontconfig version: 2.13.1
freetype version: 2.11.1
harfbuzz version: 2.7.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /home/glenn/3DPrinting/openscad/build-linux
Documents Path: /home/glenn/.local/share
User Documents Path: /home/glenn/Documents
Resource Path: /home/glenn/3DPrinting/openscad
User Library Path: /home/glenn/.local/share/OpenSCAD/libraries
User Config Path: /home/glenn/.config/OpenSCAD
Backup Path: /home/glenn/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/glenn/.local/share/OpenSCAD/libraries
  /home/glenn/3DPrinting/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /usr/share/fonts
  /usr/local/share/fonts
  /home/glenn/.local/share/fonts
  /home/glenn/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/glenn/Library/Fonts
  /usr/share/fonts/X11
[... 420 lines cut ...]
    at /home/glenn/3DPrinting/openscad/src/geometry/manifold/ManifoldGeometry.cc:45
#27 0x00005555562d32b1 in GeometryEvaluator::applyToChildren3D (this=0x7fffffffd270, node=..., op=OpenSCADOperator::UNION)
    at /home/glenn/3DPrinting/openscad/src/geometry/GeometryEvaluator.cc:153
#28 0x00005555562d2d1a in GeometryEvaluator::applyToChildren (this=0x7fffffffd270, node=..., op=OpenSCADOperator::UNION)
--Type <RET> for more, q to quit, c to continue without paging--
    at /home/glenn/3DPrinting/openscad/src/geometry/GeometryEvaluator.cc:112
#29 0x00005555562d539b in GeometryEvaluator::visit (this=0x7fffffffd270, state=..., node=...)
    at /home/glenn/3DPrinting/openscad/src/geometry/GeometryEvaluator.cc:465
#30 0x00005555562d5703 in GeometryEvaluator::visit (this=0x7fffffffd270, state=..., node=...)
    at /home/glenn/3DPrinting/openscad/src/geometry/GeometryEvaluator.cc:506
#31 0x00005555560f991d in BaseVisitable::acceptImpl<GroupNode> (state=..., node=..., visitor=...)
    at /home/glenn/3DPrinting/openscad/src/core/BaseVisitable.h:32
#32 0x00005555560f92a5 in GroupNode::accept (this=0x2000018ab90, state=..., visitor=...) at /home/glenn/3DPrinting/openscad/src/core/node.h:107
#33 0x0000555556216e57 in NodeVisitor::traverse (this=0x7fffffffd270, node=..., state=...)
    at /home/glenn/3DPrinting/openscad/src/core/NodeVisitor.cc:30
#34 0x0000555556216d68 in NodeVisitor::traverse (this=0x7fffffffd270, node=..., state=...)
    at /home/glenn/3DPrinting/openscad/src/core/NodeVisitor.cc:20
#35 0x0000555556216d68 in NodeVisitor::traverse (this=0x7fffffffd270, node=..., state=...)
    at /home/glenn/3DPrinting/openscad/src/core/NodeVisitor.cc:20
#36 0x00005555562d25a7 in GeometryEvaluator::evaluateGeometry (this=0x7fffffffd270, node=..., allownef=true)
    at /home/glenn/3DPrinting/openscad/src/geometry/GeometryEvaluator.cc:66
#37 0x0000555555fbbe0b in do_export (cmd=..., render_variables=..., export_format=FileFormat::STL, root_file=0x20000150700)
    at /home/glenn/3DPrinting/openscad/src/openscad.cc:580
#38 0x0000555555fba924 in cmdline (cmd=...) at /home/glenn/3DPrinting/openscad/src/openscad.cc:451
#39 0x0000555555fc1236 in main (argc=8, argv=0x7fffffffe2d8) at /home/glenn/3DPrinting/openscad/src/openscad.cc:1229

thehans, 2024-04-11:
@butcherg You need to include scad code which reproduces the issue.

butcherg, 2024-04-11:
[https://github.com/butcherg/DRG_168/blob/main/scad/cistern.scad](url)

UBaer21, 2024-04-11:
If you want to make it easy to track an issue it would be great if you can provide a short code snipped that produces the issue without needing any external files.  I understand this is more work for you, but this would help the dev people to have a look without costing them much time.

kintel, 2024-04-11:
A more minimal test would indeed be great.

In the meantime: @pca006132 This is the stack trace - looks like an exception thrown when calling the Manifold constructor.
```
OpenSCAD!manifold::VecView<int>::operator[](unsigned long) (/Users/kintel/code/OpenSCAD/openscad/submodules/manifold/src/utilities/include/vec_view.h:54)
OpenSCAD!(anonymous namespace)::MarkVerts::operator()(glm::vec<3, int, (glm::qualifier)0>) (/Users/kintel/code/OpenSCAD/openscad/submodules/manifold/src/manifold/src/impl.cpp:132)
OpenSCAD!void thrust::detail::wrapped_function<(anonymous namespace)::MarkVerts, void>::operator()<glm::vec<3, int, (glm::qualifier)0> >(glm::vec<3, int, (glm::qualifier)0> const&) const (/Users/kintel/code/OpenSCAD/openscad/build/_deps/thrust-src/thrust/detail/function.h:125)
OpenSCAD!glm::vec<3, int, (glm::qualifier)0> const* thrust::system::detail::sequential::for_each<thrust::system::cpp::detail::par_t, glm::vec<3, int, (glm::qualifier)0> const*, (anonymous namespace)::MarkVerts>(thrust::system::detail::sequential::execution_policy<thrust::system::cpp::detail::par_t>&, glm::vec<3, int, (glm::qualifier)0> const*, glm::vec<3, int, (glm::qualifier)0> const*, (anonymous namespace)::MarkVerts) (/Users/kintel/code/OpenSCAD/openscad/build/_deps/thrust-src/thrust/system/detail/sequential/for_each.h:55)
OpenSCAD!glm::vec<3, int, (glm::qualifier)0> const* thrust::for_each<thrust::system::cpp::detail::par_t, glm::vec<3, int, (glm::qualifier)0> const*, (anonymous namespace)::MarkVerts>(thrust::detail::execution_policy_base<thrust::system::cpp::detail::par_t> const&, glm::vec<3, int, (glm::qualifier)0> const*, glm::vec<3, int, (glm::qualifier)0> const*, (anonymous namespace)::MarkVerts) (/Users/kintel/code/OpenSCAD/openscad/build/_deps/thrust-src/thrust/detail/for_each.inl:40)
OpenSCAD!void manifold::for_each<glm::vec<3, int, (glm::qualifier)0> const*, glm::vec<3, int, (glm::qualifier)0> const*, (anonymous namespace)::MarkVerts>(manifold::ExecutionPolicy, glm::vec<3, int, (glm::qualifier)0> const*, glm::vec<3, int, (glm::qualifier)0> const*, (anonymous namespace)::MarkVerts) (/Users/kintel/code/OpenSCAD/openscad/submodules/manifold/src/utilities/include/par.h:169)
OpenSCAD!manifold::Manifold::Impl::RemoveUnreferencedVerts(manifold::Vec<glm::vec<3, int, (glm::qualifier)0> >&) (/Users/kintel/code/OpenSCAD/openscad/submodules/manifold/src/manifold/src/impl.cpp:597)
OpenSCAD!manifold::Manifold::Impl::Impl(manifold::Mesh const&, manifold::Manifold::Impl::MeshRelationD const&, std::__1::vector<float, std::__1::allocator<float> > const&, bool) (/Users/kintel/code/OpenSCAD/openscad/submodules/manifold/src/manifold/src/impl.cpp:512)
OpenSCAD!std::__1::__shared_ptr_emplace<manifold::Manifold::Impl, std::__1::allocator<manifold::Manifold::Impl> >::__shared_ptr_emplace[abi:v15006]<manifold::Mesh const&, manifold::Manifold::Impl::MeshRelationD&>(std::__1::allocator<manifold::Manifold::Impl>, manifold::Mesh const&, manifold::Manifold::Impl::MeshRelationD&) (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/c++/v1/__memory/shared_ptr.h:294)
OpenSCAD!std::__1::__shared_ptr_emplace<manifold::Manifold::Impl, std::__1::allocator<manifold::Manifold::Impl> >::__shared_ptr_emplace[abi:v15006]<manifold::Mesh const&, manifold::Manifold::Impl::MeshRelationD&>(std::__1::allocator<manifold::Manifold::Impl>, manifold::Mesh const&, manifold::Manifold::Impl::MeshRelationD&) (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/c++/v1/__memory/shared_ptr.h:288)
OpenSCAD!std::__1::shared_ptr<manifold::Manifold::Impl> std::__1::allocate_shared[abi:v15006]<manifold::Manifold::Impl, std::__1::allocator<manifold::Manifold::Impl>, manifold::Mesh const&, manifold::Manifold::Impl::MeshRelationD&, void>(std::__1::allocator<manifold::Manifold::Impl> const&, manifold::Mesh const&, manifold::Manifold::Impl::MeshRelationD&) (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/c++/v1/__memory/shared_ptr.h:953)
OpenSCAD!std::__1::shared_ptr<manifold::Manifold::Impl> std::__1::make_shared[abi:v15006]<manifold::Manifold::Impl, manifold::Mesh const&, manifold::Manifold::Impl::MeshRelationD&, void>(manifold::Mesh const&, manifold::Manifold::Impl::MeshRelationD&) (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/c++/v1/__memory/shared_ptr.h:962)
OpenSCAD!manifold::Manifold::Manifold(manifold::Mesh const&) (/Users/kintel/code/OpenSCAD/openscad/submodules/manifold/src/manifold/src/manifold.cpp:150)
OpenSCAD!std::__1::__shared_ptr_emplace<manifold::Manifold const, std::__1::allocator<manifold::Manifold const> >::__shared_ptr_emplace[abi:v15006]<manifold::Mesh>(std::__1::allocator<manifold::Manifold const>, manifold::Mesh&&) (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/c++/v1/__memory/shared_ptr.h:294)
OpenSCAD!std::__1::__shared_ptr_emplace<manifold::Manifold const, std::__1::allocator<manifold::Manifold const> >::__shared_ptr_emplace[abi:v15006]<manifold::Mesh>(std::__1::allocator<manifold::Manifold const>, manifold::Mesh&&) (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/c++/v1/__memory/shared_ptr.h:288)
[... cut ...] ed_ptr.h:962)
OpenSCAD!ManifoldUtils::createManifoldFromTriangularPolySet(PolySet const&) (/Users/kintel/code/OpenSCAD/openscad/src/geometry/manifold/manifoldutils.cc:94)
OpenSCAD!ManifoldUtils::createManifoldFromPolySet(PolySet const&) (/Users/kintel/code/OpenSCAD/openscad/src/geometry/manifold/manifoldutils.cc:110)
OpenSCAD!ManifoldUtils::createManifoldFromGeometry(std::__1::shared_ptr<Geometry const> const&) (/Users/kintel/code/OpenSCAD/openscad/src/geometry/manifold/manifoldutils.cc:168)
```

butcherg, 2024-04-11:
So, I took cistern.scad and did an "elimnation by comment", and I isolated the behavior to the trim() module.  This module calls others that do various path_extrude() to make "rope" trim on the cistern top edges.  So, I worked through each module call individually, none gave anything other than a perfect Manifold render.  But, when I uncommented them all, I go the error I originally reported.  And, since I did this in the GUI, I got a dialog complaining of a critical error and the program stopped doing any rendering, preview or mesh.

For any path_extrude(), I'm getting multiples of:

```
WARNING: PolySet -> Manifold conversion failed: NotManifold
Trying to repair and reconstruct mesh..
```
and in most cases it appears the repair works.  Only for cistern.scad does it crash, of note it also crashes with CGAL.  This same script worked with as recently as a March nightly.

I apologize, I don't know how to cut this up to make a performant snippet...

kintel, 2024-04-12:
The hard crash is likely a separate issue, so let's deal with that separately.
The warning is pretty common. Since you're modeling in pretty small coordinates, it's likely an issue with very close vertices becoming the same when we convert from OpenSCAD's 64-bit float to Manifold's 32-bit float.
That shouldn't break Manifold though. Let's wait for a bit and see if we get any feedback from them.

kintel, 2024-04-12:
We optimized the Manifold interface code quite a bit around mid-March, so we probably uncovered some instability.

kintel, 2024-04-12:
@pca006132 I slimmed down the example to this:
```
cube(0.1);

union() {
  multmatrix([[0.939693, 0, -0.34202, 0], [0, 1, 0, 0], [0.34202, 0, 0.939693, 0], [0, 0, 0, 1]])
    multmatrix([[-0, 1, 0, 0], [-1, 0, 0, 0], [-0, 0, 1, 0], [0, 0, 0, 1]])
      polyhedron(
        points = [[0.00648287, -0.0113365, -0.0135], [0.00620019, -0.0107004, -0.0128815], [0.00588995, -0.0100024, -0.0123491], [0.00555644, -0.00925199, -0.0119102], [0.00520429, -0.00845965, -0.0115708], [0.00483838, -0.00763635, -0.0113357], [0.00446378, -0.00679352, -0.0112081], [0.0040857, -0.00594282, -0.0111899], [0.00370936, -0.00509605, -0.0112811], [0.00333998, -0.00426496, -0.0114807], [0.00298269, -0.00346106, -0.0117857], [0.00264244, -0.00269549, -0.0121921], [0.00232395, -0.00197888, -0.0126941], [0.00203162, -0.00132116, -0.0132848], [0.00176953, -0.000731433, -0.013956], [0.00154128, -0.00021789, -0.0146984], [0.00135006, 0.000212356, -0.0155017], [0.00119852, 0.000553338, -0.0163547], [0.00108874, 0.00080033, -0.0172458], [0.00102226, 0.000949909, -0.0181624], [0.001, 0.001, -0.0190919], [0.001, 0.001, -0.0190919], [0.0010099, 0.000977723, -0.0197123], [0.00225708, -0.0124413, -0.0135], [0.00219227, -0.0117484, -0.0128815], [0.00212114, -0.0109878, -0.0123491], [0.00204467, -0.0101702, -0.0119102], [0.00196393, -0.00930688, -0.0115708], [0.00188004, -0.00840985, -0.0113357], [0.00179415, -0.00749153, -0.0112081], [0.00170747, -0.00656464, -0.0111899], [0.00162118, -0.00564203, -0.0112811], [0.0015365, -0.0047365, -0.0114807], [0.00145458, -0.0038606, -0.0117857], [0.00137657, -0.00302647, -0.0121921], [0.00130355, -0.00224568, -0.0126941], [0.00123652, -0.00152905, -0.0132848], [0.00117643, -0.000886505, -0.013956], [0.0011241, -0.000326967, -0.0146984], [0.00108026, 0.000141812, -0.0155017], [0.00104551, 0.000513334, -0.0163547], [0.00102035, 0.000782447, -0.0172458], [0.0010051, 0.000945423, -0.0181624], [0.001, 0.001, -0.0190919], [0.001, 0.001, -0.0190919], [0.00100227, 0.000975728, -0.0197123], [0.013102, -0.0114271, -0.0135], [0.0130371, -0.0107341, -0.0128815], [0.012966, -0.00997354, -0.0123491], [0.0128895, -0.00915594, -0.0119102], [0.0128088, -0.00829264, -0.0115708], [0.0127249, -0.0073956, -0.0113357], [0.012639, -0.00647728, -0.0112081], [0.0125523, -0.00555039, -0.0111899], [0.0124661, -0.00462778, -0.0112811], [0.0123814, -0.00372225, -0.0114807], [0.0122995, -0.00284635, -0.0117857], [0.0122214, -0.00201222, -0.0121921], [0.0121484, -0.00123143, -0.0126941], [0.0120814, -0.000514798, -0.0132848], [0.0120213, 0.000127742, -0.013956], [0.011969, 0.00068728, -0.0146984], [0.0119251, 0.00115606, -0.0155017], [0.0118904, 0.00152758, -0.0163547], [0.0118652, 0.00179669, -0.0172458], [0.01185, 0.00195967, -0.0181624], [0.0118449, 0.00201425, -0.0190919], [0.0118449, 0.00201425, -0.0190919], [0.0118471, 0.00198998, -0.0197123]],
        faces = [[0, 24, 1], [1, 25, 2], [2, 26, 3], [3, 27, 4], [4, 28, 5], [5, 29, 6], [6, 30, 7], [7, 31, 8], [8, 32, 9], [9, 33, 10], [10, 34, 11], [11, 35, 12], [12, 36, 13], [13, 37, 14], [14, 38, 15], [15, 39, 16], [16, 40, 17], [17, 41, 18], [18, 42, 19], [19, 43, 20], [20, 44, 21], [21, 45, 22], [22, 23, 0], [24, 0, 23], [25, 1, 24], [26, 2, 25], [27, 3, 26], [28, 4, 27], [29, 5, 28], [30, 6, 29], [31, 7, 30], [32, 8, 31], [33, 9, 32], [34, 10, 33], [35, 11, 34], [36, 12, 35], [37, 13, 36], [38, 14, 37], [39, 15, 38], [40, 16, 39], [41, 17, 40], [42, 18, 41], [43, 19, 42], [44, 20, 43], [45, 21, 44], [23, 22, 45], [23, 47, 24], [24, 48, 25], [25, 49, 26], [26, 50, 27], [27, 51, 28], [28, 52, 29], [29, 53, 30], [30, 54, 31], [31, 55, 32], [32, 56, 33], [33, 57, 34], [34, 58, 35], [35, 59, 36], [36, 60, 37], [37, 61, 38], [38, 62, 39], [39, 63, 40], [40, 64, 41], [41, 65, 42], [42, 66, 43], [43, 67, 44], [44, 68, 45], [45, 46, 23], [47, 23, 46], [48, 24, 47], [49, 25, 48], [50, 26, 49], [51, 27, 50], [52, 28, 51], [53, 29, 52], [54, 30, 53], [55, 31, 54], [56, 32, 55], [57, 33, 56], [58, 34, 57], [59, 35, 58], [60, 36, 59], [61, 37, 60], [62, 38, 61], [63, 39, 62], [64, 40, 63], [65, 41, 64], [66, 42, 65], [67, 43, 66], [68, 44, 67], [46, 45, 68], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22], [68, 67, 66, 65, 64, 63, 62, 61, 60, 59, 58, 57, 56, 55, 54, 53, 52, 51, 50, 49, 48, 47, 46]], convexity = 1);
  multmatrix([[1, 0, 0, 0], [0, 1, 0, -0.0035], [0, 0, 1, -0.02], [0, 0, 0, 1]])
    sphere($fn = 180, r = 0.008);
}
```

pca006132, 2024-04-12:
I do think this is a bug in manifold, at least we should not crash. I am still a bit busy recently, will look into in details a bit later.

kintel, 2024-04-13:
This may be a partial duplicate of #4824 

Note: There are two separate places where Manifold throws this exception:
1. The original example in this issue is about throwing in the `Manifold` constructor.
   Throwing in the constructor can be easily caught, and we can do some surface reconstruction/quantization work to make the mesh nicer to work on. We just need to know if this is expected behavior or an indication of a Manifold bug.
2. The other is about throwing during CSG evaluation. 
    The hard crash I mentioned was because we didn't correctly catch the `Vec out of range` exception during Manifold CSG evaluation. Fixed in #5087. However, if Manifold throws during evaluation, we don't really have a good fallback any longer, due to lazy evaluation. The workaround for this is, as usual, to evaluate greedily to detect such issues early and use a fallback solution. As discussed earlier, this potentially kills some optimization opportunities.

butcherg, 2024-06-17:
II just pulled the master branch and compiled it, and manifold rendered my problematic script.  commit d3a5ac.

kintel, 2024-06-17:
Awesome. So much stuff happened in this domain since April, so I guess we must have fixed it :)

closed 2024-06-17 by kintel