openscad/openscad #5089: Add / change Metadata for 3mf export

UBaer21, 2024-04-13:
**Is your feature request related to a problem? Please describe.**
3mf file contain only "<object id="1" name="OpenSCAD Model"  which gets confusing in object lists if multiple files are opened

**Describe the solution you'd like**
Allow to set a object name and to set metadata for the default fields like 

  |Title | A title for the 3MF document
  | Designer | A name for a designer of this document
  | Description | A description of the document
  | Copyright | A copyright associated with this document
  | LicenseTerms | License information associated with this document
  | Rating | An industry rating associated with this document
  | CreationDate | The date this documented was created by a source app
  | ModificationDate | The date this document was last modified
  | Application | The name of the source application that originally created this document

optional additional customizable tags - or adding a thumbnail (automatic) would be great

https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#341-metadata

I can see 4 options to obtain this data

1. creation / modification (if file exist) data is automatically entered 
2. a pop up dialog when exporting
3.  a metatag in the openSCAD script that can be parsed to create the 3mf metadata
4.  a saved set within openscad that is used / selected to fill the data.

tcurtin, 2024-04-13:
+1 to this.  Please please!  Extra credit for the ability to set coordinates of objects when opened in BambuLab.  But just being able name objects (even numerically number instances of modules) would be almost unspeakable helpful to my current workflow!!

UBaer21, 2024-04-13:
in Regards to BambuStudio (or Slic3r / Prusa) the coordinates are within the 3mf - these programs just handle them differently. If you import the first object they are used (because it is opened as project), if they are imported additionally they are centered - they are also dropped to bed (except if imported as parts then the object is still dropped to bed) - CURA has an option to deactivate "drop" and auto arrange (keep models apart).  So this is nothing OpenSCAD could change - you need to address this with BS/Slic3r

jordanbrown0, 2024-04-13:
This is strongly related to #4977 "Ability to set solid name", probably enough that they should be considered to be duplicates.

kintel, 2024-04-13:
We may want to downscope this to support whatever is possible without changing the OpenSCAD language, as that would be a much easier feature to add.

Laserology, 2024-05-04:
A separate labeling option using ``union()`` might be a good choice for this. Example:
```scad
union("Part1") {
    Code go here...
}
union("Part2") {
    Other code go here...
}
```
Unless i don't understand the underlying core behind this, i feel that using the method described above would probably be the best implementation for such a feature, because it integrated into already existing functions that would make sense, and it doesn't interfere with existing project files.

jordanbrown0, 2024-05-04:
Indeed, adding one or more new parameters to existing modules is one of the ideas.  You'd want them to be named parameters so that they could be in addition to any existing positional parameters, and probably want #4478 objects so that you can express complex data structures.

gsohler, 2024-05-04:
Very easy option to extend openscad is a this one:

export("file="design.3mf", partnames=["part1", "part2", "part3"])
{
  part1_module();
  part2_module();
  part3_module();
}

this would call the GeometryEvaluator right during traversing export() Node
even cache would work very well ...

Sounds like it would be very easy to implement(  it there were not certain
security concerns 🤣 )

}

Laserology, 2024-05-04:
> Very easy option to extend openscad is a this one: export("file="design.3mf", partnames=["part1", "part2", "part3"]) { part1_module(); part2_module(); part3_module(); } this would call the [...]

I feel like needing to specify the file name and mapping one array to another may slow down the development process. A generic container that gets recognized when a part is being exported may be a better option, such as the ``union`` implementation i described above.

gsohler, 2024-05-04:
I believe that openscad has exactly one Top Level and changing that Is an
enormous Change. Still feel that my proposal has least Impact and Most
value.

lgarron, 2025-05-31:
I've submitted a PR to associate names with objects in `.3mf` exports: https://github.com/openscad/openscad/pull/5942