openscad/openscad #5090: Sadly, we can't add .scad files to the issues in the openSCAD repository

JamesC1, 2024-04-14:
Github has a facility to add files to an issue.  If it's an issue with code in openSCAD, then the files that we want to add may well be .scad files.

**Environment**
Github allows a limited set of file suffixes, including .txt, which it probably takes as implying particular semantics.

**Issue**
.scad, the natural name for our code files, is not on the list.  I've made a soft-link, in order to rename a file to upload it, in another ticket.

t-paul, 2024-04-14:
Can you please elaborate on what you expect the OpenSCAD team to do here? We have no control over how github works, so I don't see any possible action on our side.

UBaer21, 2024-04-14:
[Open_an_issue_with_github_or_find_a_workaronud.scad.txt](https://github.com/openscad/openscad/files/14970551/Open_an_issue_with_github_or_find_a_workaronud.scad.txt)

JamesC1, 2024-04-14:
Hello Ulrich,

Your link triggers the start of a download in my browser.  Could you
tell me what will be downloaded, please?

Torsten, there currently either is, or is not, a github feature to
support project-specific file types (or probably actually MIME types,
but I haven't learned MIME yet).  If there is, I'm asking you to use
it.  If there isn't, it's an enhancement request on github.  Would you
prefer to write it, or if not, do you wish to point me toward the
github enhancement request queue?

Regards,
James.

UBaer21, 2024-04-14:
James - you clicked on a text file containing the scad code.  This is what happens if you upload a file and others click on that file - it will be downloaded. 

You always can paste the scad code itself into the comment and format as code.

jordanbrown0, 2024-04-15:
> You always can paste the scad code itself into the comment and format as code.

And that's usually the better answer anyway, because then people can directly read it.  If it seems too large to put into a comment, then it probably **is** too large to be a reasonable demonstration of a problem.

JamesC1, 2024-04-16:
My smallest example of 3254 is 52 lines (including some whitespace).
As far as I know, that's the smallest reliable reproduction.  We got
there after a 108 line reproduction that I found by accident in the
course of doing something else.  Are they in the right length range to
paste in?

UBaer21, 2024-04-16:
When people past their library and build info - this already exceeds 300 lines.
So yes, you can paste 150 lines.  More important is that no external lib/files/fonts etc. are needed.
If you can't avoid this it can make sense to package everything into a zip container and upload - but normally a minimal case can be constructed with some effort.

pca006132, 2024-05-04:
You can put it inside `<details> ... </details>`

nomike, 2024-05-04:
For whatever strange reason, GitHub imposes a limit on the file extensions a file you upload could happen, but on the other side they allow .gz file, meaning you can upload whatever you want by just compressing it with gzip first.

This renders the whole "You are only allowed to upload the following types of files" restriction useless.

But it's to our benefit in this case.

So, just gzip the file and everything works and you don't have to mess around with weird incorrect file extensions.