openscad/openscad #5093: Current git HEAD - src/geometry/manifold/manifoldutils.h:6:10: fatal error: manifold.h: No such file or directory

zougloub, 2024-04-19:
**Describe the bug**

Git HEAD compilation failure.

I'm compiling using [Gentoo's SCM ebuild](https://github.com/gentoo/gentoo/blob/master/media-gfx/openscad/openscad-9999.ebuild)

**To Reproduce**

1. Get the code from git, resulting in:

```
openscad:
   repository:               https://github.com/openscad/openscad.git                                                                                                                                               
   at the commit:            cc546b31bd6e78c2d16dac3cc30d49c621959a71
submodules:
   repository:               https://github.com/openscad/MCAD.git                                                                                                                                                   
   at the commit:            1ea402208c3127ffb443931e9bb1681c191dacca                                                                                                                                               
   repository:               https://github.com/microsoft/mimalloc.git                                    
   at the commit:            fe61be80bf8942e764e18d469e96f8dacab44728                                     
   repository:               https://github.com/elalish/manifold           
   at the commit:            c03a5ac3adbb002ef29646648adbbebba179000d                                                                                                                                               
   repository:               https://github.com/akuukka/quickhull                                         
   at the commit:            1ffbc6f884ea1da89e104a5996cf8a726db673d5                                     
   repository:               https://github.com/wjakob/nanobind                                                                                                                                                     
   at the commit:            1a309ba444a47e081dc6213d72345a2fbbd20795                                     
   repository:               https://github.com/Tessil/robin-map                                          
   at the commit:            68ff7325b3898fca267a103bad5c509e8861144d                                     
   repository:               https://github.com/arsenm/sanitizers-cmake.git
   at the commit:            c3dc841af4dbf44669e65b82cb68a575864326bd                                     
   repository:               https://github.com/openscad/OpenCSG.git                                      
   at the commit:            059a173f161a39a4ab28a32a5b89adabd4f77370      

```

2. Configure with -DENABLE_MANIFOLD=OFF

3. Compile and get: 

```
src/geometry/manifold/manifoldutils.h:6:10: fatal error: manifold.h: No such file or directory
```

**Expected behavior**

Compilation works

**Environment and Version info (please complete the following information):**
 - OS: linux, gentoo
 - System: x86_64
 - OpenSCAD Version: latest

t-paul, 2024-04-19:
Not likely an OpenSCAD issue. All CI builds are fine and manual build just now (after fresh `git clone --recursive` - cc546b31b) had no issues either. That is on Debian/Testing though.

The mentioned file is at: `./submodules/manifold/src/manifold/include/manifold.h`

zougloub, 2024-04-19:
cmake is doing 
```
/usr/bin/c++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DCGAL_DEBUG -DCGAL_USE_CORE=1 -DCGAL_USE_GMPXX -DCGAL_USE_GMPXX=1 -DEIGEN_DONT_ALIGN -DENABLE_CAIRO -DENABLE_CGAL -DENABLE_DBUS -DENABLE_EGL -DENABLE_GLX -DENABLE_HIDAPI -DENABLE_JOYSTICK -DENABLE_LIB3MF -DENABLE_LIBZIP -DENABLE_OPENCSG -DENABLE_SPNAV -DLIB3MF_API_2 -DMI_LINK_SHARED -DOPENSCAD_DAY=19 -DOPENSCAD_MONTH=4 -DOPENSCAD_OS=\"Unix\" -DOPENSCAD_SHORTVERSION=2024.04.19 -DOPENSCAD_VERSION=2024.04.19 -DOPENSCAD_YEAR=2024 -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_DBUS_LIB -DQT_GUI_LIB -DQT_MULTIMEDIA_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_SVG_LIB -DQT_WIDGETS_LIB -DSTACKSIZE=8388608 -DUNICODE -DUSE_GLAD -DUSE_MIMALLOC -DUSE_QOPENGLWIDGET -D_REENTRANT -D_UNICODE -D__GCC -I/home/cJ/dev/media/media-gfx/openscad/build -I/home/cJ/dev/media/media-gfx/openscad -I/home/cJ/dev/media/media-gfx/openscad/build/OpenSCAD_autogen/include -I/home/cJ/dev/media/media-gfx/openscad/src -I/home/cJ/dev/media/media-gfx/openscad/src/core -I/home/cJ/dev/media/media-gfx/openscad/src/core/customizer -I/home/cJ/dev/media/media-gfx/openscad/src/ext -I/home/cJ/dev/media/media-gfx/openscad/src/ext/json -I/home/cJ/dev/media/media-gfx/openscad/src/ext/lexertl/include -I/home/cJ/dev/media/media-gfx/openscad/src/ext/libtess2/Include -I/home/cJ/dev/media/media-gfx/openscad/src/geometry -I/home/cJ/dev/media/media-gfx/openscad/src/geometry/cgal -I/home/cJ/dev/media/media-gfx/openscad/src/geometry/manifold -I/home/cJ/dev/media/media-gfx/openscad/src/glview -I/home/cJ/dev/media/media-gfx/openscad/src/glview/preview -I/home/cJ/dev/media/media-gfx/openscad/src/glview/cgal -I/home/cJ/dev/media/media-gfx/openscad/src/gui -I/home/cJ/dev/media/media-gfx/openscad/src/io -I/home/cJ/dev/media/media-gfx/openscad/src/platform -I/home/cJ/dev/media/media-gfx/openscad/src/utils -isystem /usr/include/eigen3 -isystem /usr/include/harfbuzz -isystem /usr/include/glib-2.0 -isystem /usr/lib64/glib-2.0/include -isystem /usr/include/freetype2 -isystem /usr/include/libxml2 -isystem /usr/include/hidapi -isystem /usr/include/cairo -isystem /usr/include/libpng16 -isystem /usr/include/pixman-1 -isystem /usr/include/lib3mf/Bindings/Cpp -isystem /usr/include/qt5 -isystem /usr/include/qt5/QtCore -isystem /usr/lib64/qt5/mkspecs/linux-g++ -isystem /usr/include/qt5/QtWidgets -isystem /usr/include/qt5/QtGui -isystem /usr/include/qt5/QtMultimedia -isystem /usr/include/qt5/QtNetwork -isystem /usr/include/qt5/QtOpenGL -isystem /usr/include/qt5/QtConcurrent -isystem /usr/include/qt5/QtSvg -isystem /usr/include/qt5/QtDBus -frounding-math -std=c++17 -fPIE -frounding-math -fPIC -MD -MT CMakeFiles/OpenSCAD.dir/src/geometry/GeometryUtils.cc.o -MF CMakeFiles/OpenSCAD.dir/src/geometry/GeometryUtils.cc.o.d -o CMakeFiles/OpenSCAD.dir/src/geometry/GeometryUtils.cc.o -c /home/cJ/dev/media/media-gfx/openscad/src/geometry/GeometryUtils.cc
```
However manifold.h is in submodules/manifold/src/manifold/include/manifold.h

zougloub, 2024-04-19:
Other issue:
```
In file included from /home/cJ/dev/media/media-gfx/openscad/src/geometry/cgal/CGALCache.cc:5:
/home/cJ/dev/media/media-gfx/openscad/src/geometry/manifold/ManifoldGeometry.h: In member function ‘virtual void ManifoldGeometry::accept(GeometryVisitor&) const’:
/home/cJ/dev/media/media-gfx/openscad/src/geometry/manifold/ManifoldGeometry.h:16:71: error: no matching function for call to ‘GeometryVisitor::visit(const ManifoldGeometry&)’
```

zougloub, 2024-04-19:
Other issue:
```
In file included from /home/cJ/dev/media/media-gfx/openscad/src/geometry/cgal/CGALCache.h:5,
                 from /home/cJ/dev/media/media-gfx/openscad/src/geometry/cgal/CGALCache.cc:1:
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:58:16: note: candidate: ‘virtual void GeometryVisitor::visit(const GeometryList&)’
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:58:42: note:   no known conversion for argument 1 from ‘const ManifoldGeometry’ to ‘const GeometryList&’
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:59:16: note: candidate: ‘virtual void GeometryVisitor::visit(const PolySet&)’
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:59:37: note:   no known conversion for argument 1 from ‘const ManifoldGeometry’ to ‘const PolySet&’
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:60:16: note: candidate: ‘virtual void GeometryVisitor::visit(const Polygon2d&)’
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:60:39: note:   no known conversion for argument 1 from ‘const ManifoldGeometry’ to ‘const Polygon2d&’
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:62:16: note: candidate: ‘virtual void GeometryVisitor::visit(const CGAL_Nef_polyhedron&)’
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:62:49: note:   no known conversion for argument 1 from ‘const ManifoldGeometry’ to ‘const CGAL_Nef_polyhedron&’
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:63:16: note: candidate: ‘virtual void GeometryVisitor::visit(const CGALHybridPolyhedron&)’
/home/cJ/dev/media/media-gfx/openscad/src/geometry/Geometry.h:63:50: note:   no known conversion for argument 1 from ‘const ManifoldGeometry’ to ‘const CGALHybridPolyhedron&’
```

zougloub, 2024-04-19:
@t-paul `target_link_libraries(OpenSCAD PRIVATE manifold)` however there's no include to it?

zougloub, 2024-04-19:
My bad, it's

> @t-paul `target_link_libraries(OpenSCAD PRIVATE manifold)` however there's no include to it?

Oh, are we supposed to build with EXPERIMENTAL=ON ? manifold isn't included if EXPERIMENTAL is off.

zougloub, 2024-04-19:
Side note, I've been pointed to https://github.com/gentoo/gentoo/pull/35936 which has de-vendoring CMake patches to OpenSCAD, perhaps you've not seen this and it could be of interest.

t-paul, 2024-04-19:
Yes, at this point building with EXPERIMENTAL=OFF is not in focus. In theory it should always work but with the big changes going on and no CI verifying that, it' not feasible to maintain. It's likely going to be setup once some of the transitions conclude.

kintel, 2024-04-19:
See also #4875

kintel, 2024-04-19:
Once every few months, I do a non-experimental build locally and fix stuff up, but as t-paul says, it's not really maintained. Also, it kind of makes sense that HEAD is experimental as that's our main dev branch.
Setting up builds with Manifold working would be great, as that will at some point become a requirement.

zougloub, 2024-04-19:
Thank you! Off-topic but I think you'll find in that gentoo PR above some interesting cmake additions, I'm not sure if the contributors reached out to you already.

t-paul, 2024-04-19:
Yes, @cjmayo has, specifically regarding Qt6. If that covers everything, I don't know.

t-paul, 2024-04-22:
Fixed via #5095, CI via #5097.

closed 2024-04-22 by t-paul