openscad/openscad #5094: OpenScad will not load on Mac

RetiredJake, 2024-04-20:
**Describe the bug**
A clear and concise description of what the bug is.
Downloaded "OpenSCAD-2021.01.dmg" on 4/19/24. When trying to install, received the following error message:
“OpenSCAD.app” cannot be opened because the developer cannot be verified.
Perhaps some verification certificate has expired ?
Can really use some help so I can get the program loaded and start learning it.
**To Reproduce**
Steps to reproduce the behavior:
1. Download "OpenSCAD-2021.01.dmg"
2. Click on downloaded file
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Software to be installed and opened
**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Environment and Version info (please complete the following information):**
 - OS: macOS 12.7.4 Monterey
 - System: 2015 MacBook Pro
 - OpenSCAD Version [OpenSCAD-2021.01.dmg] downloaded from "https://openscad.org"

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

kintel, 2024-04-20:
If you Ctrl-click on the icon, you can bypass the "unverified developer" check: https://support.apple.com/en-ca/guide/mac-help/mh40616/mac

We're currently not signing our binaries due to Apple's stringent developer policy (they only accept incorporated entities for non-personal account and charges an annual fee), so we haven't been motivated to jump through the hoops.

kintel, 2024-04-20:
Closing as duplicate of #880

closed 2024-04-20 by kintel